MichaIng/DietPi #4854: Geo Location not accurate (wrong state/province)

maartenlangeveld, 2021-10-17:
DietPi v7.7.3

The new geo location routine is not accurate: my country is good but state/province not.

It says "Nort Brabant NL" whilst it should be "North Holland NL" (this is where my provider is located, I muyself live in "South Holland NL").

maartenlangeveld, 2021-10-17:
Not the same with me.

G_GET_WAN_IP gives a false state/province (not accurate), whilst curl -sSfL https://freegeoip.app/csv/ 2>&1 | mawk -F, '($5){r=$5" "}{print $1" "r$3}' only displays country (which is accurate)

```
2021-10-17 15:05:53 root@M-DNS:~# G_GET_WAN_IP
----:----:---:-----:----:----:---- North Brabant NL2021-10-17 15:05:56 root@M-DNS:~# curl -sSfL https://freegeoip.app/csv/ 2>&1 | mawk -F, '($5){r=$5" "}{print $1" "r$3}'
----:----:---:-----:----:----:---- Netherlands
```

MichaIng, 2021-10-17:
Ah so freegeoip.app doesn't have a region at all for that IP.

Yeah not sure, I mean we could remove the region as it is obviously not accurate in European countries. On the other hand in the USA it seems to be accurately showing the state. The idea behind showing the location is especially for cases with a VPN or proxy to show whether it's working as expected. Usually the country is sufficient for this, but in large countries the federal state may be interesting as well. In theory we could make it complicated and show the region only in cases where we know it is accurate, but I'd rather keep it simple. Does it disturb to have an inaccurate region shown?

MichaIng, 2021-10-17:
Out of interest, does it make a difference whether you access with IPv4 or IPv6?
```sh
curl -sSf4 https://dietpi.com/geoip
```

maartenlangeveld, 2021-10-17:
(1) It does not disturb me,m I just noticed it.
(2) IPv4 makes a difference, i.e. accurate location (state/province):

```
2021-10-17 15:40:08 root@M-DNS:/mnt/sda1# curl -sSf4 https://dietpi.com/geoip
147.78.236.125 South Holland NL2021-10-17 15:40:53 root@M-DNS:/mnt/sda1#
```

MichaIng, 2021-10-17:
Interesting. In my case it doesn't make a difference, but if we find some more hints where it is accurate with IPv4 but not with IPv6 (and not the other way round), we could force IPv4 usage for this request.

maartenlangeveld, 2021-10-17:
Ok, thanks!

maartenlangeveld, 2021-10-17:
Thans a lot for your explanation, seems very logical. Querying different online geo location databases with my IPv6 address give four different results for the region: North Holland, South Holland, North Brabant and <none>. With IPv4 there are also differences. So conclusion is that those databases are only realibale when country is concerned.