MichaIng/DietPi #4855: QUEST: Shutdown not working using PXE boot

Joulinar, 2021-10-17:
Hi,

I'm still playing around trying to use my test systems using PXE boot. Booting themselves is not an issue and my RPi4B is starting using RootFS located on a NFS share 😃 

```
root@DietPi4:~# df -h
Filesystem                                Size  Used Avail Use% Mounted on
192.168.0.10:/volume1/NFS/pxe/rpi4_armv6  2.7T  806G  1.9T  31% /
devtmpfs                                  3.9G     0  3.9G   0% /dev
tmpfs                                     3.9G     0  3.9G   0% /dev/shm
tmpfs                                     1.6G  8.6M  1.6G   1% /run
tmpfs                                     5.0M     0  5.0M   0% /run/lock
tmpfs                                     3.9G     0  3.9G   0% /tmp
tmpfs                                      50M  8.0K   50M   1% /var/log
```

Currently my challenge is to showdown/reboot the system. It's going to hang on bringing down the interfaces. I guess issue is somewhere in the order of actions. NFS share seems to disconnect before all scrips finished. Maybe you know what status the system is hanging on and if there is a way to shift thinks around and to have NFS share disconnected last? Probably a chicken/ egg challenge 🤔 

[image]

MichaIng, 2021-10-17:
Probably it's the other way round. `ifup@eth0` does `ifdown eth0` when stopping, hence deconfigures the network interface and would hence cut the NFS connection. Probably this is prevented somehow when this NFS mount serves the root filesystem.

Two ideas:
- We forced `ifup@.service` to be a oneshot service type (instead of default "simple"), so that it needs to finish (instead of only just start) before other targets and services are reached/start which order themselves after it. This is why `network-online.target` now really means that the network is online, at least that the interfaces have been fully configured, before it was basically meaningless, only the loopback interface was up for sure. But when `ifup@eth0` is stopped now, the effect is probably the same: The shutdown sequence does not continue until `ifdown eth0` finished, which hangs. To test this:
    ```sh
    echo -e '[Service]\nType=simple' > /etc/systemd/system/ifup@.service.d/override.conf
    systemctl daemon-reload
    ```
- Since the above breaks the positive intention of using a oneshot interface, another idea is to simply stop the interface deconfiguration at shutdown. I'm not aware of any important effect of this when the system is powered of anyway. At least with PXE it makes sense that the NFS mount is active as long as possible for all important shutdown steps to finish:
    ```sh
    echo -e '[Service]\nExecStop=' > /etc/systemd/system/ifup@.service.d/override.conf
    systemctl daemon-reload
    ```

Joulinar, 2021-10-17:
ok `ifup` seems to be fixed by setting `ExecStop=` to an empty value. However it's gonna stale now on `Raise network interfaces` 🙃

MichaIng, 2021-10-17:
So we gonna fix it the same way then:
```sh
mkdir -p /etc/systemd/system/networking.service.d
echo -e '[Service]\nExecStop=' > /etc/systemd/system/networking.service.d/override.conf
systemctl daemon-reload
```
Let's see, here it is actually strange since it does not deconfigure any actual interface:
```sh
ifdown -a --read-environment --exclude=lo
```
With the `-a` flag, all interfaces defined with `auto <iface>` in `/etc/network/interfaces` are affected, but we use `allow-hotplug <iface>`, which are deconfigured by `ifup@<iface>.service` only. And the loopback device is explicitly excluded here. However, let's see if it still helps.

Joulinar, 2021-10-17:
I guess I will flash a RPi OS to have it stored on NFS as well. Let's see how it behaves. At least on all the guides, even the Raspberry own one, I haven't seen such steps. Description was always in how to boot and not on how to shutdown 😂

Joulinar, 2021-10-17:
ok i just copied the partitions from RPi OS image file to my NFS share and was able to boot RPi OS arm64 without issues. As well reboot is working without issues.

Joulinar, 2021-10-17:
```
root@raspberrypi:~# systemctl status networking
● networking.service - Raise network interfaces
   Loaded: loaded (/lib/systemd/system/networking.service; enabled; vendor preset: enabled)
   Active: active (exited) since Mon 2021-10-18 00:08:33 BST; 20min ago
     Docs: man:interfaces(5)
  Process: 179 ExecStart=/sbin/ifup -a --read-environment (code=exited, status=0/SUCCESS)
 Main PID: 179 (code=exited, status=0/SUCCESS)

Oct 18 00:08:31 raspberrypi systemd[1]: Starting Raise network interfaces...
Oct 18 00:08:33 raspberrypi systemd[1]: Started Raise network interfaces.
```
```
root@raspberrypi:~# ls -l /etc/network/*.d/*
-rwxr-xr-x 1 root root  301 Jul 25  2011 /etc/network/if-down.d/resolvconf
lrwxrwxrwx 1 root root   32 Apr 16  2021 /etc/network/if-down.d/wpasupplicant -> ../../wpa_supplicant/ifupdown.sh
-rwxr-xr-x 1 root root 1409 Jun  5  2018 /etc/network/if-post-down.d/wireless-tools
lrwxrwxrwx 1 root root   32 Apr 16  2021 /etc/network/if-post-down.d/wpasupplicant -> ../../wpa_supplicant/ifupdown.sh
-rwxr-xr-x 1 root root  344 Jun 30  2016 /etc/network/if-pre-up.d/ethtool
-rwxr-xr-x 1 root root 4191 Sep 15  2018 /etc/network/if-pre-up.d/wireless-tools
lrwxrwxrwx 1 root root   32 Apr 16  2021 /etc/network/if-pre-up.d/wpasupplicant -> ../../wpa_supplicant/ifupdown.sh
-rwxr-xr-x 1 root root  703 Jul 25  2011 /etc/network/if-up.d/000resolvconf
-rwxr-xr-x 1 root root 1685 Jun 30  2016 /etc/network/if-up.d/ethtool
lrwxrwxrwx 1 root root   32 Apr 16  2021 /etc/network/if-up.d/wpasupplicant -> ../../wpa_supplicant/ifupdown.sh
root@raspberrypi:~#

```

MichaIng, 2022-02-06:
I lost strack about this issue. Is there still something missing on current DietPi images for PXE boot to work OOTB?

Joulinar, 2022-02-06:
booting is working fine, but in my setup horrable slow. But this is due to the NFS server I guess. Will close this one.

closed 2022-02-06 by Joulinar