MichaIng/DietPi #4857: Updating to Bullseye

barukh27, 2021-10-17:
Hi there!

I'm upgrading from Buster to Bullseye following this tutorial: https://dietpi.com/blog/?p=811#2.2-manual-upgrade

I think I've followed every step but when I run the command  apt upgrade I receive the following error:

[image]

What should I do?

Thanks in advance.

MichaIng, 2021-10-17:
Many thanks for your report.

Has the Raspberry Pi repository migrated to Bullseye as well?
```sh
cat /etc/apt/sources.list.d/raspi.list
```

MichaIng, 2021-10-17:
And if you upgrade `g++` separately?
```sh
apt install --autoremove g++
```

barukh27, 2021-10-17:
Idem

[image]

MichaIng, 2021-10-17:
Okay, then we need to purge it. This will remove a bunch of other development packages and compilers, but shouldn't be any end user software package:
```sh
apt purge --autoremove libgcc-8-dev
```
If you require them, you can re-install them later:
```sh
apt install g++
```
But if you installed software via `dietpi-software` only, then it is not required for regular usage and will be automatically reinstall when you (re)install software which requires the C/C++ toolkit for installation.

barukh27, 2021-10-17:
Hi again.

Now I've got this issue.

[image]

Joulinar, 2021-10-17:
Could you try running

```
apt full-upgrade
```

barukh27, 2021-10-17:
thank you for your answer.

That's the error I got.