MichaIng/DietPi #4861: Simple routine upgrade: unable to, exited with error code

MAGA6226, 2021-10-17:
#### Details:
- Date           | Sun Oct 17 14:15:51 CDT 2021
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux dietpi20053101 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ... Nothing much to say, I try not to disturb the system as much as possible but e/month or so I try to update/upgrade, however, it doesn't fail that every time, there's something that prevents to. 
2. ... People do not maintain the sites, links, etc., therefore, the process cannot continue unimpeded every time. 
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Setting up bind9 (1:9.11.5.P4+dfsg-5.1+deb10u5) ...
bind9-pkcs11.service is a disabled or a static unit not running, not starting it.
bind9-resolvconf.service is a disabled or a static unit not running, not starting it.
Job for bind9.service failed because the control process exited with error code.
See "systemctl status bind9.service" and "journalctl -xe" for details.
invoke-rc.d: initscript bind9, action "restart" failed.

MichaIng, 2021-10-17:
The bind9 services fail to start. Do you actively use them (a DNS nameserver)?

Check the service logs:
```sh
journalctl -u bind9
```

MichaIng, 2021-11-24:
I'll mark this as closed. Feel free to reopen when required.

closed 2021-11-24 by MichaIng