MichaIng/DietPi #4863: Partition missing after Bullseye upgrade

barukh27, 2021-10-17:
Creating a bug report/issue
Required Information
DietPi version |
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Distro version |bullseye 1

Kernel version | Linux DietPi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux

SBC model | RPi4

Power supply used | (EG: 5V 1A RAVpower)

Steps to reproduce
1. I've upgraded to Bullseye.
2. The RPi boots from a sdd.
3. Previous to the upgrade process I had a partition in the same SSD mounted in /mnt/HAL
4. Now the partition (HAL) is gone and some programs can't access the files on it.

Joulinar, 2021-10-17:
following could be interesting as well

`lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid`

barukh27, 2021-10-17:
I think I was wrong about the partition issue. My bad. I thought so because Jellyfin was unable to play any single file. An error was prompted that the file wasn't playable. This issue appears when the files are missing. I managed to find the folder where I store the files in HAL and I've rebuilt the libraries.

The problem is that Jellyfin can´t play some files. It's not a compatibility issue because I've tried with some videos that I've already watched.

MichaIng, 2021-10-17:
Then I guess it's because of the broken Jellyfin FFmpeg which requires a rebuild: https://github.com/jellyfin/jellyfin-ffmpeg/issues/75

Check out: #4821

barukh27, 2021-10-17:
I think so too. Let's close this thread. Thank you.

closed 2021-10-17 by barukh27