MichaIng/DietPi #4864: strange text DietPi-DDNS

maartenlangeveld, 2021-10-17:
DietPi v7.7.3 (bullseye), NanoPi NEO2 (aarch64)

When configuring DietPi-DDNS some strange text appears before ('nochg') and after ('^M') the IPv4 address

> DDNS update test succeeded:
> **nochg** 147.xxx.xxx.xxx **^M**

```

                    ┌───────────────────────────────────────────────────┤ DietPi-DDNS ├────────────────────────────────────────────────────┐
                    │ DDNS update test succeeded:                                                                                          │
                    │ nochg 147.xxx.xxx.xxx^M                                                                                              │
                    │                                                                                                                      │
                    │                               Provider : [No-IP]                                                                     │
                    │                               Domains  : [---.ddns.net]                                                              │
                    │                               Username : [---]                                                                       │
                    │                               Password : [*********]                                                                 │
                    │                               Timespan : [10 minutes]                                                                │
                    │                                        ●─ Apply ─────────────────────────────────────●                               │
                    │                               Apply    : Create or update Cron job with above settings                               │
                    │                               Remove   : Remove any DDNS updates from this system                                    │
                    │                                                                                                                      │
                    │                                                                                                                      │
                    │                                  <Ok>                                      <Exit>                                    │
                    │                                                                                                                      │
                    └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

```

Joulinar, 2021-10-17:
If I'm not mistaken this is the message provided by DDNS provider

`nochg` means the domain already maps to the IP address from where the IP update is sending from

maartenlangeveld, 2021-10-18:
@Joulinar

Thanks for your explanation ! Makes sense :-)

closed 2021-10-18 by maartenlangeveld

MichaIng, 2021-10-18:
Probably we can prefix that line with `DDNS server response:` or so. This was mostly added to identify when a "positive" (code 2xx) HTTP response happened but the content actually means an error. Bad practice IMHO but some providers do it like that and we added manual keyword detection for those. In this case it is fine, just informing you that nothing needs to be updated, i.e. your public IP didn't change from last request.