MichaIng/DietPi #4865: DietPi won't boot after initial setup

Megum1n, 2021-10-18:
Just installed DietPi and it won't boot. 
Tried to reinstall couple of times. Also tried to upgrade to bullseye. Same result.
I'm not familiar with arm devices bootloaders, so please help.
I can boot from setup iso again, update os and get some files/run some scripts for you if needed.

## Info
**Device**: NanoPi NEO
**Version**: Latest DietPi atm (buster) (6f21077ad0a7ec0acce7c05016be8ad5b9600251 shasum)
Ethernet and power connected.

## Steps to reproduce
1. Flash rom on sdcard
2. Boot from sdcard
3. Do initial setup (passwords/upgrade/etc)
4. Reboot

## Expected behaviour
Device is booted to OS

## Actual behaviour (serial output)
```
U-Boot SPL 2021.04-armbian (May 06 2021 - 17:50:40 +0000)
DRAM: 512 MiB
Trying to boot from MMC1

U-Boot 2021.04-armbian (May 06 2021 - 17:50:40 +0000) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: FriendlyARM NanoPi NEO
DRAM:  512 MiB
MMC:   mmc@1c0f000: 0, mmc@1c11000: 1
Loading Environment from FAT... Card did not respond to voltage select! : -110
In:    serial
Out:   serial
Err:   serial
Net:   phy interface0

Error: ethernet@1c30000 address not set.
No ethernet found.

starting USB...
Bus usb@1c1a000: USB EHCI 1.00
Bus usb@1c1a400: USB OHCI 1.0
Bus usb@1c1d000: USB EHCI 1.00
Bus usb@1c1d400: USB OHCI 1.0
scanning bus usb@1c1a000 for devices... 1 USB Device(s) found
scanning bus usb@1c1a400 for devices... 1 USB Device(s) found
scanning bus usb@1c1d000 for devices... 1 USB Device(s) found
scanning bus usb@1c1d400 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Autoboot in 1 seconds, press <Space> to stop
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
3967 bytes read in 3 ms (1.3 MiB/s)
## Executing script at 43100000
U-boot loaded from SD
Boot script loaded from mmc
282 bytes read in 3 ms (91.8 KiB/s)
Failed to load '/boot/uInitrd'
7995288 bytes read in 599 ms (12.7 MiB/s)
Found mainline kernel configuration
32430 bytes read in 11 ms (2.8 MiB/s)
504 bytes read in 9 ms (54.7 KiB/s)
Applying kernel provided DT overlay sun8i-h3-usbhost1.dtbo
504 bytes read in 9 ms (54.7 KiB/s)
Applying kernel provided DT overlay sun8i-h3-usbhost2.dtbo
4185 bytes read in 9 ms (454.1 KiB/s)
Applying kernel provided DT fixup script (sun8i-h3-fixup.scr)
[... 364 lines cut ...]
mdio_register: non unique device name 'ethernet@1c30000'

Error: ethernet@1c30000 address not set.
phy interface0
mdio_register: non unique device name 'ethernet@1c30000'

Error: ethernet@1c30000 address not set.
phy interface0
mdio_register: non unique device name 'ethernet@1c30000'

Error: ethernet@1c30000 address not set.
phy interface0
mdio_register: non unique device name 'ethernet@1c30000'

Error: ethernet@1c30000 address not set.
phy interface0
mdio_register: non unique device name 'ethernet@1c30000'

Error: ethernet@1c30000 address not set.
No ethernet found.
phy interface0
mdio_register: non unique device name 'ethernet@1c30000'

Error: ethernet@1c30000 address not set.
```

MichaIng, 2021-10-18:
As said, Bullseye image is up now (also linked from download page and docs), so no need to run PREP anymore 😉.

MichaIng, 2021-10-18:
Yes. Issue was that DietPi-FirstBoot didn't use the `-q` (=quiet) flag when applying network interfaces when G_GET_NET unintentionally prints errors (no matching interface found) to STDOUT instead of STDERR. The letter is not a big issue but in combination both lead to an error message in `/etc/network/interfaces` instead of the intended default/fallback eth0/wlan0. DietPi-PREP now applies the `-q` flag to the `dietpi-firstboot.bash` script until next release has this master.