MichaIng/DietPi #4876: Firefly III Personal Finance Manager

walker93, 2021-10-20:
### Creating a software request

#### Formal software information
- Software title | Firefly III
- Short description | self-hosted personal expenses and financial manager
- Official URL (if available) | https://www.firefly-iii.org/
- GitHub URL (if available) | https://github.com/firefly-iii/firefly-iii
- (Official) install documentation | https://docs.firefly-iii.org/firefly-iii/installation/self_hosted/

#### Are there similar/alternative software titles available with DietPi-Software?

- Didn't found any

#### What makes your requested software better than the above solutions, if available?
It is fully open source, without any (feature) limitation to free users. It has a large feature range based on available apps and a large active community.
Didn't find any other self-hosted solutions capable of all the features this software supports. 

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
- it also supports docker and docker compose installations if it's easier to automate.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. Install a LAMP stack with php 8.0
2. it also requires the following php modules:
   - PHP BCMath Arbitrary Precision Mathematics
   - PHP Internationalization extension
   - PHP Curl
   - PHP Zip
   - PHP GD
   - PHP XML
   - PHP MBString
   - PHP LDAP

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
- Sorry, I am very unexperienced in bash coding.

walker93, 2021-10-21:
Yeah, that was my concern too.
Unfortunately in the docs they don't provide any support to older versions of PHP, the good news is that somehow this proves how up to date the project is and how is currently actively maintained.

Do you think would be possible to automate the DietPi-Software installation using docker compose?
[https://docs.firefly-iii.org/firefly-iii/installation/docker/]( https://docs.firefly-iii.org/firefly-iii/installation/docker/)

Any suggestions on how I can install it independently without messing my current Nextcloud Installation?
Thanks

Joulinar, 2021-10-21:
Usually we try to avoid including docker containers into `dietpi-software` as there is not that much we can automate. We offer all needed tools like docker, docker compose as install option. For starting the container you simply could follow the guide you linked. Only thing needed would be to change the port exposed as port 80 is already used by Nextcloud web server.