MichaIng/DietPi #4913: First install of nzbget fails due to lack of permssion on config file (also tries to install vpn)

kerryland, 2021-10-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.7.3
- Distro version | bullseye 0
- Kernel version | Linux blackbox 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | (5V 1A Official PI Power supply)
- SDcard used | (Panasonic)

#### Additional Information (if applicable)
- Software title | nzbget 
- Was the software title installed freshly or updated/migrated? -- fresh install
- Can this issue be replicated on a fresh installation of DietPi? -- this is a fresh install

#### Steps to reproduce

- Use dietpi-software to install nzbget
- Watch services automatically stop and restart
- Notice that nzbget is 'Failed'
- Start dietpi-services, select 'nzbget' and choose "Status", which tells me "Could not open file /mnt/EE40769B407669EB/dietpi_userdata/nzbget/nzbget.conf".
- Look at nzbget.conf and notice it has interesting access rights:
```
---------- 1 nzbget root   66974 Oct 31 10:54 nzbget.conf
```

#### Expected behaviour
- nzbget should start with working config
 
#### Actual behaviour
- nzbget does not start
- dietpi-vpn is also installed, despite not being selected. It should not be auto-installed. It too fails to start:
```
  │ Oct 31 10:59:38 blackbox systemd[1]: Starting VPN Client (DietPi)...                                                 │
  │ Oct 31 10:59:39 blackbox systemd[138971]: dietpi-vpn.service: Failed to locate executable /usr/sbin/openvpn: No such │
  │ file or directory                                                                                                    │
  │ Oct 31 10:59:39 blackbox systemd[138971]: dietpi-vpn.service: Failed at step EXEC spawning /usr/sbin/openvpn: No     │
  │ such file or directory                                                                                               │
  │ Oct 31 10:59:39 blackbox systemd[1]: dietpi-vpn.service: Main process exited, code=exited, status=203/EXEC 
```

#### Extra details
```
 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing NZBGet: nzb download manager

[  OK  ] DietPi-Software | curl -sSfL https://nzbget.net/download/nzbget-latest-bin-linux.run -o package.run
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/nzbget
[ INFO ] DietPi-Software | dash package.run --destdir /mnt/dietpi_userdata/nzbget, please wait...
Installer for nzbget-21.1
Verifying package...
Checking system...
CPU-Architecture: aarch64
Unpacking...
Configuring...
  Free memory detected: 4938 MB
  Activating article cache (ArticleCache=100)
  Increasing write buffer (WriteBuffer=1024)
  Increasing par repair buffer (ParBuffer=100)
  Activating direct rename (DirectRename=yes)
  CPU speed detected: 432 BogoMIPS
  Disabling download during par check/repair (ParPauseQueue=yes)
  Disabling download during unpack (UnpackPauseQueue=yes)
  Disabling download during post-processing (ScriptPauseQueue=yes)
Installation completed

Quick help (from nzbget-directory):
   ./nzbget -s        - start nzbget in console mode
   ./nzbget -D        - start nzbget in daemon mode (in background)
[... 82 lines cut ...]
-rw-r--r-- 1 nzbget root   15876 Jun  4 06:15 README
drwxr-xr-x 1 nzbget root       0 Jun  4 06:15 scripts
-rwxr-xr-x 1 nzbget root 1047368 Jun  4 06:15 unrar
drwxr-xr-x 1 nzbget root    4096 Jun  4 06:15 webui

  ┌──────────────────────────────────────────┤ DietPi-Services | File viewer ├───────────────────────────────────────────┐
  │                                                                                                                      │
  │ ● nzbget.service - NZBget (DietPi)                                                                                   │
  │      Loaded: loaded (/etc/systemd/system/nzbget.service; disabled; vendor preset: enabled)                           │
  │      Active: failed (Result: exit-code) since Sun 2021-10-31 10:59:36 NZDT; 14min ago                                │
  │     Process: 138834 ExecStart=/mnt/dietpi_userdata/nzbget/nzbget -D (code=exited, status=1/FAILURE)                  │
  │         CPU: 12ms                                                                                                    │
  │                                                                                                                      │
  │ Oct 31 10:59:36 blackbox systemd[1]: Starting NZBget (DietPi)...                                                     │
  │ Oct 31 10:59:36 blackbox nzbget[138834]: nzbget.conf(0): Could not open file                                         │
  │ /mnt/EE40769B407669EB/dietpi_userdata/nzbget/nzbget.conf                                                             │
  │ Oct 31 10:59:36 blackbox systemd[1]: nzbget.service: Control process exited, code=exited, status=1/FAILURE           │
  │ Oct 31 10:59:36 blackbox systemd[1]: nzbget.service: Failed with result 'exit-code'.                                 │
  │ Oct 31 10:59:36 blackbox systemd[1]: Failed to start NZBget (DietPi).                                                │
  │                                                                                                                      │
  │                                                        <Ok>                                                          │
  │                                                                                                                      │
  └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

```

Joulinar, 2021-10-30:
Looks like you have located you user directory on an external disk. Could it be an NTFS drive?

`lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid`

kerryland, 2021-10-30:
Yes, it is an NTFS drive. I'm setting up a pi for the first time, and moving from a Windows machine.

Joulinar, 2021-10-30:
Ok that's basically a limitation of NTFS. Personally I would recommend using ext4 on your external disk. To get `nzbget` working now, you would need to set r/w permissions on the config file.

```
chmod 600 /mnt/EE40769B407669EB/dietpi_userdata/nzbget/nzbget.conf
```

 In future we might prevent moving user data to NTFS drive's at all to avoid such issues.

MichaIng, 2021-10-31:
Strange that only this one file is affected. The installer archive contains correct modes. It extracts everything, then only the configuration file is copied via regular `cp` from the template in place, so that the default mode 644 should apply:
<details>

```
#!/bin/sh
#
#  This file is part of nzbget. See <http://nzbget.net>.
#
#  Copyright (C) 2015-2017 Andrey Prygunkov <hugbug@users.sourceforge.net>
#
#  This program is free software; you can redistribute it and/or modify
#  it under the terms of the GNU General Public License as published by
#  the Free Software Foundation; either version 2 of the License, or
#  (at your option) any later version.
#
#  This program is distributed in the hope that it will be useful,
#  but WITHOUT ANY WARRANTY; without even the implied warranty of
#  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
#  GNU General Public License for more details.
#
#  You should have received a copy of the GNU General Public License
#  along with this program.  If not, see <http://www.gnu.org/licenses/>.
#

# Strict error handling for debugging
set -o nounset
set -o errexit

# Installer title
TITLE="nzbget-21.1"
# Size of installer script (package header)
HEADER=17253
# Size of installer package (header + payload)
TOTAL=31535485
# Md5 sum of payload
MD5="29a07f5d8f982a84591f32fb2071991b"
# List of included CPU architecture binaries
DISTARCHS="i686 x86_64 armel armhf aarch64 mipsel mipseb ppc6xx ppc500"
# Target platform
PLATFORM="linux"

SILENT=no
ALLARCHS="$DISTARCHS all"
ARCH=""
SELECT=auto
OUTDIR="nzbget"
PRINTEDTITLE=no
JUSTUNPACK=no
UPDATE=no
VERIFY=yes
OS=""

Info()
{
    if test "$SILENT" = "no"; then
        echo "$1"
    fi
}

Error()
{
    Info "ERROR: $1"
    exit 1
}

ValidArch()
{
    for TARG in $ALLARCHS; do
        if test "$TARG" = "$1"; then
            return 0
[... 499 lines cut ...]
            # OK, a default route 0.0.0.0 + corresponding interface was found
            # Now find the IPv4 address on that interface:
            {
                IP=`ifconfig "$INTERFACE" | sed -rn 's/.*r:([^ ]+) .*/\1/p'`
            } > /dev/null 2>&1
        fi
        if test "$IP" = ""; then
            IP="localhost"
        fi

        Info "Web-interface is on http://$IP:6789 (login:nzbget, password:tegbzn6789)"
    else
        Info "Successfully installed into $ABSOUTDIR"
    fi
    Info "For support please visit http://nzbget.net/forum"
    Linux2FreeBSD "kernel-check"
else
    Info "Unpacked into $ABSOUTDIR"
fi

exit
```
</details>

The installer is actually an interesting file. The top is this regular shell script but the remaining 30 MiB are a tarball.

Joulinar, 2021-10-31:
@MichaIng 
Issue is similar to AGH configuration file that had none permissions after copied to a NTFS drive.

MichaIng, 2021-10-31:
True, I remember. I pushed a commit which sets the mode for the config file explicitly, stricter than it was before, since it contains the control password in plain text. So basically the same workaround that we applied for AGH.

I'll try to play around with copying files to NTFS. Usually this should all work fine with ntfs-3g driver, while it is true that these emulated UNIX permissions are not reliable enough to allow userdata being stored on it.

MichaIng, 2021-11-11:
Moving userdata to NTFS drives won't be supported anymore with v7.8. Good to keep in mind that some users might still have them there, in case any permissions issues are reported, but I'm going to close the issue. Feel free to reopen when required.

closed 2021-11-11 by MichaIng