MichaIng/DietPi #4914: "Catch 22" in DietPi-Software for Browser and Desktop

kerryland, 2021-10-30:
#### Required Information

    DietPi version | 7.7.3
    Distro version | bullseye 0
    Kernel version | Linux blackbox 5.10.63-v8+ 1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
    SBC model | RPi 4 Model B (aarch64)
    Power supply used | (5V 1A Official PI Power supply)
    SDcard used | (Panasonic)

#### Additional Information (if applicable)
- Software title | DietPi-Software
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? This is a fresh install

#### Steps to reproduce
1. Think "I wonder how this Raspberry Pi 4 runs a desktop"
2. Start `dietpi-software`
3. Set "Desktop Preference" to xfce, and see it will be installed (only) when another selected software requires any desktop.
4. Set "Browser Preference" to Chromium, and see it will be installed (only) when a desktop is installed.
5. Click "Install" and be told "No changes have been detected. Unable to start installation"

Interestingly, I see a category called "Desktop Utilities" which I guess are all the software that requires a desktop. There are 5 items only the list, and two are Browsers.

#### Expected behaviour
When I click **Install** the system should notice that I have requested a browser, and a desktop, and so install them.

#### Actual behaviour
Nothing. There is no obvious way to start a web browser in a desktop GUI

#### Extra details
The work around is to go into "Browse Software" and select the Desktop environment and browser that you want.

Joulinar, 2021-10-30:
I guess a misunderstanding. Just setting the preference will not install anything automatically, as it is a preference setting only. It will be installed once another software require it, like a VNC server. Or if you select it actively.

kerryland, 2021-10-30:
I'm a first time user, so I'm going to encounter problems that experienced users won't. 

According to my understanding, a "preference" is a thing I'd like to use, although I'd be happy with something else if it isn't available. Maybe "preference" is a poor choice of wording.

A "default" is something that will be used unless I choose something else, so may I suggest that "Desktop Preference" and "Browser Preference" become "Desktop Default" and "Browser Default" instead?

I guess it doesn't matter much. I just found it a weird user experience.

kerryland, 2021-10-30:
I don't understand when "Browser Preference" would be used. Do you automatically install it when a desktop is installed? Is it used for anything else? Seems odd, but again, I'm a complete newbie.

I will leave you alone now.

Joulinar, 2021-10-30:
If you install a desktop, you could choose the browser using browser preference. It could be chromium, Firefox or none.

MichaIng, 2021-10-31:
Did you read the text(s) given in the menu?
>  Please select your preferred webserver, which will be installed (only) when another selected software requires any webserver:

> Please select your preferred desktop environment, which will be installed (only) when another selected software requires any desktop:

> Please select your preferred web browser, which will be installed (only) when a desktop is installed:

If you have a suggestion to make this clearer, that would be great. As opposed to the file server and SSH server "choices" this is indeed only used when another software title pulls a webserver/desktop/browser in as dependency, and yes browsers are as stated pulled in by desktops only currently.

kerryland, 2021-10-31:
I did read the text(s) in the menu. I even referenced them in the "steps to reproduce" :-)

How about this alternative wording:

"Which webserver should be installed if you download software that requires a webserver. e.g. ?????"

"Which desktop environment should be installed if you download software that requires a desktop. e.g. vnc"

"Which web browser should be installed if you download software that requires a browser e.g. LXDE or Xfce"

MichaIng, 2021-10-31:
> I did read the text(s) in the menu. I even referenced them in the "steps to reproduce" :-)

Ah yes, so looks like they are not clear enough for everyone. Actually I like your idea to turn them into a questions. I would replace "download" with "install", but otherwise to me it indeed looks clearer than the current sentences with the `(only)` inside. @Joulinar do you agree to change them that way? I mean it's not the first time this question/confusion arises, was present for the webserver preference for ages and now to added two other preferences.

Joulinar, 2021-10-31:
Fine with me @MichaIng

kerryland, 2021-10-31:
> I would replace "download" with "install", but otherwise to me it indeed looks clearer than the current sentences with the (only) inside.

Except that my first attempt at new wording loses the meaning behind that "only". Here's another version that tries to replicate the "only":

"If you install software (eg: vnc) that requires a desktop environment, and none is installed, which desktop should we install for you?"

MichaIng, 2021-11-02:
Found the "No session for PID" error, which is indeed a Bullseye-only problem: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=864402
https://github.com/WhitewaterFoundry/Pengwin/issues/429

As a workaround we will now unset the default polkit entry from the global lxsession configuration on Bullseye. I also found a way to set the desktop icon shadow and wallpaper/mode via global config file, so we'll do this now instead of restarting PCManFM on first login.

MichaIng, 2021-11-02:
PR up to fix the two issues: #4920
Also we do not override the whole default PCManFM and Openbox config files anymore but only apply the few settings to the existing one. This should generally enhance compatibility and also prevents manual customisations. Only for the lxpanel we override the default completely, as there is too much changed.

MichaIng, 2021-11-08:
I'll mark this as closed as of the fixes which are now in beta. Feel free to reopen when required.

closed 2021-11-08 by MichaIng

MichaIng, 2021-11-27:
@kerryland 
We forgot about the preference texts with last release. I now changed them for DietPi v7.9: https://github.com/MichaIng/DietPi/pull/4965/commits/a9c586fc2b1f72473a90fd1472c6f21a6311db3d
It is close to your first suggestions. I like those most as they are short and clear (I hope).