MichaIng/DietPi #4915: Install of Chromium fails with "libgtk-3-0" has no installation candidate

kerryland, 2021-10-30:
#### Required Information
DietPi version | 7.7.3
Distro version | bullseye 0
Kernel version | Linux blackbox 5.10.63-v8+ 1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
SBC model | RPi 4 Model B (aarch64)
Power supply used | (5V 1A Official PI Power supply)
SDcard used | (Panasonic)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated? No
- Can this issue be replicated on a fresh installation of DietPi? This is very fresh
- Bug report ID: 887f1797-48ee-431c-ab26-092ea9582b1c

#### Steps to reproduce
1. Start `dietpi-software` and select `Browse Software`.
2. Check boxes for `23 LXDE` and `113 Chromium`
3. Select `OK` and `Install`
4. Wait for failure advising `"libgtk-3-0" has no installation candidate`.

I've tried a couple of times with the same result. Works OK with Firefox.

#### Expected behaviour
- Chromium should install

#### Actual behaviour
```
│ APT install for: chromium-browser libgtk-3-0                                                                         │
                     │  - Command: apt-get -qq --allow-change-held-packages install chromium-browser libgtk-3-0                             │
                     │  - Exit code: 100                                                                                                    │
                     │  - DietPi version: v7.7.3 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 6                                   │
                     │  - Image creator: DietPi Core Team                                                                                   │
                     │  - Pre-image: From scratch                                                                                           │
                     │  - Error log:                                                                                                        │
                     │ E: Package 'libgtk-3-0' has no installation candidate  
```

kerryland, 2021-10-31:
After the Firefox install worked I decided to start the UI. "startx" was, apparently, a poor choice because the result was underwhelming, and I couldn't find a way back to the shell (the old standby of CTRL-ALT-BACKSPACE didn't work) so I killed a random process with lxde in its name, which "worked", and decided to reboot.

During shutdown I'm told that the OS can't unmount my drive, and upon restart I find myself looking at something orange called "CloudShell".

I love the _concept_ of Linux so much.

kerryland, 2021-10-31:
Uninstalling X-Windows, LXDE and Firefox worked well. I have no idea what turned on "CloudShell", but I found out how to turn it off. Fingers crossed, everything is working well.

MichaIng, 2021-10-31:
First of all, `libgtk-3-0` is a regular Debian package, available on all versions: https://packages.debian.org/bullseye/libgtk-3-0
And it is present on Raspbian: http://raspbian.raspberrypi.org/raspbian/pool/main/g/gtk+3.0/
And it is also present on the Raspberry Pi repository, newer override packages will are the ones that would be installed: https://archive.raspberrypi.org/debian/pool/main/g/gtk+3.0/

On my test system:
```
2021-10-31 02:36:05 root@Testing-ARMv6-Bullseye:~# apt show libgtk-3-0
Package: libgtk-3-0
Version: 3.24.24-4+rpt2
Priority: optional
Section: libs
Source: gtk+3.0
Maintainer: Debian GNOME Maintainers <pkg-gnome-maintainers@lists.alioth.debian.org>
Installed-Size: 8775 kB
Provides: gtk3-binver-3.0.0
Depends: adwaita-icon-theme, hicolor-icon-theme, shared-mime-info, libatk-bridge2.0-0 (>= 2.5.3), libatk1.0-0 (>= 2.35.1), libc6 (>= 2.29), libcairo-gobject2 (>= 1.14.0), libcairo2 (>= 1.14.0), libcolord2 (>= 0.1.10), libcups2 (>= 1.7.0), libepoxy0 (>= 1.4.3), libfontconfig1 (>= 2.12.6), libfreetype6 (>= 2.2.1), libfribidi0 (>= 0.19.7), libgdk-pixbuf-2.0-0 (>= 2.40.0), libglib2.0-0 (>= 2.59.0), libharfbuzz0b (>= 1.2.6), libjson-glib-1.0-0 (>= 1.5.2), libpango-1.0-0 (>= 1.45.5), libpangocairo-1.0-0 (>= 1.41.0), libpangoft2-1.0-0 (>= 1.41.0), librest-0.7-0 (>= 0.7), libwayland-client0 (>= 1.14.91), libwayland-cursor0 (>= 1.14.91), libwayland-egl1 (>= 1.15.0), libx11-6 (>= 2:1.4.99.1), libxcomposite1 (>= 1:0.4.5), libxcursor1 (>> 1.1.2), libxdamage1 (>= 1:1.1), libxext6, libxfixes3, libxi6 (>= 2:1.2.99.4), libxinerama1, libxkbcommon0 (>= 0.5.0), libxrandr2 (>= 2:1.5.0), libgtk-3-common (>= 3.24.24-4+rpt2)
Recommends: libgtk-3-bin, librsvg2-common
Suggests: gvfs
Homepage: https://www.gtk.org/
Download-Size: 2277 kB
APT-Sources: https://archive.raspberrypi.org/debian bullseye/main armhf Packages
Description: GTK graphical user interface library
 GTK is a multi-platform toolkit for creating graphical user
 interfaces. Offering a complete set of widgets, GTK is suitable
 for projects ranging from small one-off tools to complete application
 suites.
 .
 This package contains the shared libraries.
```
Can you run these commands on your system and check whether you see the same?
```sh
apt update
apt show libgtk-3-0
```

What was the "underwhelming" problem with LXDE? It runs perfectly smooth even on a Raspberry Pi Zero W here, so on an RPi 4 it should be overwhelming in terms of performance. To return to the console: Open the menu via button left panel icon and hit "logout". On the appearing prompt again hit "logout", as you can also shutdown or reboot from there.

Not sure about default hotkeys, but you can configure them, there is lxhotkey or so available by default as well from the panel menu.

Which drive could not be unmounted during shutdown? This means that there is something still accessing it, which shouldn't be the case on a regular shutdown. But probably something was hanging has of your manually killed desktop session.

There is no mechanism which would change the autostart option non-interactively. When you install a desktop, you should have seen a prompt which offers you to change the autostart option (as one may want to start a desktop session or desktop login prompt automatically), probably you selected DietPi-CloudShell that way? Generally to change it (which you seem to have found already), run `dietpi-autostart` from console.

> Uninstalling X-Windows, LXDE and Firefox worked well. I have no idea what turned on "CloudShell", but I found out how to turn it off. Fingers crossed, everything is working well.

You mean you uninstalled and reinstalled those, hence LXDE and Firefox do now work as intended?

kerryland, 2021-10-31:
> What was the "underwhelming" problem with LXDE?

On starting (via "startx") it immediately showed an error dialog. "No session for pid 30332", the desktop had no icons at all, and the horizontal task bar at the bottom of the screen was slicing off the top of some icons.

> You mean you uninstalled and reinstalled those, hence LXDE and Firefox do now work as intended?

No, I actually just uninstalled them completely -- I don't really need a UI, and was just taking a look out of interest. I have since done the installation again and although I still get a "No session for pid 30332" error dialog, this time the desktop looks much better with icons for "DietPi-Con...", "DietPi-Lau..." and "Htop".

In the task bar is a Firefox icon. When clicked it asks "This text file 'Firefox ESR' seems to be an executable script. What do you want to do with it". Maybe another NTFS permission problem? Firefox starts fine if I start it from the menu in the bottom left of the screen.

I've also tried xfce which I much prefer. No error dialog, and Firefox starts without asking any questions :-)

kerryland, 2021-10-31:
> ``` apt update```
```
root@blackbox:~# apt update
Hit:1 https://download.mono-project.com/repo/debian buster InRelease
Hit:2 https://deb.debian.org/debian bullseye InRelease
Hit:3 https://deb.debian.org/debian bullseye-updates InRelease
Hit:4 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:5 https://apt.sonarr.tv/debian buster InRelease
Hit:6 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:7 https://deb.debian.org/debian bullseye-backports InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
root@blackbox:~#
```

> `apt show libgtk-3-0`

```
root@blackbox:~# apt show libgtk-3-0
Package: libgtk-3-0
Version: 3.24.24-4+rpt2
Priority: optional
Section: libs
Source: gtk+3.0
Maintainer: Debian GNOME Maintainers <pkg-gnome-maintainers@lists.alioth.debian.org>
Installed-Size: 10.2 MB
Provides: gtk3-binver-3.0.0
Depends: adwaita-icon-theme, hicolor-icon-theme, shared-mime-info, libatk-bridge2.0-0 (>= 2.5.3), libatk1.0-0 (>= 2.35.1), libc6 (>= 2.29), libcairo-gobject2 (>= 1.14.0), libcairo2 (>= 1.14.0), libcolord2 (>= 0.1.10), libcups2 (>= 1.7.0), libepoxy0 (>= 1.4.3), libfontconfig1 (>= 2.12.6), libfreetype6 (>= 2.2.1), libfribidi0 (>= 0.19.7), libgdk-pixbuf-2.0-0 (>= 2.40.0), libglib2.0-0 (>= 2.59.0), libharfbuzz0b (>= 1.2.6), libjson-glib-1.0-0 (>= 1.5.2), libpango-1.0-0 (>= 1.45.5), libpangocairo-1.0-0 (>= 1.41.0), libpangoft2-1.0-0 (>= 1.41.0), librest-0.7-0 (>= 0.7), libwayland-client0 (>= 1.14.91), libwayland-cursor0 (>= 1.14.91), libwayland-egl1 (>= 1.15.0), libx11-6 (>= 2:1.4.99.1), libxcomposite1 (>= 1:0.4.5), libxcursor1 (>> 1.1.2), libxdamage1 (>= 1:1.1), libxext6, libxfixes3, libxi6 (>= 2:1.2.99.4), libxinerama1, libxkbcommon0 (>= 0.5.0), libxrandr2 (>= 2:1.5.0), libgtk-3-common (>= 3.24.24-4+rpt2)
Recommends: libgtk-3-bin, librsvg2-common
Suggests: gvfs
Homepage: https://www.gtk.org/
Download-Size: 2,516 kB
APT-Manual-Installed: no
APT-Sources: https://archive.raspberrypi.org/debian bullseye/main arm64 Packages
Description: GTK graphical user interface library

N: There is 1 additional record. Please use the '-a' switch to see it
```

but THEN I tried `apt show libgtk-3-0` again, after failing to install chromium again. This time with different results:

```
root@blackbox:~# apt show libgtk-3-0
Package: libgtk-3-0
State: not a real package (virtual)
N: Can't select candidate version from package libgtk-3-0 as it has no candidate
N: There are 2 additional records. Please use the '-a' switch to see them.
N: No packages found
```
So I did as I was told, and added `-a`:
```
root@blackbox:~# apt show -a libgtk-3-0
Package: libgtk-3-0
Version: 3.24.24-4+rpt2
Priority: optional
Section: libs
Source: gtk+3.0
Maintainer: Debian GNOME Maintainers <pkg-gnome-maintainers@lists.alioth.debian.org>
Installed-Size: 10.2 MB
Provides: gtk3-binver-3.0.0
Depends: adwaita-icon-theme, hicolor-icon-theme, shared-mime-info, libatk-bridge2.0-0 (>= 2.5.3), libatk1.0-0 (>= 2.35.1), libc6 (>= 2.29), libcairo-gobject2 (>= 1.14.0), libcairo2 (>= 1.14.0), libcolord2 (>= 0.1.10), libcups2 (>= 1.7.0), libepoxy0 (>= 1.4.3), libfontconfig1 (>= 2.12.6), libfreetype6 (>= 2.2.1), libfribidi0 (>= 0.19.7), libgdk-pixbuf-2.0-0 (>= 2.40.0), libglib2.0-0 (>= 2.59.0), libharfbuzz0b (>= 1.2.6), libjson-glib-1.0-0 (>= 1.5.2), libpango-1.0-0 (>= 1.45.5), libpangocairo-1.0-0 (>= 1.41.0), libpangoft2-1.0-0 (>= 1.41.0), librest-0.7-0 (>= 0.7), libwayland-client0 (>= 1.14.91), libwayland-cursor0 (>= 1.14.91), libwayland-egl1 (>= 1.15.0), libx11-6 (>= 2:1.4.99.1), libxcomposite1 (>= 1:0.4.5), libxcursor1 (>> 1.1.2), libxdamage1 (>= 1:1.1), libxext6, libxfixes3, libxi6 (>= 2:1.2.99.4), libxinerama1, libxkbcommon0 (>= 0.5.0), libxrandr2 (>= 2:1.5.0), libgtk-3-common (>= 3.24.24-4+rpt2)
Recommends: libgtk-3-bin, librsvg2-common
Suggests: gvfs
Homepage: https://www.gtk.org/
Download-Size: 2,516 kB
APT-Manual-Installed: no
APT-Sources: https://archive.raspberrypi.org/debian bullseye/main arm64 Packages
Description: GTK graphical user interface library
[... 6 lines cut ...]
Maintainer: Debian GNOME Maintainers <pkg-gnome-maintainers@lists.alioth.debian.org>
Installed-Size: 10.2 MB
Provides: gtk3-binver-3.0.0
Depends: adwaita-icon-theme, hicolor-icon-theme, shared-mime-info, libatk-bridge2.0-0 (>= 2.5.3), libatk1.0-0 (>= 2.35.1), libc6 (>= 2.29), libcairo-gobject2 (>= 1.14.0), libcairo2 (>= 1.14.0), libcolord2 (>= 0.1.10), libcups2 (>= 1.7.0), libepoxy0 (>= 1.4.3), libfontconfig1 (>= 2.12.6), libfreetype6 (>= 2.2.1), libfribidi0 (>= 0.19.7), libgdk-pixbuf-2.0-0 (>= 2.40.0), libglib2.0-0 (>= 2.59.0), libharfbuzz0b (>= 1.2.6), libjson-glib-1.0-0 (>= 1.5.2), libpango-1.0-0 (>= 1.45.5), libpangocairo-1.0-0 (>= 1.41.0), libpangoft2-1.0-0 (>= 1.41.0), librest-0.7-0 (>= 0.7), libwayland-client0 (>= 1.14.91), libwayland-cursor0 (>= 1.14.91), libwayland-egl1 (>= 1.15.0), libx11-6 (>= 2:1.4.99.1), libxcomposite1 (>= 1:0.4.5), libxcursor1 (>> 1.1.2), libxdamage1 (>= 1:1.1), libxext6, libxfixes3, libxi6 (>= 2:1.2.99.4), libxinerama1, libxkbcommon0 (>= 0.5.0), libxrandr2 (>= 2:1.5.0), libgtk-3-common (>= 3.24.24-4)
Recommends: libgtk-3-bin, librsvg2-common
Suggests: gvfs
Homepage: https://www.gtk.org/
Tag: role::shared-lib
Download-Size: 2,516 kB
APT-Sources: https://deb.debian.org/debian bullseye/main arm64 Packages
Description: GTK graphical user interface library
```
So I have two versions of `libgtk-3-0`.

[... cut ...]
Sorry for wasting your time.

Perhaps there is some code in your installer that says "if there is one of the required libraries then we can continue" which fails where there are two versions of the required library?

Again, sorry for wasting your time.

I'm actually running into another installation problem, so I can't *prove* the above, but that seems to be because my hard-drive is no longer mounted. Probably another NTFS problem. `dietpi-drive_manager` is taking a lifetime to start. No need to worry about it here.

MichaIng, 2021-10-31:
> So I have two versions of `libgtk-3-0`.

As I said above, one is from the Debian repository (the second one) and the other one from the Raspberry Pi repository, so that is all as expected. If a package pulls in `libgtk-3-0`, it will use the newer version, hence the one from the RPi repo.

`vim` isn't related, it doesn't depend on `libgtk-3-0` in any way.

> I have uninstalled vim and `apt-get remove libgtk-3-0=3.24.24-4`, and I'm sure that will fix things.

And this command did actually uninstall something? I'm currently trying to replicate. Works well with the ARMv6 image, currently doing so with the ARMv7 image and afterwards with the ARMv8 image. So far I do not see a reason why the package from the Debian version would be installed and even if, why it doesn't satisfy a regular `libgtk-3-0` dependency (without a version explicitly defined) and why it wouldn't simply upgrade the package, in case. Also the error message doesn't really fit to the fact where this package is installed already, just an older version/one from a different repo. However, let's see what happens on my test systems.

> Perhaps there is some code in your installer that says "if there is one of the required libraries then we can continue" which fails where there are two versions of the required library?

No both is not the case. APT simply selects the newest version, that's all. It is pretty common that multiple versions of a package are available, especially when using 3rd party repos (the RPi repo in this case) and if this causes any issue, then it is a bug in the 3rd party repo.

When `dietpi-drive_manager` takes long to start, it may be that you mounted a drive with it and then simply unplugged it. An automount entry in `/etc/fstab` is created for the drive when you add it, so that it is mounted an future boot sessions as fast as anything accesses the mount point. If you simply unplug the drive and something accesses the mount point (`dietpi-drive_manager` does this to gather mount information) that it hangs for 90 seconds before giving up. It is an issue with this systemd automount implementation, at least for network drives I'd see it as an issue, which can be worked around: #4870
For local drives, you should simply unmount them via `dietpi-drive_manager` if you mounted them with it, before unplugging the drive. For temporary external drives, either mount them manually `mount /dev/sda1 /mnt/external` or use a daemon/package which mounts drives automatically when being attached, e.g. [udisks2](https://packages.debian.org/bullseye/udisks2). I think some desktops ship which such by default, but LXDE does not.

> On starting (via "startx") it immediately showed an error dialog. "No session for pid 30332", the desktop had no icons at all, and the horizontal task bar at the bottom of the screen was slicing off the top of some icons.

Actually I've seen this as well sometimes on the very first login. A simple logout and restart the desktop solves it. Without a reboot, probably some dbus/systemd user session is missing, which triggers the error message. I'll have a look into that. The other issue is that we install the desktop icons and background image on first login of a user, as this cannot be done on a global level before. We then restart PCManFM (which is also used for the desktop icons and context menu on LXDE), and probably either the stop or the restart is done too fast so that it fails partly. I'll also have a look how to make this more reliable.

kerryland, 2021-10-31:
>> I have uninstalled vim and apt-get remove libgtk-3-0=3.24.24-4, and I'm sure that will fix things.
> And this command did actually uninstall something? 

No. I'm deeply ashamed at my bug reporting. I swear I'm normally better than this. My excuse is that I'm also being tripped up by unreliable behaviour of my disks. `dietpi-drive_manager` was hanging for much longer than 90 seconds, and I'm not unplugging the drives. I was blaming NTFS (which might still be true), but I have just discovered that the ext4 disk I created last night disappeared halfway through copying data from the ntfs drive. The physical keyboard is also not responding. Maybe it's a USB problem, although I don't think I'm pushing it hard; USB hub for mouse and keyboard, 1 unpowered external disk and 1 powered external disk. I'll get rid of the USB hub and do some Googling.

MichaIng, 2021-10-31:
Probably a voltage issue. Do you power the external drives via dedicated PSU or are they USB-powered only? The latter is at least for 2,5" drives (and up) known to be unreliable. On RPi, all USB ports share an overall maximum 1.2A (6W), which is easily hit by a single 2,5" drive during peaks, e.g. startup or access peaks. This may then cause various issues.

Generally, SoC voltage issues can be checked on RPi via:
```sh
vcgencmd get_throttled
```
which should return `throttled=0x0` if everything is fine, else there is or was either a thermal related or voltage related throttling or frequency cap active.

kerryland, 2021-11-02:
I finally resolved the problems with the unreliability of my self-powered USB disk. Not voltage or NTFS related, but caused by Seagate using a sub-optimal chip. Repaired by adding "usb-storage.quirks", per [this sticky](https://forums.raspberrypi.com/viewtopic.php?f=28&t=245931)).

So I have been able to take another look. I didn't do anything to attempt to fix the problem, but did try installing Chromium again. It worked perfectly.

` apt show -a  libgtk-3-0` still shows multiple results, so maybe that isn't the underlying problem after all.

MichaIng, 2021-11-02:
Ah, so UAS was the issue. It could be completely disabled/blacklisted as well, but the using the needle USB quirk instead of the hammer is the best solution.

Basically a known issue with Raspberry Pi 4 and some USB drives which do not support UAS. Raspberry Pi 4 uses the dwc2 USB driver by default which does support UAS. Earlier RPi models use dwc_otg by default, which does not support UAS and hence does not suffer from these issues, but in theory lower performance instead.

> `apt show -a libgtk-3-0` still shows multiple results, so maybe that isn't the underlying problem after all.

That is not a problem but totally expected and not a reason for any issues. It can be actually safely tested. Run the following command and see the output, but do not confirm with `y`, instead hit CTRL+C or any other key+return:
```sh
apt install chromium-browser libgtk-3-0
```
If the same error is thrown, could you show the following:
```
apt policy libgtk-3-0
```

kerryland, 2021-11-03:
Just to repeat, I have subsequently installed chromium by simply trying again. That said, the results of your request are below:
```
root@blackbox:~# cat << '_EOF_' > /etc/apt/preferences.d/dietpi-lxde
Package: openbox* obconf* libob* pcmanfm* libfm* libgtk* lx*
Pin: origin archive.raspberrypi.org
Pin-Priority: -1
_EOF_
apt policy libgtk-3-0
apt install chromium-browser libgtk-3-0 # and cancel it again
libgtk-3-0:
  Installed: 3.24.24-4+rpt2
  Candidate: (none)
  Version table:
 *** 3.24.24-4+rpt2 -1
        500 https://archive.raspberrypi.org/debian bullseye/main arm64 Packages
        100 /var/lib/dpkg/status
     3.24.24-4 500
        500 https://deb.debian.org/debian bullseye/main arm64 Packages
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libgtk-3-0 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libgtk-3-0' has no installation candidate
```
Also, `apt install chromium-browser libgtk-3-0` gave:
```

root@blackbox:~# apt install chromium-browser libgtk-3-0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
chromium-browser is already the newest version (92.0.4515.98-rpt2).
Suggested packages:
  gvfs
Recommended packages:
  libgtk-3-bin
The following packages will be upgraded:
  libgtk-3-0 libgtk-3-common
2 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 6,272 kB of archives.
After this operation, 6,144 B of additional disk space will be used.
Get:1 https://archive.raspberrypi.org/debian bullseye/main arm64 libgtk-3-common all 3.24.24-4+rpt2 [3,756 kB]
Get:2 https://archive.raspberrypi.org/debian bullseye/main arm64 libgtk-3-0 arm64 3.24.24-4+rpt2 [2,516 kB]
Fetched 6,272 kB in 6s (1,118 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 58916 files and directories currently installed.)
Preparing to unpack .../libgtk-3-common_3.24.24-4+rpt2_all.deb ...
Unpacking libgtk-3-common (3.24.24-4+rpt2) over (3.24.24-4) ...
Preparing to unpack .../libgtk-3-0_3.24.24-4+rpt2_arm64.deb ...
Unpacking libgtk-3-0:arm64 (3.24.24-4+rpt2) over (3.24.24-4) ...
Setting up libgtk-3-common (3.24.24-4+rpt2) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1) ...
Processing triggers for libglib2.0-0:arm64 (2.66.8-1) ...
Setting up libgtk-3-0:arm64 (3.24.24-4+rpt2) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1) ...
root@blackbox:~#
```

MichaIng, 2021-11-03:
You called the commands in this order, or did you remove `/etc/apt/preferences.d/dietpi-lxde` again? Because `apt policy` shows clearly that the package from the RPi repo is blocked as intended, the one from Debian perfectly available and below it can be seen that it was also installed while it was then upgraded to the one from the RPi repository 🤔.

I'm still confused about the `E: Package 'libgtk-3-0' has no installation candidate` as the candidate is clearly there and even was successfully installed. However, as things are running now and I wasn't able to replicate, let's ignore it for now. Please reopen the issue if you run in any similar issue, like `has no installation candidate` or `you have held packages` and such errors during APT upgrades or installs.

closed 2021-11-03 by MichaIng

Joulinar, 2021-11-03:
I did a test today `dietpi-software install 23 113` on RPI4B Bullseye ARMv8 and all was working fine. Below is the full log.

<details>
  <summary>Logfile attached. Click to expand!</summary>

```
root@DietPi4:~# dietpi-software install 23 113
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated install

[  OK  ] DietPi-Software | Installing LXDE: ultra lightweight desktop
[  OK  ] DietPi-Software | Installing Chromium: web browser for desktop or autostart
[  OK  ] DietPi-Software | Free space check: path=/ | available=12281 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists...
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[ INFO ] DietPi-Software | ALSA will be installed
[ INFO ] DietPi-Software | X.Org X Server will be installed

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing ALSA: Advanced Linux Sound Architecture

[ SUB1 ] DietPi-Set_hardware > soundcard (rpi-bcm2835-auto)
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: alsa-utils
[ INFO ] DietPi-Set_hardware | Flagged for install: alsa-utils
[ INFO ] DietPi-Set_hardware | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists...
[  OK  ] DietPi-Set_hardware | APT update
[ INFO ] DietPi-Set_hardware | APT install for: alsa-utils, please wait...
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package libasound2-data.
(Reading database ... 17607 files and directories currently installed.)
Preparing to unpack .../0-libasound2-data_1.2.4-1.1+rpt1_all.deb ...
Unpacking libasound2-data (1.2.4-1.1+rpt1) ...
Selecting previously unselected package libasound2:arm64.
Preparing to unpack .../1-libasound2_1.2.4-1.1+rpt1_arm64.deb ...
Unpacking libasound2:arm64 (1.2.4-1.1+rpt1) ...
Selecting previously unselected package libatopology2:arm64.
Preparing to unpack .../2-libatopology2_1.2.4-1.1+rpt1_arm64.deb ...
[... 1141 lines cut ...]
[  OK  ] DietPi-Set_hardware | sed -i /^[[:blank:]]*hdmi_ignore_hotplug=/c\#hdmi_ignore_hotplug=0 /boot/config.txt
[  OK  ] DietPi-Set_hardware | sed -i /^[[:blank:]]*enable_tvout=/c\#enable_tvout=0 /boot/config.txt
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_HEADLESS=0
[  OK  ] headless disable | Completed
[ SUB1 ] DietPi-Set_hardware > gpumemsplit (96)
[  OK  ] DietPi-Set_hardware | Setting in /boot/config.txt adjusted: gpu_mem_1024=96
[  OK  ] DietPi-Set_hardware | Setting in /boot/config.txt adjusted: gpu_mem_512=96
[  OK  ] DietPi-Set_hardware | Setting in /boot/config.txt adjusted: gpu_mem_256=96
[  OK  ] DietPi-Set_hardware | sed -i /dtoverlay=dietpi-disable_vcsm/d /boot/config.txt
[  OK  ] DietPi-Set_hardware | rm /boot/overlays/dietpi-disable_vcsm.dtbo
[  OK  ] DietPi-Set_hardware | rm /etc/modprobe.d/dietpi-disable_vcsm.conf
[  OK  ] gpumemsplit 96 | Completed

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Install completed

[  OK  ] DietPi-Survey | Purging survey data
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : cron
root@DietPi4:~#
```
</details>

`libgtk-3-0` package was already pulled as part of the LXDE installation as you can see on my log.

MichaIng, 2021-11-03:
Jep, that matches my test results, thanks for verifying. I was thinking whether installing Chromium first and LXDE afterwards might cause issues, as then `libgtk-3-0` is installed form the RPi repo and blocked afterwards. However, as/until the version is newer, any `apt install libgtk-3-0` or dependency should then simply keep the current package installed without complaining.

Joulinar, 2021-11-03:
Even installing them individually Chromium first, followed by LXDE is working without issues.

<details>
  <summary>Chromium Logfile attached. Click to expand!</summary>

```
root@DietPi4:~# dietpi-software install 113
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated install

[  OK  ] DietPi-Software | Installing Chromium: web browser for desktop or autostart
[  OK  ] DietPi-Software | Free space check: path=/ | available=12281 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists...
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[ INFO ] DietPi-Software | ALSA will be installed
[ INFO ] DietPi-Software | X.Org X Server will be installed

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing ALSA: Advanced Linux Sound Architecture

[ SUB1 ] DietPi-Set_hardware > soundcard (rpi-bcm2835-auto)
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: alsa-utils
[ INFO ] DietPi-Set_hardware | Flagged for install: alsa-utils
[ INFO ] DietPi-Set_hardware | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists...
[  OK  ] DietPi-Set_hardware | APT update
[ INFO ] DietPi-Set_hardware | APT install for: alsa-utils, please wait...
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package libasound2-data.
(Reading database ... 17607 files and directories currently installed.)
Preparing to unpack .../0-libasound2-data_1.2.4-1.1+rpt1_all.deb ...
Unpacking libasound2-data (1.2.4-1.1+rpt1) ...
Selecting previously unselected package libasound2:arm64.
Preparing to unpack .../1-libasound2_1.2.4-1.1+rpt1_arm64.deb ...
Unpacking libasound2:arm64 (1.2.4-1.1+rpt1) ...
Selecting previously unselected package libatopology2:arm64.
Preparing to unpack .../2-libatopology2_1.2.4-1.1+rpt1_arm64.deb ...
Unpacking libatopology2:arm64 (1.2.4-1.1+rpt1) ...
[... 1191 lines cut ...]
[  OK  ] DietPi-Software | systemctl daemon-reload
[  OK  ] DietPi-Software | systemctl unmask dbus
[  OK  ] DietPi-Software | systemctl start dbus
[  OK  ] DietPi-Software | systemctl unmask systemd-logind
[  OK  ] DietPi-Software | systemctl start systemd-logind
[ SUB1 ] DietPi-Services > dietpi_controlled
[  OK  ] DietPi-Services | dietpi_controlled : cron
[ SUB1 ] DietPi-Set_hardware > headless (disable)
[  OK  ] DietPi-Set_hardware | sed -i /^[[:blank:]]*max_framebuffers=/c\#max_framebuffers=2 /boot/config.txt
[  OK  ] DietPi-Set_hardware | sed -i /^[[:blank:]]*hdmi_blanking=/c\#hdmi_blanking=0 /boot/config.txt
[  OK  ] DietPi-Set_hardware | sed -i /^[[:blank:]]*hdmi_ignore_hotplug=/c\#hdmi_ignore_hotplug=0 /boot/config.txt
[  OK  ] DietPi-Set_hardware | sed -i /^[[:blank:]]*enable_tvout=/c\#enable_tvout=0 /boot/config.txt
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_HEADLESS=0
[  OK  ] headless disable | Completed

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Install completed

[  OK  ] DietPi-Survey | Purging survey data
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : cron
root@DietPi4:~#
```
</details>