MichaIng/DietPi #4916: Could not find DRM device! Qt v5.15.2 Application

csharpee, 2021-10-31:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
- 
- ```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applied'
```
- Distro version | buster 0
- Kernel version | Linux oDinPi 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
- SBC model | RPi 400 (aarch64)

#### Additional Information (if applicable)
- Software title | Debian Direct Rendering Manager (DRM)
- Was the software title installed freshly or updated/migrated? Fresh 

#### Steps to reproduce
<!-- Explain how to reproduce the issue -->
1. ... Compile Qt application on RPI 400
2. ... Launch fresh built binary

#### Expected behaviour
- ...Application runs without DRM error

#### Actual behaviour
<!-- What IS happening? -->
```
- Could not find DRM device!
Aborted
```

#### Extra details
<!-- Please post any extra details that might help solve the issue -->
- I configured my Qt application, make the application and make install the Qt application.

I am wondering specifically if the DRM error is associated with this Debian in particular? I found this thread from 2020 and wondering if it applies to debian diet-pi?

MichaIng, 2021-10-31:
Many thanks for your report.

Are the DRM devices really not present?
```sh
ls -l /dev/dri
```
If so, is the KMS of fMKS driver enabled?
```sh
grep 'vc4-.*kms-v3d' /boot/config.txt
```
If not, please enable it. This can be done via `dietpi-config` > "Display Options" > "Resolution" (reminds me that we need to move it out there...), or by manually adding:
```
dtoverlay=vc4-fkms-v3d
```
for the fake KMS driver, or
```
dtoverlay=vc4-kms-v3d-pi4
```
for the full KMS driver to `/boot/config.txt`.

Hmm, I wanted to add an offical RPi docs link for this, but sadly the new documentation has it scattered and no section which explains the three graphics drivers explicitly 😢: https://www.raspberrypi.com/documentation/

csharpee, 2021-10-31:
Awesome :birthday: :+1: . Thanks for your speedy response. 

`ls -l /dev/dri`
ls: cannot access '/dev/dri': No such file or directory

`grep 'vc4-.*kms-v3d' /boot/config.txt`
`a cricket playing his fiddle at night`

On going to dietpi-config >> display >> resolution, I see I didn't have any driver loaded for some odd reason. It said Current resolution: 0.

I am testing the **vc4-kms-v3d-pi4** now; rebooting...

MichaIng, 2021-10-31:
Not uncommon. By default no KMS driver is enabled on DietPi, hence the legacy framebuffer based driver is used for GUI. But some applications obviously require DRM, hence a KMS driver enabled.

Resolution `0` looks odd, probably this is shown when none has been explicitly selected. We should then show `auto`, as this is what is effectively done, using the native screen resolution minus overscan, which is usually perfectly fine. When starting any GUI application via DRM/KMS, this framebuffer resolution isn't effective anyway, but only for the bare console.

I guess you don't need to rebuild the app. If the required development headers were installed, it compiles regardless whether the DRM/KMS is enabled or not. But if it still does not work, then a rebuild is definitely worth an attempt.

csharpee, 2021-10-31:
> I guess you don't need to rebuild the app. If the required development headers were installed, it compiles regardless whether the DRM/KMS is enabled or not. But if it still does not work, then a rebuild is definitely worth an attempt.

I tested the original bin file `./app` and didn't have the DRM error. Although, sudo timer tricked me on Halloween and built the binaries with sudo, therfore, I was receiving permission for HDMI. 

```
Failed to move cursor on screen HDMI1: -13
Could not queue DRM page flip on screen HDMI1 (Permission denied)
Could not set DRM mode for screen HDMI1 (Permission denied)
```

I need to `sudo visudo` 

`Defaults env_reset, timestamp_timeout=00`

csharpee, 2021-10-31:
> Not uncommon. By default no KMS driver is enabled on DietPi, hence the legacy framebuffer based driver is used for GUI. But some applications obviously require DRM, hence a KMS driver enabled.

According to another KMS is required for RPI 4. 

"Note: For a Qt 5 that will be run on a Raspberry Pi 4 you must use KMS, Broadcom EGL is not supported on that model and never will. Make sure you have installed the packages: libgles2-mesa-dev libgbm-dev"

MichaIng, 2021-10-31:
This should be true for all RPi models, good that it is clear then.

Jep, initial Vulkan support will be part of Mesa 21. Bullseye however ships only Mesa 20. Mesa 21 has been merged to Bookworm already: https://packages.debian.org/search?keywords=libegl1-mesa
With some luck, it will be either available via backports for Bullseye, and/or the RPi repo maintainers add a build to the RPi repository: https://archive.raspberrypi.org/debian/pool/main/m/mesa/?C=M;O=D

csharpee, 2021-11-01:
Ok, after rebuilding and everything else, all I needed to do is use -platform xcb (X11). 

`./app -platform xcb`

I am super new to this whole process, including Qt. It is a super complex piece of software. 

At first, without adding platform specific options, it was providing lots of errors. I shouldn't have to add `-platform xcb` when running the binary...then again, I haven't created a app.deb for it.

MichaIng, 2021-11-01:
Okay great. I'm also no expert in this, but XCB is a generally optional platform plugin/X11 abstraction, through on Linux seems to be required now: https://doc.qt.io/qt-5/linux-requirements.html
Probably there is a build option to force XCB, respectively disable direct/low-level alternatives in the first place.

oniongarlic, 2021-11-02:
> Ok, after rebuilding and everything else, all I needed to do is use -platform xcb (X11) after enabling [#4916 (comment)](https://github.com/MichaIng/DietPi/issues/4916#issuecomment-955786469)
> 
> `./app -platform xcb`
> 

My Qt instructions are using "-qpa eglfs" to set the default platform to eglfs as I prefer to run without X11. Just change (or remove) that option to change the default. I haven't yet tested Vulkan on Qt 5 but on 6 it works nicely.

csharpee, 2021-11-02:
@oniongarlic I thought my RPi  4 made use of eglfs as default. I didn't know how to verify or check for this.

I am still unsure why I had to use X11 (xcb) on my RPi 400. 

I updated my configuration file with: 

-bundled-xcb-xinput
-qpa xcb

I'm compiling now.; testing later. 

P.S> @oniongarlic  Do I need to **make** and make install every configuration change? it currently takes about 8-10hrs on my RPi 400.
I would like to build for v6 & v7 ARMs.

csharpee, 2021-11-02:
My compiled C++ code with Qt5.1.5.2 compiled and I didn't need to add ./app -platform xcb to run the application.

MichaIng, 2021-11-02:
`make -j3` and `make -j 3` should be 100% equivalent 🤔. Did you have a look into `htop` or so during compilation to see whether 3 build processes are running or not? However, if `-j 3` works for you, no need to debug 😄.

MichaIng, 2021-11-11:
I'll mark this issue as closed. Feel free to reopen if questions remain.

closed 2021-11-11 by MichaIng