MichaIng/DietPi #4919: Dietpi-update fails on debian testing repos

tandy-1000, 2021-11-02:
### Creating a bug report/issue
I am using debian testing repos, and I can't seem to run `dietpi-update`, I tried exporting new `G_DISTRO_NAME` and `G_RASPBIAN` based on values in `/DietPi/dietpi/func/dietpi-obtain_hw_model` but this didn't work.

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux bookworm/sid
Release:	testing
Codename:	bookworm
```

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=2
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
Unknown
- Kernel version | `uname -a`
Linux server 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
rockpro64
- Power supply used | (EG: 5V 1A RAVpower)
pine64 provided
- SDcard used | (EG: SanDisk ultra)
Emmc

#### Steps to reproduce
1. Run `sudo dietpi-update`

#### Expected behaviour
- Update should happen successfully

#### Actual behaviour
```
[ INFO ] DietPi-Update | Your Debian version is too low to update to the current branch: Unknown (0)
[ INFO ] DietPi-Update | We're switching to an alternative branch: jessie-support
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/jessie-support/.update/version
curl: (22) The requested URL returned error: 404 
[FAILED] DietPi-Update | Unable to get latest version.
[ INFO ] DietPi-Update | Please check the target Git repository and your network connection, then rerun dietpi-update.
[ INFO ] DietPi-Update | If this error persists, please report at: https://github.com/MichaIng/DietPi/issues
```

Joulinar, 2021-11-02:
in general `bookworm` is already supported. Could you check output of following

`cat /etc/debian_version`

tandy-1000, 2021-11-02:
```
cat /etc/debian_version
bookworm/sid
```
I also tried changing my sources.list to bookworm instead of testing

Joulinar, 2021-11-02:
ahh `bookworm` support has been added with 7.6 and you are on 7.5 (still). That's where the issue is coming from. I guess you need to update `DietPi-Obtain_HW_model` manually first.

https://github.com/MichaIng/DietPi/pull/4698

tandy-1000, 2021-11-02:
This worked, thank you for your help / time.

closed 2021-11-02 by tandy-1000