MichaIng/DietPi #4921: Deluge web UI Preferences are not saved when I reboot the raspberry

martinlaune, 2021-11-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applied'
- Distro version | buster 1
- Kernel version | Linux DietPi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux
- SBC model | RPi 3 Model B (armv7l)

#### Additional Information (if applicable)
- Software title | Deluge
- Was the software title installed freshly or updated/migrated? Yes
- Can this issue be replicated on a fresh installation of DietPi? I did not try
- Bug report ID | ccb6a4c8-1e9a-4b97-afbf-0100e4cc046f

#### Steps to reproduce

1. ... install Deluge through dietpi-software
2. ... open deluge web UI 1.3.15 and change whatever preference, for exemple the password
3.  Reboot the raspberry
4. Open again deluge web UI and you will see that the UI preferences are not saved

Deluge web UI Preferences are not saved when I reboot the raspberry

#### Expected behaviour

- the UI preferences should be saved by reboot the raspberry 

#### Actual behaviour

- ...Deluge web UI Preferences are not saved when I reboot the raspberry

#### Extra details

- ...

MichaIng, 2021-11-02:
Many thanks for your report.

I tried to replicate, but it works well here:
- Installed Deluge
- Accessed to web UI
- Disabled UPnP, changed max connections and the password
- Rebooted
- Verified that the settings are still changed in the web UI
- Also verified that Deluge created a new config file (with the changes) after doing a backup

Can you check:
```sh
ls -l /mnt/dietpi_userdata/deluge/.config/deluge/
```

martinlaune, 2021-11-02:
root@DietPi:~# ls -l /mnt/dietpi_userdata/deluge/.config/deluge/
total 48
-rw------- 1 debian-deluged root     76 nov.   1 21:03 auth
-rw-r--r-- 1 debian-deluged root   2040 nov.   1 21:03 core.conf
-rw-rw-r-- 1 debian-deluged dietpi   10 nov.   2 21:00 deluged.pid
-rw-rw-r-- 1 debian-deluged dietpi 2603 nov.   2 20:59 dht.state
-rw-rw-r-- 1 debian-deluged dietpi   51 nov.   2 20:55 execute.conf
-rw-r----- 1 debian-deluged dietpi  168 nov.   1 21:03 hostlist.conf.1.2
drwxr-x--- 2 debian-deluged dietpi 4096 nov.   1 21:03 icons
drwxrwxr-x 2 debian-deluged dietpi 4096 nov.   2 09:40 plugins
-rw-rw-r-- 1 debian-deluged dietpi 3419 nov.   2 20:59 session.state
drwxrwxr-x 2 debian-deluged dietpi 4096 nov.   1 21:03 ssl
drwxrwxr-x 2 debian-deluged dietpi 4096 nov.   2 23:37 state
-rw-r--r-- 1 debian-deluged root    527 nov.   1 21:03 web.conf

MichaIng, 2021-11-02:
Looks fine, though the `core.conf` timestamp and missing backup file indicates that indeed nothing was written. When you change the password, does it say "Password successfully changed" or something like that? And you hit "Apply" after changing other settings, right? Please try these again and then check the service logs:
```sh
journalctl -u deluged -u deluge-web
```

martinlaune, 2021-11-02:
[deluge_debug.log](https://github.com/MichaIng/DietPi/files/7464440/deluge_debug.log)

MichaIng, 2021-11-02:
Did you apply the live patch? Looks like the config file syntax errors are there. Try:
```sh
systemctl stop deluge-web deluged
```
Then in `/mnt/dietpi_userdata/deluge/.config/deluge/core.conf` at the "max_active_limit" setting there is a single quote character `'` behind the number which does not belong there. Remove it.
And in `/mnt/dietpi_userdata/deluge/.config/deluge/web.conf` at the `pwd_sha1` line, the hash needs to be wrapped in to double quotes, like here: https://github.com/MichaIng/DietPi/blob/dev/.conf/dps_45/deluge_web.conf#L14
After doing so, restart the services:
```sh
systemctl start deluged deluge-web
```

martinlaune, 2021-11-03:
problem solved thank you

closed 2021-11-03 by martinlaune

Joulinar, 2021-11-03:
@martinlaune did you solved the issue with the help of the commands @MichaIng provided in https://github.com/MichaIng/DietPi/issues/4921#issuecomment-958397570