MichaIng/DietPi #4923: Wifi modules disabled on fresh install

lupa18, 2021-11-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | v7.7
- Distro version | bullseye
- Kernel version | Linux rock64 5.10.63-rockchip64 aarch64 GNU/Linux
- SBC model | ROCK64 (aarch64)
- Power supply used | 5V 1A
- SDcard used | SanDisk Ultra

#### Additional Information (if applicable)
- Software title | Dietpi
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Create the SD card
2. Then I modify dietpi-wifi.txt to fit my wifi configuration
3. Finally couldn't find dietpi on my network. 

#### Expected behaviour

- I must be able to reach dietpi using ssh

#### Actual behaviour

- No wifi network available

#### Extra details

- What I finally did: plug a monitor, login, run dietpi-config and enable wifi modules
- Maybe is related with the fact that rock64 didn't have a builtin wifi card.

MichaIng, 2021-11-03:
Many thanks for your report.

Did you enable WiFi in [dietpi.txt](https://github.com/MichaIng/DietPi/blob/a4cf2c5/dietpi.txt#L28)?

Whether onbaord WiFi or not shouldn't play a role. WiFi modules are enabled by default just like when you do that via `dietpi-config`, they are only disabled when WiFi is not enabled via `dietpi.txt`.

lupa18, 2021-11-03:
Oh, sorry. It seems I didn't enabled there. I only put the SSID and pass. 

Thanks!

closed 2021-11-03 by lupa18