MichaIng/DietPi #4924: Home assistant installation failed

lupa18, 2021-11-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.7
- Distro version | bullseye
- Kernel version | Linux DietPi 5.10.63-rockchip64
- SBC model | ROCK64 (aarch64)
- Power supply used | 5V 1A
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | Home Assistant
- Was the software title installed freshly or updated/migrated? Both
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID |10ff456b-4fea-4f92-ac51-ed25c093725d

#### Steps to reproduce

1. I updated to bullseye twice, one manual and one fresh. In both cases the same
2. When I tried to update or fresh install HA it crashes

#### Expected behaviour

- HA should be installed without problem

#### Actual behaviour

- Installation crashes

#### Extra details

- Error includes: 
```
Objects/bytesobject.c:1261:1: internal compiler error: Segmentation fault
 1261 | PyBytes_Repr(PyObject *obj, int smartquotes)
      | ^~~~~~~~~~~~
```
(I have already sent a bug report)

MichaIng, 2021-11-04:
I tried to replicate, Bullseye on ARMv8, but it works fine here.

What is the size of your tmp directory?
```
df /tmp
```

Joulinar, 2021-11-04:
`Segmentation fault` usually points to some corruption or incompatibility. Isn't it?

lupa18, 2021-11-04:
> I tried to replicate, Bullseye on ARMv8, but it works fine here.
> 
> What is the size of your tmp directory?
> 
> ```
> df /tmp
> ```

Ho and thanks! df -h /tmp -> 2GB

lupa18, 2021-11-04:
I had also discovered this message: 
`[    9.603313] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_component_of_xlate_dai_id on hdmi-audio-codec.4.auto: -22`

MichaIng, 2021-11-04:
The HDMI audio error is not related.

/tmp size is sufficient. Yes indeed a segmentation fault is usually an issue with the binary itself, in this case hence probably with the C compiler. Can you run it from console?
```sh
gcc --help
```

lupa18, 2021-11-04:
> Can you run it from console?
> 
> ```shell
> gcc --help
> ```

yes ! and output is gcc's help

MichaIng, 2021-11-04:
Strange, however seems to be a systematical error as it failed the same way at least on two system (upgraded + fresh image) in your case. I did the successful test install in an arm64 container, so probably not 100% comparable.

I'll have a look into your bug report and will try to replicate on a real SBC with Armbian based DietPi. I have no ROCK64, but a NanoPi R4S at least, which shares the same kernel.

MichaIng, 2021-11-11:
Worked fine here as well on NanoPi R4S. Though I have to admin while fixing #4928 I ran into 2 failed Python compilations as well, though a rebuild then worked. I though it was either due to insufficient memory or as the RPi host became unstable, e.g. caused a some messed lines in the source code of Python (HA install ran in container), just a few very mixed up lines in the middle of the file, no idea how this can happen. Later a backup produced kernel errors on drive access, and finally I needed to raise minimum voltage a little to get it back stable.

So I don't think it is a general issue. Did you try to simply rerun the command or the installation, e.g. after a reboot?

lupa18, 2021-11-23:
I did a fresh install and problem persists. Attached full error log
[python-build.20211123142824.7856.log](https://github.com/MichaIng/DietPi/files/7588833/python-build.20211123142824.7856.log)
.

MichaIng, 2021-11-23:
When you retry the build from the error prompt, is the final error exactly the same or does it fail at a different module?

lupa18, 2021-11-23:
I installed python3 from dietpi-software and other software. Then installed HA and now it worked. It's weird.

MichaIng, 2021-11-23:
A system wide Python is not required, as one is compiled for the `pyenv` HA run in.

I guess it was a similar issue I ran into with the container, where for some reason something with the download or build of Python failed at a random point. As said, in my case a source code file was malformed, but no idea how this can happen, or whether some failsafe/integrity check should be implemented in `pyenv` to avoid such issues. Makes sense to keep this in mind when doing more tests and facing or receiving similar reports.

To check whether it is due to malformed source files, the file the build fails at could be check, in your case `./Modules/clinic/_codecsmodule.c.h` line 363 within the `/tmp/python-build.*` build directory. If indeed there is malformed code, then we should report this to the `pyenv` team to ask for an integrity check and/or review the Python download/extract code: https://github.com/pyenv/pyenv/issues

MichaIng, 2022-01-04:
I'll mark this as closed. Feel free to reopen when required.

closed 2022-01-04 by MichaIng