MichaIng/DietPi #4925: Tor Relay wont work

cptechnik, 2021-11-04:
**1st ... missing geoip package**
""Nov 04 03:21:45.000 [warn] Failed to open GEOIP file /usr/share/tor/geoip.  We've been configured to see which countries can access us as a bridge, and we need GEOIP information to tell which countries clients are in. Do you have the tor-geoipdb package installed?
Nov 04 03:21:45.000 [warn] Failed to open GEOIP file /usr/share/tor/geoip6.  We've been configured to see which countries can access us as a bridge, and we need GEOIP information to tell which countries clients are in. Do you have the tor-geoipdb package installed?
Nov 04 03:21:45.000 [notice] Configured to measure directory request statistics, but no GeoIP database found. Please specify a GeoIP database using the GeoIPFile option.
""
>> apt install tor-geoipdb

...accepted

---

**2nd** ... ""At least one protocol listed as required in the consensus is not supported by this version of Tor. You should upgrade. This version of Tor will not work as a relay on the Tor network. The missing protocols are: DirCache=2 HSDir=2 HSIntro=4 Link=4-5 LinkAuth=3""

 ...>>> so...

**_please remove or sign it with "beta"_**

---

### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
stretch 1

- Kernel version | `uname -a`
Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
echo $G_HW_MODEL_NAME

- Power supply used | (EG: 5V 1A RAVpower)
äh, yes

- SDcard used | (EG: SanDisk ultra)
also

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Tor Relay

- Was the software title installed freshly or updated/migrated?
fresh

chhrrrrrrr......chhrrrrrrr......chhrrrrrrr......chhrrrrrrr......

- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2021-11-04:
Many thanks for your report.

You use a very old image based on Debian Stretch and it seems the Tor version shipped by the Stretch repo is too old now for all features network-side: https://packages.debian.org/stretch/tor
"Solution" our end is to disable Tor on Stretch systems. On non-Raspbian systems backports can be used, but those are not available through the Raspbian repository (your system).

Also, we plan to drop support for Debian Stretch at the end of the year due to raising problems with it. I highly recommend to upgrade to Debian Buster, and in case further to the current stable Debian Bullseye, to solve an increasing number of issues you'll run into: https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

About the GeoIP database, I think it is fine to leave it up to the user whether location info is wanted or not. We do not actively enable it in any config, but probably it needs to be actively disabled to mute this error. I'll verify this on Buster and Bullseye and in case add the required setting to the default configs on install.

cptechnik, 2021-11-09:
...in my case, the easiest way to spend time how to upgrade... 
thanks!

MichaIng, 2021-11-09:
Okay, indeed I needed to disable the software option for Debian Stretch systems: https://github.com/MichaIng/DietPi/commit/4470dae04df75d038259ea21a770bf2bbc82a227
In theory we could keep it enabled on non-Raspbian based systems, using backports, but the effort to implement more and more workarounds for Stretch is not worth it, unless we get a large number of requests.

Also there is another warning about an OpenSSL header mismatch, and there are no backports for OpenSSL/libSSL. Also the GeoIP warning is present on Stretch only. It looks like `GeoIPFile` and `GeoIPv6File` settings had some hardcoded default (not visible in `/usr/share/tor/tor-service-defaults-torrc`) on the Tor version/build, which is not the case with the versions shipped on Buster and Bullseye.

MichaIng, 2021-11-09:
Marking this as closed. Feel free to reopen if required.

closed 2021-11-09 by MichaIng