MichaIng/DietPi #4926: Mosquitto installation fails

Bluehmchen, 2021-11-04:
#### Details:
- Date           | Thu Nov  4 09:14:35 GMT 2021
- DietPi version | v7.7.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -qq --allow-change-held-packages install mosquitto`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. dietpi-software install 123
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```

Joulinar, 2021-11-04:
Hi,

could you share full output of `apt update && apt upgrade`

Bluehmchen, 2021-11-04:
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

root@DietPi:~# apt update && apt upgrade
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 https://repo.mosquitto.org/debian bullseye InRelease
Hit:3 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPi:~#

Bluehmchen, 2021-11-04:
It is a new installation of DiePi and the first attempt to install software

Joulinar, 2021-11-04:
could you try following and post full output

```
apt install mosquitto
```

Bluehmchen, 2021-11-04:
root@DietPi:~# apt install mosquitto
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mosquitto : Depends: libwebsockets18 (>= 4.2.1) but it is not installable
E: Unable to correct problems, you have held broken packages.
root@DietPi:~#

Joulinar, 2021-11-04:
ok I found another issue reporting same https://github.com/MichaIng/DietPi/issues/4757

It's an issue with `mosquitto` repository. I described a workaround on this post https://github.com/MichaIng/DietPi/issues/4757#issuecomment-923286197

Bluehmchen, 2021-11-04:
It works. Thank you

Joulinar, 2021-11-04:
ok I will close the issue for now. Feel free to reopen if needed.

closed 2021-11-04 by Joulinar

MichaIng, 2021-11-04:
I gave this a bump at Mosquitto, I hope they'll resolve it soon, otherwise we'll add the workaround to our code.