MichaIng/DietPi #4928: DietPi-Software | Home Assistant: Python 3.9 piwheels Buster/Stretch mismatch

MichaIng, 2021-11-04:


seniorm0ment, 2021-11-04:
I'm having issues with Home Assistant installation. After install, systemctl status shows it's using sqlite 3.27.2 and it wants at minimim 3.32.1.
I can't finish setting this up

Joulinar, 2021-11-04:
Pls open an own issue and fill the bug report with your system details. Thx for understanding.

MichaIng, 2021-11-04:
No need, I'm pretty sure it is exactly the issue I'm addressing here.

Joulinar, 2021-11-04:
#### Required 
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

seniorm0ment, 2021-11-04:
**Specs**
```
# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

# echo $G_DISTRO_NAME $G_RASPBIAN
buster 0

# uname -a
Linux 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux

# echo $G_HW_MODEL_NAME
RPi 4 Model B (aarch64)
```
**PSU**
I don't know.

**SD Card**
Sandisk Ultra

**Error**

```
# systemctl status home-assistant
● home-assistant.service - Home Assistant (DietPi)
   Loaded: loaded (/etc/systemd/system/home-assistant.service; disabled; v
endor preset: enabled)
   Active: active (running) since Thu 2021-11-04 16:36:37 EDT;
11min ago
 Main PID: 6159 (hass)
    Tasks: 9 (limit: 2198)
   CGroup: /system.slice/home-assistant.service
           └─6159 /home/homeassistant/.pyenv/versions/3.9.7/bin/python3.9
/home/homeassistant/.pyenv/versions/3.9.7/bin/hass -c /mnt/dietpi_userdata
/homeassistant

Nov 04 16:36:37 systemd[1]: Started Home Assistant (DietPi).
Nov 04 16:36:41 homeassistant[6159]: 2021-11-04 16:36:41 WARNING (R
ecorder) [homeassistant.components.recorder.util] The system could not val
idate that the sqlite3 database at //mnt/dietpi_userdata/homeassistant/hom
e-assistant_v2.db was shutdown cleanly
Nov 04 16:36:41 homeassistant[6159]: 2021-11-04 16:36:41 WARNING (R
ecorder) [homeassistant.components.recorder.util] Version 3.27.2 of SQLite
 is not supported; minimum supported version is 3.32.1. Starting with Home
 Assistant 2022.2 this will prevent the recorder from starting. Please upg
rade your database software before then
```

MichaIng, 2021-11-04:
Fits very well. If you can wait until tomorrow, I'll ship a workaround then. The alternative would be to upgrade the system to Debian Bullseye: https://dietpi.com/blog/?p=811

seniorm0ment, 2021-11-04:
Can you confirm whether or not Mopidy is working yet on Bullseye?
If so, then yeah I'll try to run the manual upgrade and hope it goes smoothly.
If not, then I'll wait tomorrow for the workaround.

MichaIng, 2021-11-04:
Jep, Mopidy it works pretty well on Bullseye.

seniorm0ment, 2021-11-04:
Weird.
When I ran apt-update it complained about Mopidy.
I noticed on the blog (although 2 months old) it states Mopidy didn't have a Bullseye release yet.

MichaIng, 2021-11-04:
As mentioned in the block, the Mopidy repo does not have a Bullseye suite yet, but the Buster suite works pretty fine. In `/etc/apt/sources.list.d/mopidy.list` exchange "bullseye" again with "buster".

seniorm0ment, 2021-11-04:
Looks like that worked to upgrade and fix Mopidy!
Trying to reinstall Home Assistant now, and will report back. 
Thanks.

MichaIng, 2021-11-04:
> Will I need to change mopidy.list back to Bullseye when it has a Bullseye suite? Or will it do it automatically?

We will adjust this on a future update automatically, once the suite is available.

seniorm0ment, 2021-11-04:
```
systemctl status home-assistant
assistant.service - Home Assistant (DietPi) loaded (/etc/systemd/system/home-assistant.service; disabled; vendor preset: enabled)                                                                                                                                                                                                                                       Nov 04 17:54:00 aurras systemd[1]: Started Home Assistant (DietPi).                                                                                                 Nov 04 17:54:05 aurras homeassistant[18063]: 2021-11-04 17:54:05 WARNING (Recorder) [homeassistant.components.recorder.util] The system could not validate that the sqlite3 database at //mnt/dietpi_userdata/homeassistant/home-assistant_v2.db was shutdown cleanly  
```

I can't reach the homeassistant webgui at http://ip:8123/
Not sure if tis is an actual error or not.

MichaIng, 2021-11-04:
Some pre-build Python 3.9 wheels provided by piwheels.org depend on external libraries of Debian Bullseye. There seems to be no check/field preventing the installation of those on Stretch/Buster systems, where older library versions are used. So we need to block `pip` from pulling those from piwheels.org to have them instead compiled against the actually present libraries.

Another solution is to upgrade the system to Debian Bullseye: https://dietpi.com/blog/?p=811

seniorm0ment, 2021-11-04:
> Another solution is to upgrade the system to Debian Bullseye: https://dietpi.com/blog/?p=811

That is what I did I thought?
I did

apt update
apt install --reinstall librtmp1

sed -i 's/buster/bullseye/g' /etc/apt/sources.list{,.d/*.list}
sed -i 's|bullseye/updates|bullseye-security|' /etc/apt/sources.list
[[ -f '/etc/apt/sources.list.d/raspi.list' ]] && sed -i 's/ ui$//' /etc/apt/sources.list.d/raspi.list
rm -f /etc/apt/sources.list.d/dietpi-{php,wireguard}.list
rm -f /etc/apt/trusted.gpg.d/dietpi-php.gpg
rm -f /etc/apt/preferences.d/dietpi-{php,openssl,xrdp,wireguard}
rm -f /etc/apt/sources.list.d/pivpn-bullseye.list
rm -f /etc/apt/sources.list.d/pivpn-bullseye-repo.list
rm -f /etc/apt/preferences.d/pivpn-limit-bullseye

apt update
dietpi-services stop
apt upgrade
apt full-upgrade

cat /etc/os-release                                                                                                                                  PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"                                                                                                                        NAME="Debian GNU/Linux"                                                                                                                                             VERSION_ID="11"                                                                                                                                                     VERSION="11 (bullseye)"                                                                                                                                             VERSION_CODENAME=bullseye                                                                                                                                           ID=debian                                                                                                                                                           HOME_URL="https://www.debian.org/"                                                                                                                                  SUPPORT_URL="https://www.debian.org/support"                                                                                                                        BUG_REPORT_URL="https://bugs.debian.org/"

MichaIng, 2021-11-04:
Another solution would have been to install either Python 3.8 or Python 3.10, but:
- Python 3.8 support has been dropped by HA: https://github.com/home-assistant/core/commit/b024d88
- Python 3.10 still causes some issues: https://github.com/home-assistant/core/issues/59006
_______
@seniorm0ment 
After upgrading to Bullseye, you can reinstall HA which should then succeed:
```sh
dietpi-software reinstall 157
```

seniorm0ment, 2021-11-04:
I had done an uninstall and install through dietpi-software
didn't try reinstall like that, let me try.

MichaIng, 2021-11-04:
> I had done an uninstall and install through dietpi-software

This should work the same way then. And it does still not work?

seniorm0ment, 2021-11-04:
> 
> 
> > I had done an uninstall and install through dietpi-software
> 
> This should work the same way. And it does still not work?

It wasn't throwing the original sqlite error about being an outdated version.
But just was throwing: [homeassistant.components.recorder.util] The system could not validate that the sqlite3 database at //mnt/dietpi_userdata/homeassistant/home-assistant_v2.db was shutdown cleanly

And I couldn't reach the webgui. Unless there was some other command I needed to run. It said the service was active.
I looked at ss and didn't see it trying to do anything over port 8123. I'm more used to netstat so maybe I was reading ss wrong though.

MichaIng, 2021-11-04:
Please check `htop` whether it is still active doing something (one or more core close to 100%). Probably it is still compiling some Python modules.

seniorm0ment, 2021-11-04:
Nope pretty much 0% cpu

MichaIng, 2021-11-04:
After CPU usage turned to 0% and only one HA process shown in `htop`, try a service restart:
```sh
systemctl restart home-assistant
```
This is usually required once, still not sure why: https://github.com/home-assistant/core/issues/28361

Yes the database unclean shutdown warning shouldn't break anything. Of course it is never good when database connection was lost uncleanly, but you'd see a real error if it was corrupted.

seniorm0ment, 2021-11-04:
Yeah nope. Gonna try rebooting the system.

seniorm0ment, 2021-11-04:
Upon rebooting, on boot, it complained 68 packages could be upgraded. 
Makes no sense because I did that after the Bullseye upgrade, and rebooted and it was fine. But I just upgraded them again, and rebooted again, and it doesn't complain.
But that didn't fix the home-assistant issue. 
Not sure what's going on.

MichaIng, 2021-11-04:
Hmm, probably there is/was an issue on a different level. Can you check:
```sh
dmesg -l emerg,alert,crit,err
```
To check which services are listening, instead of `nmap`, run on the server itself:
```sh
ss -tlpn
```
And the again HA logs:
```sh
journalctl -u home-assistant
```

seniorm0ment, 2021-11-04:
Ok, I had to reinstall python3-pip and then reinstall Mopidy-Jellyfin and now THAT works.
So something happened there with the upgrade I guess.

```
ss -tlpn
LISTEN          0               128                            0.0.0.0:8123                        0.0.0.0:*              users:(("hass",pid=430,fd=15))
```

```
# journalctl -u home-assistant                                                                                                                         Journal file /var/log/journal/c5842551d8154a289bb7080f0ac7dff0/system.journal is truncated, ignoring file.                                                          -- Journal begins at Thu 2021-11-04 18:58:07 EDT, ends at Thu 2021-11-04 19:00:39 EDT. --                                                                           Nov 04 18:58:09 systemd[1]: Started Home Assistant (DietPi).   **
```

```
# dmesg -l emerg,alert,crit,err                                                                                                                        [    0.576932] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.                                                                              [    1.539039] hid-generic 0003:0D8C:0319.0001: No inputs registered, leaving     
```

MichaIng, 2021-11-04:
Ah yes makes sense as pip + modules are no APT packages. I'll add those two to the reinstall list on the blog.

So HA is running and listening and the web UI should hence be accessible.

seniorm0ment, 2021-11-04:
>So HA is running and listening and the web UI should hence be accessible.

From the looks of it, yes. Which is what I don't understand.
I set this device up a while back, but it doesn't look like I put any firewall software on it from what I'm seeing. So that doesn't seem to be the issue which was my first though but nope.

http://dietpiIP:8123/ doesn't work

MichaIng, 2021-11-04:
Did you try it from a different client? And what error message does it show? Does it work from the server itself?
```sh
curl -IL http://127.0.0.1:8123
```
And to check for possibly unknown firewall rules:
```sh
iptables -L
```

seniorm0ment, 2021-11-04:
For some reason, I had to go directly to ip:8123/onboarding.html and it works now.
Weird but I guess I got it.

MichaIng, 2021-11-04:
Hmm, such redirects should happen automatically. However, good that it works now.

seniorm0ment, 2021-11-04:
Thank you very much for your time and help :)

MichaIng, 2021-11-09:
Finally, solved with: https://github.com/MichaIng/DietPi/commit/266a3cb

That was a long process of trial and error 😄. I strongly recommend to upgrade to Bullseye with ARMv6 and ARMv7 systems when aiming to install or update Home Assistant. On Stretch and Buster it takes a long time since most Python modules now need to be compiled, including `cryptography` which in turn requires the Rust compiler installed for the `homeassistant` user.

closed 2021-11-10 by MichaIng

seniorm0ment, 2021-11-10:
Sorry haven't gotten around to working on Home Assistant been busy lately.
Are you suggesting doing something different? Because you already had me upgrade to Bullseye didn't you?

MichaIng, 2021-11-10:
If you are on Bullseye, no need to change anything. This is only affecting Stretch and Buster systems.

Nhoya, 2021-11-12:
Hi,

I don't know if that's the right issue to reopen but installing home assistant  on a fresh version of bullseye I get multiple dependencies missing after the installation

```
systemctl status home-assistant
● home-assistant.service - Home Assistant (DietPi)
     Loaded: loaded (/etc/systemd/system/home-assistant.service; disabled; vendor preset: enabled)
     Active: active (running) since Fri 2021-11-12 19:52:30 GMT; 19min ago
   Main PID: 404173 (hass)
      Tasks: 8 (limit: 4532)
        CPU: 55.649s
     CGroup: /system.slice/home-assistant.service
             └─404173 /home/homeassistant/.pyenv/versions/3.9.7/bin/python3.9 /home/homeassistant/.pyenv/versions/3.9.7/bin/hass -c /mnt/dietpi_userdata/homeassistant

Nov 12 19:53:28 DietPi homeassistant[404173]: 2021-11-12 19:53:28 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of analytics. Setup failed for dependencies: api, websocket_api
Nov 12 19:53:28 DietPi homeassistant[404173]: 2021-11-12 19:53:28 ERROR (MainThread) [homeassistant.setup] Setup failed for analytics: Could not set up all dependencies.
Nov 12 19:53:28 DietPi homeassistant[404173]: 2021-11-12 19:53:28 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of search. Setup failed for dependencies: websocket_api
Nov 12 19:53:28 DietPi homeassistant[404173]: 2021-11-12 19:53:28 ERROR (MainThread) [homeassistant.setup] Setup failed for search: Could not set up all dependencies.
Nov 12 19:53:28 DietPi homeassistant[404173]: 2021-11-12 19:53:28 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of onboarding. Setup failed for dependencies: analytics, auth, http, person
Nov 12 19:53:28 DietPi homeassistant[404173]: 2021-11-12 19:53:28 ERROR (MainThread) [homeassistant.setup] Setup failed for onboarding: Could not set up all dependencies.
Nov 12 19:53:28 DietPi homeassistant[404173]: 2021-11-12 19:53:28 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of frontend. Setup failed for dependencies: api, auth, config, http, onboarding, search, system_log, websocket_api
Nov 12 19:53:28 DietPi homeassistant[404173]: 2021-11-12 19:53:28 ERROR (MainThread) [homeassistant.setup] Setup failed for frontend: Could not set up all dependencies.
Nov 12 19:53:28 DietPi homeassistant[404173]: 2021-11-12 19:53:28 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of safe_mode. Setup failed for dependencies: frontend, cloud
Nov 12 19:53:28 DietPi homeassistant[404173]: 2021-11-12 19:53:28 ERROR (MainThread) [homeassistant.setup] Setup failed for safe_mode: Could not set up all dependencies.
```

DietPi version | cat /boot/dietpi/.version
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
root@DietPi:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
```
   
Distro version | echo $G_DISTRO_NAME $G_RASPBIAN
```
bullseye 0
```
Kernel version | uname -a
```
Linux DietPi 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux
```
SBC model | echo $G_HW_MODEL_NAME or (EG: RPi3)
```
RPi 4 Model B (aarch64)
```

MichaIng, 2021-11-12:
This is not related. Please restart the service:
```sh
systemctl restart home-assistant
```
Please see the first run notes on our docs: https://dietpi.com/docs/software/home_automation/#home-assistant
The reason why this sometimes happens is still unknown and the issue at the Home Assistant repository pending (faced not only by us).