MichaIng/DietPi #4929: Home Assistant: possible Python 3.9 incompatibilty when enabling DHT sensor

advicepyro, 2021-11-04:
#### Required Information
DietPi version:
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
```

- Distro version: `bullseye 1`
- Kernel version | `Linux raspy 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux`
- SBC model | `RPi 3 Model B+ (armv7l)`

#### Additional Information (if applicable)
- Software title: **Home Assistant**
- Was the software title installed freshly or updated/migrated? **updated**

#### Steps to reproduce
Add the DHT sensor to Home Assistant. Example in configuration.yaml:

```yaml
sensor:
  - platform: dht
    sensor: AM2302
    pin: 4
    monitored_conditions:
      - temperature
      - humidity
```
Home Assistant sees this on the next reload and will automatically install the [adafruit-circuitpython-dht](https://pypi.org/project/adafruit-circuitpython-dht/) package. This causes an error, see below.

#### Expected behaviour
Package installs without errors.

#### Actual behaviour
Following instructions here to login as the homeassistant user and use pip3 to install this package, this is the error I get:

```
homeassistant@raspy:~$ pip3 install adafruit-circuitpython-dht
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Collecting adafruit-circuitpython-dht
  Downloading https://www.piwheels.org/simple/adafruit-circuitpython-dht/adafruit_circuitpython_dht-3.6.3-py3-none-any.whl (9.1 kB)
Collecting Adafruit-Blinka
  Using cached https://www.piwheels.org/simple/adafruit-blinka/Adafruit_Blinka-6.15.0-py3-none-any.whl (187 kB)
Collecting rpi-ws281x>=4.0.0
  Using cached rpi_ws281x-4.3.1-cp39-cp39-linux_armv7l.whl (117 kB)
Collecting Adafruit-PureIO>=1.1.7
  Using cached https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-1.1.9-py3-none-any.whl (12 kB)
Collecting pyftdi>=0.40.0
  Using cached pyftdi-0.53.3-py3-none-any.whl (141 kB)
Collecting sysv-ipc>=1.1.0
  Using cached https://www.piwheels.org/simple/sysv-ipc/sysv_ipc-1.1.0-cp39-cp39-linux_armv7l.whl (63 kB)
Collecting RPi.GPIO
  Using cached RPi.GPIO-0.7.0.tar.gz (30 kB)
Collecting Adafruit-PlatformDetect>=3.13.0
  Using cached https://www.piwheels.org/simple/adafruit-platformdetect/Adafruit_PlatformDetect-3.17.1-py3-none-any.whl (19 kB)
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting pyusb!=1.2.0,>=1.0.0
  Using cached pyusb-1.2.1-py3-none-any.whl (58 kB)
Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/homeassistant/.pyenv/versions/3.9.7/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-7prq9nl2/rpi-gpio/setup.py'"'"'; __file__='"'"'/tmp/pip-install-7prq9nl2/rpi-gpio/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-yceg036m
       cwd: /tmp/pip-install-7prq9nl2/rpi-gpio/
  Complete output (87 lines):
  running bdist_wheel
  running build
[... 165 lines cut ...]
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_gpio.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/constants.h:29: multiple definition of `output'; build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/constants.h:29: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_gpio.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/constants.h:28: multiple definition of `input'; build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/constants.h:28: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_gpio.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/constants.h:27: multiple definition of `low'; build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/constants.h:27: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_gpio.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/constants.h:26: multiple definition of `high'; build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/constants.h:26: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_pwm.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:38: multiple definition of `gpio_direction'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:38: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_pwm.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/py_pwm.h:23: multiple definition of `PWMType'; build/temp.linux-armv7l-3.9/source/py_gpio.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/py_pwm.h:23: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_pwm.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:41: multiple definition of `module_setup'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:41: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_pwm.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:40: multiple definition of `setup_error'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:40: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_pwm.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:39: multiple definition of `rpiinfo'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:39: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_pwm.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:37: multiple definition of `pin_to_gpio'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:37: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_pwm.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:36: multiple definition of `pin_to_gpio_rev3'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:36: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_pwm.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:35: multiple definition of `pin_to_gpio_rev2'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:35: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_pwm.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:34: multiple definition of `pin_to_gpio_rev1'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:34: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/py_pwm.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:33: multiple definition of `gpio_mode'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/common.h:33: first defined here
    /usr/bin/ld: build/temp.linux-armv7l-3.9/source/soft_pwm.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/soft_pwm.c:28: multiple definition of `threads'; build/temp.linux-armv7l-3.9/source/event_gpio.o:/tmp/pip-install-7prq9nl2/rpi-gpio/source/event_gpio.c:60: first defined here
    collect2: error: ld returned 1 exit status
    error: command '/usr/bin/gcc' failed with exit code 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/homeassistant/.pyenv/versions/3.9.7/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-7prq9nl2/rpi-gpio/setup.py'"'"'; __file__='"'"'/tmp/pip-install-7prq9nl2/rpi-gpio/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-5gd8qrmi/install-record.txt --single-version-externally-managed --compile --install-headers /home/homeassistant/.pyenv/versions/3.9.7/include/python3.9/RPi.GPIO Check the logs for full command output.
WARNING: You are using pip version 20.2.4; however, version 21.3.1 is available.
You should consider upgrading via the '/home/homeassistant/.pyenv/versions/3.9.7/bin/python3.9 -m pip install --upgrade pip' command.
```

#### Extra details
Workaround might be to force an older Python version and reinstall Home Assistant that way - would appreciate instructions on how to do so.

Joulinar, 2021-11-04:
Hi,

issue is with `RPi.GPIO-0.7.0` package as it is not yet supported for Python 3.9 on `piwheels.org`. https://www.piwheels.org/project/rpi-gpio/

But there is a pre-release available that could be installed manually. 

similar to this forum post https://dietpi.com/phpbb/viewtopic.php?t=9509

advicepyro, 2021-11-04:
Thanks @Joulinar! `pip3 install RPi.GPIO==0.7.1a4` then `pip3 install adafruit-circuitpython-dht` did the trick; my sensors are now showing up in HA.

closed 2021-11-04 by advicepyro