MichaIng/DietPi #4930: Wireguard fails on Odroid

StormCh, 2021-11-04:
### Creating a bug report/issue

#### Required Information
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version: stretch
- Kernel version: Linux DietPi 3.16.85+ #1 SMP PREEMPT Tue Jun 30 19:02:35 CEST 2020 aarch64 GNU/Linux
- SBC model: Odroid C2 (aarch64)

#### Additional Information (if applicable)
- Bug report ID: aabaaac3-f858-4f37-b56a-f53bbe7aa436

#### Steps to reproduce
uninstall and reinstall wireguard via dietpi-software

#### Expected behaviour
Wireguard works
- ...

#### Actual behaviour
Wireguard fails
- ...

#### Extra details
`dietpi-software
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | Free space check: path=/ | available=13455 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask 
[  OK  ] DietPi-Services | unmask : avahi-daemon
[  OK  ] DietPi-Services | unmask : proftpd
[  OK  ] DietPi-Services | unmask : nmbd
[  OK  ] DietPi-Services | unmask : smbd
[  OK  ] DietPi-Services | unmask : redis-server
[  OK  ] DietPi-Services | unmask : mariadb
[  OK  ] DietPi-Services | unmask : php7.3-fpm
[  OK  ] DietPi-Services | unmask : lighttpd
[  OK  ] DietPi-Services | unmask : plexmediaserver
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | ncc maintenance:mode --on
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : plexmediaserver
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : redis-server
[  OK  ] DietPi-Services | stop : smbd
[  OK  ] DietPi-Services | stop : nmbd
[  OK  ] DietPi-Services | stop : proftpd
[  OK  ] DietPi-Services | stop : avahi-daemon
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ INFO ] DietPi-Software | APT update, please wait...
Ign:1 https://deb.debian.org/debian stretch InRelease
Hit:2 https://downloads.plex.tv/repo/deb public InRelease
Hit:3 https://deb.debian.org/debian stretch-updates InRelease
Hit:4 https://packages.sury.org/php stretch InRelease
Hit:5 https://deb.debian.org/debian-security stretch/updates InRelease
Hit:6 https://dietpi.com/meveric all InRelease
[... 171 lines cut ...]
DKMS make.log for wireguard-0.0.20190123 for kernel 3.16.85+ (aarch64)
Thu Nov  4 20:01:42 GMT 2021
make: Entering directory '/usr/src/linux-headers-3.16.85+'
  LD      /var/lib/dkms/wireguard/0.0.20190123/build/built-in.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190123/build/main.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190123/build/noise.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190123/build/device.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190123/build/peer.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190123/build/timers.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190123/build/queueing.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190123/build/send.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190123/build/receive.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190123/build/socket.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190123/build/hashtables.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190123/build/allowedips.o
/var/lib/dkms/wireguard/0.0.20190123/build/socket.c: In function ‘send6’:
/var/lib/dkms/wireguard/0.0.20190123/build/socket.c:143:18: error: ‘const struct ipv6_stub’ has no member named ‘ipv6_dst_lookup’; did you mean ‘ipv6_dst_lookup_flow’?
   ret = ipv6_stub->ipv6_dst_lookup(sock_net(sock), sock, &dst,
                  ^~
scripts/Makefile.build:257: recipe for target '/var/lib/dkms/wireguard/0.0.20190123/build/socket.o' failed
make[1]: *** [/var/lib/dkms/wireguard/0.0.20190123/build/socket.o] Error 1
make[1]: *** Waiting for unfinished jobs....
Makefile:1363: recipe for target '_module_/var/lib/dkms/wireguard/0.0.20190123/build' failed
make: *** [_module_/var/lib/dkms/wireguard/0.0.20190123/build] Error 2
make: Leaving directory '/usr/src/linux-headers-3.16.85+'

MichaIng, 2021-11-04:
Many thanks for your report.

It looks like you have an old kernel installed aside of the used one. Can you please share the output of the following commands:
```sh
dpkg -l | grep 'linux-image'
ls -l /lib/modules
ls -l /boot
```

StormCh, 2021-11-04:
dpkg -l | grep 'linux-image':
ii  linux-image-3.16.78+              3.16.78-20191210-C2                               arm64        Linux kernel binary image for version 3.16.78+
ii  linux-image-3.16.85+              3.16.85-20200630-C2                               arm64        Linux kernel binary image for version 3.16.85+
ii  linux-image-arm64-odroid-c2       3.16.85-1                                         arm64        Linux Image/Headers Meta Package for ODROIDs

ls -l /lib/modules:
drwxr-xr-x 3 root root 4096 Mar 24  2021 3.16.78+
drwxr-xr-x 3 root root 4096 Nov  4 18:39 3.16.85+

ls -l /boot:
-rwxr-xr-x 1 root root     2316 May  4  2020 boot.ini
-rwxr-xr-x 1 root root   114064 Dec 10  2019 config-3.16.78+
-rwxr-xr-x 1 root root   114065 Jun 30  2020 config-3.16.85+
drwxr-xr-x 4 root root     4096 Nov  3 11:04 dietpi
-rwxr-xr-x 1 root root     6285 Sep 10  2018 dietpi-README.md
-rwxr-xr-x 1 root root    11511 Nov  4 18:42 dietpi.txt
-rwxr-xr-x 1 root root 13810864 Jan 23  2021 Image
-rwxr-xr-x 1 root root  4570811 Nov  2  2020 initrd.img-3.16.78+
-rwxr-xr-x 1 root root  4583848 Aug 31 16:40 initrd.img-3.16.85+
-rwxr-xr-x 1 root root    30004 Jun 30  2020 meson64_odroidc2.dtb
-rwxr-xr-x 1 root root  3095003 Dec 10  2019 System.map-3.16.78+
-rwxr-xr-x 1 root root  3096490 Jun 30  2020 System.map-3.16.85+
drwxr-xr-x 2 root root     2048 Feb 23  2018 System Volume Information
-rwxr-xr-x 1 root root  4575452 Aug 31 16:40 uInitrd
-rwxr-xr-x 1 root root  4567503 Jan 18  2020 uInitrd-3.16.57+
-rwxr-xr-x 1 root root  4570875 Jan 23  2021 uInitrd-3.16.78+
-rwxr-xr-x 1 root root  4575452 Aug 31 16:40 uInitrd-3.16.85+
-rwxr-xr-x 1 root root 13810608 Dec 10  2019 vmlinuz-3.16.78+
-rwxr-xr-x 1 root root 13810864 Jun 30  2020 vmlinuz-3.16.85+

MichaIng, 2021-11-04:
Please remove the old one:
```sh
apt purge linux-image-3.16.78+
rm -Rf /lib/modules/3.16.78+ /boot/*-3.16.{57,78}+
```
Then try to finish the DKMS build:
```sh
dpkg --configure -a
```

StormCh, 2021-11-04:
I did your hints:

After that I tried:
```
dpkg-reconfigure wireguard-dkms

------------------------------
Deleting module version: 0.0.20190123
completely from the DKMS tree.
------------------------------
Done.
Loading new wireguard-0.0.20190123 DKMS files...
Building for 3.16.85+
Building initial module for 3.16.85+
Error! Bad return status for module build on kernel: 3.16.85+ (aarch64)
Consult /var/lib/dkms/wireguard/0.0.20190123/build/make.log for more information.
```

Still the same issue.

MichaIng, 2021-11-04:
What I'm also wondering is why such an outdated WireGuard version is tried to be installed. A much newer one is available: https://packages.debian.org/bullseye/wireguard-dkms

Can you please show:
```sh
dpkg -l | grep wireguard
apt policy wireguard wireguard-dkms
```

Ah and since the kernel was indeed upgraded, please do you a reboot as well to assure the new kernel is loaded.

StormCh, 2021-11-04:
dpkg -l | grep wireguard:
```
ii  wireguard-dkms                    0.0.20190123-1                                    all          fast, modern, secure kernel VPN tunnel (DKMS version)
ii  wireguard-tools                   1.0.20210223-1                                    arm64        fast, modern, secure kernel VPN tunnel (userland utilities)
```

apt policy wireguard wireguard-dkms:
```
wireguard:
  Installed: (none)
  Candidate: 0.0.20190123-1
  Version table:
     1.0.20210223-1 100
         -1 https://deb.debian.org/debian bullseye/main arm64 Packages
     0.0.20190123-1 500
        500 https://dietpi.com/meveric stretch/backports arm64 Packages
wireguard-dkms:
  Installed: 0.0.20190123-1
  Candidate: 0.0.20190123-1
  Version table:
     1.0.20210219-1 100
         -1 https://deb.debian.org/debian bullseye/main arm64 Packages
 *** 0.0.20190123-1 500
        500 https://dietpi.com/meveric stretch/backports arm64 Packages
        100 /var/lib/dpkg/status
```

MichaIng, 2021-11-04:
Something is blocking the packages from Debian Bullseye and instead old ones from Meverics repository are installed. Can you please show:
```sh
for i in /etc/apt/preferences{,.d/*}
do
echo "$i:"
cat "$I"
done
```

StormCh, 2021-11-04:
are you sure with uppercase I ?

```
/etc/apt/preferences:
cat: '': No such file or directory
/etc/apt/preferences.d/backports:
cat: '': No such file or directory
/etc/apt/preferences.d/deb-multimedia-pin-99:
cat: '': No such file or directory
/etc/apt/preferences.d/dietpi-openssl:
cat: '': No such file or directory
/etc/apt/preferences.d/dietpi-wireguard:
cat: '': No such file or directory
```

MichaIng, 2021-11-04:
Ui, your system seems to be a very old one. Those two files are obsolete an can be removed:
```
rm /etc/apt/preferences.d/backports /etc/apt/preferences.d/deb-multimedia-pin-99
```

And then for some reason in `/etc/apt/preferences.d/dietpi-wireguard` the package specific pin priority `100` seems to not override the global Bullseye `-1` priority. I remember I was tinkering with those a while back, probably the behaviour has changed after Stretch. Can you try:
```sh
echo -e 'Package: wireguard wireguard-dkms wireguard-tools\nPin: release n=bullseye\nPin-Priority: 100\n
Package: *\nPin: release n=bullseye\nPin-Priority: -1' > /etc/apt/preferences.d/dietpi-wireguard
apt update
apt policy wireguard-dkms
```
This exchanges the order.

StormCh, 2021-11-04:
Ok, now I have:

```
wireguard-dkms:
  Installed: 0.0.20190123-1
  Candidate: 0.0.20190123-1
  Version table:
     1.0.20210219-1 100
         -1 https://deb.debian.org/debian bullseye/main arm64 Packages
 *** 0.0.20190123-1 500
        500 https://dietpi.com/meveric stretch/backports arm64 Packages
        100 /var/lib/dpkg/status
```

MichaIng, 2021-11-04:
Ah, lol it was "fine" already, same as before:
```
     1.0.20210219-1 100
         -1 https://deb.debian.org/debian bullseye/main arm64 Packages
```
Means the list generally has priority `-1` but the particular package/version has priority `100`, so that is all fine. But Meverics packages have priority 500 and are hence preferred. So we need to pin them down:
```
echo -e 'Package: wireguard wireguard-dkms wireguard-tools\nPin: origin dietpi.com\nPin-Priority: -1' > /etc/apt/preferences.d/dietpi-wireguard-meveric
apt update
apt install wireguard-dkms wireguard-tools
```

MichaIng, 2021-11-04:
> Thanks a lot, and yes the server is "very" old, but still good...

I recommend to upgrade the Debian base version to Buster, and probably even further to Bullseye. Stretch => Buster works pretty well 🙂: https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

The two preferences files are then not required anymore as Debian Buster backports natively ship WireGuard.

closed 2021-11-11 by MichaIng