MichaIng/DietPi #4931: Problem with nre version python 3.9 with some object of HA

maury77, 2021-11-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | `7.7.2
- Distro version | `bullseye`
- Kernel version | `linuxux sdomotica 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux`
- SBC model | RPi 3 Model B+ (armv7l)
- Power supply used | (EG: 5V 3A)

I upgrade to dietPi 7.7.2 and  Home assistant with python 3.9.7

after upgrade I have this error

#### Steps to reproduce
nov 04 22:29:35 sdomotica homeassistant[3274]: IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!
nov 04 22:29:35 sdomotica homeassistant[3274]: Importing the numpy C-extensions failed. This error can happen for
nov 04 22:29:35 sdomotica homeassistant[3274]: many reasons, often due to issues with your setup or how NumPy was
nov 04 22:29:35 sdomotica homeassistant[3274]: installed.
nov 04 22:29:35 sdomotica homeassistant[3274]: We have compiled some common reasons and troubleshooting tips at:
nov 04 22:29:35 sdomotica homeassistant[3274]:     https://numpy.org/devdocs/user/troubleshooting-importerror.html
nov 04 22:29:35 sdomotica homeassistant[3274]: Please note and check the following:
nov 04 22:29:35 sdomotica homeassistant[3274]:   * The Python version is: Python3.9 from "/home/homeassistant/.pyenv/versions/3.9.7/bin/python3.9"
nov 04 22:29:35 sdomotica homeassistant[3274]:   * The NumPy version is: "1.21.3"
nov 04 22:29:35 sdomotica homeassistant[3274]: and make sure that they are the versions you expect.
nov 04 22:29:35 sdomotica homeassistant[3274]: Please carefully study the documentation linked above for further help.
nov 04 22:29:35 sdomotica homeassistant[3274]: Original error was: libcblas.so.3: cannot open shared object file: No such file or directory

MichaIng, 2021-11-04:
Many thanks for your report.

This is the issue, I'm working on a workaround for tomorrow: https://github.com/MichaIng/DietPi/issues/4928

I'll close this in favour if the linked issue.

Joulinar, 2021-11-04:
@MichaIng 
but this is a `bullseye` system

MichaIng, 2021-11-04:
Ah whoops, right.

@maury77 
Please try:
```sh
apt install libblas3
systemctl restart home-assistant
```

Which additional modules have you installed in Home Assistant? This library seems to be not required on a fresh HA, but probably some extension does.

maury77, 2021-11-05:
> Ah whoops, right.
> 
> @maury77 Please try:
> 
> ```shell
> apt install libblas3
> systemctl restart home-assistant
> ```
> 
> Which additional modules have you installed in Home Assistant? This library seems to be not required on a fresh HA, but probably some extension does.

Now is ok thanks...

the problem is the upgrade to bullseye....

MichaIng, 2021-11-05:
> the problem is the upgrade to bullseye....

No it is not generally required on Bullseye, but likely an extension you use pulls a Python modules from piwheels.org which requires it. It is expected that additional packages are required when using HA extension, which is the reason we have these `dietpi.txt` settings 🙂: https://github.com/MichaIng/DietPi/blob/a4cf2c5/dietpi.txt#L261-L266

closed 2021-11-07 by maury77