MichaIng/DietPi #4932: Use two WiFi Dongle for hotspot

iw4p, 2021-11-06:
Hello, 

I have RP zero w, I want to connect to the internet with one of my WiFi adaptors (For example onboard) and hotspot it via the external dongle WiFi which is connected to board, but DietPi forces me to use Ethernet for hotspot and I don't want to use LAN.

I searched it in issues but I didn't find any solution.

Can you help me, please?
Thanks

Joulinar, 2021-11-06:
Hi,

Unfortunately this is a hard code dependency inside of our script. We are aware on this limitation and on a long term, we are planning to rework the network stack to get more flexible. But this is a larger task and will take time.

iw4p, 2021-11-06:
Oh Ok, Thanks for your response
Is there anything I can do? or any alternative like DietPi with two dongles?

Joulinar, 2021-11-06:
If you are familiar with bash coding, you could try to adjust `dietpi-software` script to bypass the test and to set your wlan adapter. An alternative you could look into might be `RaspAP` https://raspap.com/

iw4p, 2021-11-06:
I think [this is](https://github.com/MichaIng/DietPi/blob/04007528e54db4511d298e59318deddcd74912fd/dietpi/dietpi-software#L17506) where I'm looking for, right? Line 17506

Joulinar, 2021-11-06:
would be more this one to bypass the check in total

https://github.com/MichaIng/DietPi/blob/04007528e54db4511d298e59318deddcd74912fd/dietpi/dietpi-software#L17489

Just set `INSTALL_STATE` higher than `3` and the check should not be triggered at all.

However this will not solve your problem as the install scrip still expect `eth` interface. You would need to set `eth_iface` and `wifi_iface` manually

https://github.com/MichaIng/DietPi/blob/04007528e54db4511d298e59318deddcd74912fd/dietpi/dietpi-software#L11366-L11367

I guess `eth_iface` would need to be your `wlan1` and `wifi_iface` to `wlan0`

As well it would be needed to establish internet connection via `wlan1` first

But this is just guessing. I never tested it and I don't have such WiFi dongle to try. Maybe better to wait on our developer @MichaIng as he knows what to change and what is possible.

iw4p, 2021-11-06:
Thanks, I think these guesses are fine and they'll work.
One another thing, after modifying it, is there any help or doc to make it installable? I mean compiling(?) or something like this ...
literally, I don't know what is the next step after fixing this problem for myself... any good way to debug it or just try to run it each time ...?

Joulinar, 2021-11-06:
I'm not sure what you mean exactly with compiling and debugging? Maybe you could describe it more in detail. 😃

iw4p, 2021-11-06:
I have experience in bash coding, but I've never done it on an OS source :D, so is there any recipe for me to know what I have to do after I git clone this repo and make my changes :D

Joulinar, 2021-11-06:
no need to clone the git. No need to compile anything. DietPi is not an own OS. It's just a set of bash scripts. You could edit the files directly on your device. Just check `/boot/dietpi/` 😉

iw4p, 2021-11-06:
Oh, I got it :D nice ... I'll try ASAP and if It's possible I like to contribute this feature ...

MichaIng, 2021-11-06:
Instead of changing the scripts is is probably easier to set it up manually 😉.

Before you spend time to contribute, note that we will do a complete rewrite of the network settings in `dietpi-config` which will then include the WiFi Hotspot as an option in network settings instead of a dedicated software option.

Here is the related project, which contains as very first request a duplicate of yours 😉: https://github.com/MichaIng/DietPi/projects/1

MichaIng, 2022-02-06:
Marking this as closed. Feel free to reopen when required.

closed 2022-02-06 by MichaIng