MichaIng/DietPi #4933: PiHole + PiVPN + UFW strange, external, UDP requests on port 443

GvY85, 2021-11-06:
Hi,

I have a DietPi installation running on an x84_64 mini pc with Pihole, PiVPN (Wireguard) and UFW.
This behind a FritzBox fiber router with the only port open being the incomming Wireguard port (`12000`).

Since a few days I am seeying strange UFW blocks on `port 443 UDP` (so not HTTPS on TCP) comming from ip adresses `31.13.64.16` , `31.13.64.51`, `31.13.64.11` and `157.240.201.60` 
These all seem to be ip adresses belonging to Facebook.
They come in short burts a few time an hour.

Some examples (`192.168.178.3 `being the PiHole device):
```
[Fri Nov  5 15:06:15 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=60 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=49163 LEN=40
[Fri Nov  5 15:06:15 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=49163 LEN=39
[Fri Nov  5 15:06:15 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=49163 LEN=39
[Fri Nov  5 15:06:15 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=49163 LEN=39
[Fri Nov  5 15:06:16 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=49163 LEN=39
[Fri Nov  5 15:06:16 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=49163 LEN=39
[Fri Nov  5 15:06:27 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=49163 LEN=39
[Fri Nov  5 15:28:44 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.51 DST=192.168.178.3 LEN=60 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=62186 LEN=40
[Fri Nov  5 15:28:44 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.51 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=62186 LEN=39
[Fri Nov  5 15:28:44 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.51 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=62186 LEN=39
[Fri Nov  5 15:28:44 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.51 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=62186 LEN=39
[Fri Nov  5 15:28:45 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.51 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=62186 LEN=39
[Fri Nov  5 15:28:45 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.51 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=62186 LEN=39
[Fri Nov  5 15:28:47 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.51 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=62186 LEN=39
[Fri Nov  5 15:28:50 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.51 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=62186 LEN=39
[Fri Nov  5 15:28:57 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.51 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=62186 LEN=39
[Fri Nov  5 15:29:11 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.51 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=62186 LEN=39
[Fri Nov  5 15:29:38 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.51 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=62186 LEN=39
[Fri Nov  5 15:47:01 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.11 DST=192.168.178.3 LEN=52 TOS=0x00 PREC=0x00 TTL=88 ID=59110 DF PROTO=TCP SPT=443 DPT=51454 WINDOW=269 RES=0x00 ACK URGP=0
[Fri Nov  5 15:48:07 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=60 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=56318 LEN=40
[Fri Nov  5 15:48:08 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=56318 LEN=39
[Fri Nov  5 15:48:09 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=56318 LEN=39
[Fri Nov  5 15:48:11 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=56318 LEN=39
[Fri Nov  5 15:48:16 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=56318 LEN=39
[Fri Nov  5 15:48:24 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=56318 LEN=39
[Fri Nov  5 15:48:42 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=56318 LEN=39
[Fri Nov  5 15:49:16 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=56318 LEN=39
[Fri Nov  5 15:50:26 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=56318 LEN=39
[Fri Nov  5 15:54:44 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.11 DST=192.168.178.3 LEN=52 TOS=0x00 PREC=0x00 TTL=88 ID=53016 DF PROTO=TCP SPT=443 DPT=51479 WINDOW=273 RES=0x00 ACK URGP=0
[Fri Nov  5 16:07:22 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=60 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=58342 LEN=40
[Fri Nov  5 16:07:22 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=58342 LEN=39
[Fri Nov  5 16:07:22 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=58342 LEN=39
[Fri Nov  5 16:07:22 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=58342 LEN=39
[Fri Nov  5 16:07:23 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=58342 LEN=39
[Fri Nov  5 16:07:23 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=58342 LEN=39
[Fri Nov  5 16:07:25 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=58342 LEN=39
[Fri Nov  5 16:07:28 2021] [UFW BLOCK] IN=eth0 OUT= MAC=00:07:32:6b:9b:17:dc:15:c8:b7:ef:1a:08:00 SRC=31.13.64.16 DST=192.168.178.3 LEN=59 TOS=0x00 PREC=0x00 TTL=88 ID=0 DF PROTO=UDP SPT=443 DPT=58342 LEN=39
[... cut ...]
What I find strange is that it is `UDP` traffic on `port 443` while you perhaps would expect traffic on` port 443 `to be `TCP `and thus being `HTTPS `traffic?
According to [Wikipedia ](https://en.wikipedia.org/wiki/List_of_TCP_and_UDP_port_numbers) `UDP `traffic on `port 443` could be `QUIC`?

But what I find the strange is that this is traffic comming from the internet, outside of the router, and is somehow able to go straight to the PiHole device?
I double checked and the only open port on my router is port `12000 `bound to `192.168.178.3`

Joulinar, 2021-11-06:
On your FritzBox you could activate some package capturing to see if the traffic is indeed received from Internet. As well you could do some `tcpdump` tracing on your DietPi device to see if there is more communication ongoing.

GvY85, 2021-11-06:
With `http://fritz.box/html/capture.html `you mean?

Joulinar, 2021-11-06:
yeap this one. You could try to capture your internet connection. Be careful it could be a large file if your keep it running for to long. Once stopped, your FritzBox should create a file. This could be analyzed using a tool like Wireshark https://www.wireshark.org/

GvY85, 2021-11-06:
I managed to capture one and it is traffic directed to my public IP:

[image]

AVMAudio?

Joulinar, 2021-11-06:
maybe check with AVM why such traffic is not blocked and forward into your local network

MichaIng, 2021-11-06:
These are QUIC requests sent by HTTP/3 capable clients connected through your VPN. It uses UDP port 443 indeed. I'm however not aware that any currently available webserver supports HTTP/3 QUIC already, so allowing these requests in UFW may only cause overhead and longer delay until the client falls back to HTTP/2 or HTTP/1.1.

GvY85, 2021-11-06:
I was thinking something like that.
I find it very strange that it looks like it is outside of Wireguard and able to bypass the NAT. 
Only port open (also UDP) is another one that I use for Wireguard

MichaIng, 2021-11-06:
> I cannot trigger them manualy by reconnecting Wireguard, opening the PiHole admin page on clients or opening Whatsapp/Insta on clients.

Hmm strange. Probably with help of the timestamps you're able to identify the client and software/action which triggers it. Also the `wg` command may help to check which client connected when.

GvY85, 2021-11-12:
I have a reply from AVM (Fritzbox manufacturer):

> Based on your feedback, we would like to share some information with you and have a question.
> 
> Based on the data and information provided, it looks like the data shown are responses to packets from the PiHole. Due to the fact that the PiHole has previously requested packets, the responses are naturally allowed through the firewall.
> 
> The local port is random, but the port on the Internet is 443.
> 
> We suspect that this is an error on the part of the PiHole.
> Why is it sending these packets, or why are you wondering about the replies? 

So, what would pihole be sending fo Facebook and Apple servers in order for them to start sending these packets?

MichaIng, 2021-11-12:
So they are saying it is an outbound connection, initiated from Pi-hole and the packets from Apple and Facebook are hence answers? It fits to the fact that `443` is shown as `Source Port` while the `Dst Port` is a random >50.000 one 👍.

Looks like you need to identify the process on the Pi-hole server which sends this requests. Note that it doesn't need to be Pi-hole but can be anything else on that system, of course.

This can be used to monitor outgoing UDP requests with destination for 443:
```sh
ss -upE '( dport = 443 )'
```
Sadly the `-p` flag to show processes seems to not work, at least no when I test it with some `curl` or `ping` requests. Indeed both trigger such kind of requests with specific hosts. E.g. `dietpi.com`, probably because we are behind Cloudflare which allows QUIC/HTTP3 connections to their edge servers.

GvY85, 2021-11-13:
Hmm, let it run for a while (overnight) but did not really got any wiser. The Facebook and Apple ip's dont show up so the traffic below seems to be the device checking for Apt updates etc.

`ss -upE '( dport = 443 )'`
```
Recv-Q         Send-Q                 Local Address:Port                   Peer Address:Port         Process
0              -1                     192.168.178.3:46393                 172.67.209.242:https
0             -1                     192.168.178.3:51169                 185.199.111.133:https
0             -1                     192.168.178.3:51611                  172.67.153.242:https
0             -1                     192.168.178.3:57667                 185.199.109.133:https
0             -1                     192.168.178.3:60731                 185.199.111.133:https
0             -1                     192.168.178.3:33311                 185.199.109.133:https
0             -1                     192.168.178.3:47397                 185.199.108.153:https
0             -1                     192.168.178.3:58870                  172.67.193.211:https
0             -1                     192.168.178.3:43512                  172.67.193.211:https
0             -1                     192.168.178.3:35532                  172.67.193.211:https
0             -1                     192.168.178.3:34318                  172.67.193.211:https
0             -1                     192.168.178.3:38326                  172.67.193.211:https
0             -1                     192.168.178.3:58266                  172.67.193.211:https
0             -1                     192.168.178.3:54409                  172.67.193.211:https
0             -1                     192.168.178.3:48555                  172.67.193.211:https
0             -1                     192.168.178.3:56647                    45.60.103.34:https
0             -1                     192.168.178.3:53897                 185.199.111.133:https
0             -1                     192.168.178.3:53025                 185.199.109.133:https
0             -1                     192.168.178.3:34502                 185.199.111.133:https
0             -1                     192.168.178.3:37085                 185.199.109.133:https
0             -1                     192.168.178.3:47770                 185.199.111.133:https
0             -1                     192.168.178.3:37202                 185.199.109.133:https
0             -1                     192.168.178.3:37562                  172.67.193.211:https
0             -1                     192.168.178.3:37419                  172.67.193.211:https
0             -1                     192.168.178.3:36444                  172.67.193.211:https
0             -1                     192.168.178.3:35208                    172.67.158.5:https
0             -1                     192.168.178.3:33083                 185.199.111.133:https
0             -1                     192.168.178.3:49424                 185.199.109.133:https
0             -1                     192.168.178.3:58779                 185.199.111.133:https
0             -1                     192.168.178.3:36663                  172.67.153.242:https
0             -1                     192.168.178.3:50461                   104.21.12.243:https
0             -1                     192.168.178.3:51140                 185.199.111.153:https
0             -1                     192.168.178.3:55903                 185.199.109.133:https
0             -1                     192.168.178.3:56380                 185.199.111.133:https
0             -1                     192.168.178.3:58389                 185.199.109.133:https
0             -1                     192.168.178.3:43850                 185.199.111.133:https
0             -1                     192.168.178.3:34532                  172.67.193.211:https
0             -1                     192.168.178.3:40073                 185.199.109.133:https
0             -1                     192.168.178.3:47457                  172.67.193.211:https
0             -1                     192.168.178.3:38455                   172.65.251.78:https
0             -1                     192.168.178.3:36638                 185.199.111.133:https
0             -1                     192.168.178.3:34772                 185.199.109.133:https
0             -1                     192.168.178.3:50973                   104.192.141.1:https
0             -1                     192.168.178.3:48626                   172.65.251.78:https
0             -1                     192.168.178.3:33487                 185.199.111.133:https
0             -1                     192.168.178.3:33211                 185.199.109.133:https
0             -1                     192.168.178.3:55456                 185.199.111.133:https
0             -1                     192.168.178.3:54228                 185.199.108.133:https
0             -1                     192.168.178.3:37218                 185.199.109.133:https
0             -1                     192.168.178.3:45939                 185.199.110.133:https
0             -1                     192.168.178.3:48189                 185.199.111.133:https
0             -1                     192.168.178.3:50138                 185.199.108.133:https
0             -1                     192.168.178.3:36488                 185.199.109.133:https
0             -1                     192.168.178.3:52955                 185.199.110.133:https
[... cut ...]
0           -1               [::ffff:192.168.178.3]:42586            [::ffff:172.67.209.242]:https
0           -1                        192.168.178.3:49489                    185.199.111.133:https
0           -1                        192.168.178.3:41593                     172.67.209.242:https
```

So I have a feeling that it are the devices connection via PiVPN (Wireguard) that are cousing these 'answers' but they do not show up as outgoing requests from the PiHole server, just the responses are targeted to it.

MichaIng, 2021-11-13:
> but they do not show up as outgoing requests from the PiHole server

Probably just not on UDP port 443? Sounds pretty reasonable that it's the VPN clients.

MichaIng, 2021-11-30:
The most IPs are from GitHub CDN. Probably QUIC doesn't work well with a VPN in general, the client sends a request, the QUIC capable webserver answer with this UDP packet to indicate that it supports HTTP3/QUIC, but the VPN server isn't able to detect that it's meant for the VPN client, or so? Traffic then hence continues on TCP.

However, I'll mark the issue as closed now. Interesting but if deeper investigation is wanted, I suggest to ask about this at StackExchange.

closed 2021-11-30 by MichaIng