MichaIng/DietPi #4935: Upgrade is failing. Getting 404 errors

generalsnipe, 2021-11-06:
#### Details:
- Date           | Sat Nov  6 21:39:08 GMT 2021
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-zip_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-intl_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-json_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-opcache_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-fpm_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-readline_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-gd_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-xml_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-curl_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-sqlite3_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-mbstring_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-cli_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-common_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

MichaIng, 2021-11-06:
Many thanks for your report.

Looks like a temporary repo inconsistency. Please try it from console:
```sh
apt update
apt upgrade
```

generalsnipe, 2021-11-06:
Yep. Tried that. No Luck

`root@DietPi:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Err:2 https://packages.sury.org/php stretch InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
Reading package lists... Done
Building dependency tree
Reading state information... Done
77 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php stretch InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Failed to fetch https://packages.sury.org/php/dists/stretch/InRelease  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  apt apt-transport-https apt-utils bind9-host ca-certificates cron curl dnsutils e2fslibs e2fsprogs git git-man isc-dhcp-client klibc-utils libapt-inst2.0 libapt-pkg5.0 libbind9-140 libbsd0
  libcomerr2 libcurl3 libcurl3-gnutls libdns-export162 libdns162 libelf1 libfreetype6 libgcrypt20 libgnutls30 libgssapi-krb5-2 libhogweed4 libisc-export160 libisc160 libisccc140 libisccfg140
  libjpeg62-turbo libk5crypto3 libklibc libkrb5-3 libkrb5support0 libldap-2.4-2 libldap-common liblwres141 liblz4-1 libnettle6 libnghttp2-14 libp11-kit0 libsqlite3-0 libss2 libssl1.0.2 libssl1.1
  libsystemd0 libtiff5 libudev1 libwebp6 libx11-6 libx11-data login openssl passwd php7.3-cli php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-intl php7.3-json php7.3-mbstring
  php7.3-opcache php7.3-readline php7.3-sqlite3 php7.3-xml php7.3-zip sqlite3 sudo systemd systemd-sysv tzdata udev
77 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 3,830 kB/33.4 MB of archives.
After this operation, 44.0 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 https://packages.sury.org/php stretch/main armhf php7.3-zip armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
Err:2 https://packages.sury.org/php stretch/main armhf php7.3-intl armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
Err:3 https://packages.sury.org/php stretch/main armhf php7.3-json armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
Err:4 https://packages.sury.org/php stretch/main armhf php7.3-opcache armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
Err:5 https://packages.sury.org/php stretch/main armhf php7.3-fpm armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
Err:6 https://packages.sury.org/php stretch/main armhf php7.3-readline armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
Err:7 https://packages.sury.org/php stretch/main armhf php7.3-gd armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
Err:8 https://packages.sury.org/php stretch/main armhf php7.3-xml armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
Err:9 https://packages.sury.org/php stretch/main armhf php7.3-curl armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
Err:10 https://packages.sury.org/php stretch/main armhf php7.3-sqlite3 armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
Err:11 https://packages.sury.org/php stretch/main armhf php7.3-mbstring armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
Err:12 https://packages.sury.org/php stretch/main armhf php7.3-cli armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
Err:13 https://packages.sury.org/php stretch/main armhf php7.3-common armhf 7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2
  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-zip_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-intl_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-json_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-opcache_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-fpm_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-readline_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-gd_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-xml_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-curl_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
[... cut ...]
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-mbstring_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-cli_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-common_7.3.21-1+0~20200807.66+debian9~1.gbp18a1c2_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
`

MichaIng, 2021-11-06:
The package list I can see contains the correct version `7.3.31` instead of `7.3.21`: https://packages.sury.org/php/dists/stretch/main/binary-armhf/Packages

Ah the list was not updated as of certification issue. A good case where we can debug our pre-patch which should have solved it. Can you please do the following and show the output here:
```sh
cd /tmp
bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/pre-patch_file')"
```

generalsnipe, 2021-11-06:
Thanks here is the output:
```
root@DietPi:~# cd /tmp
root@DietPi:/tmp# bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/pre-patch_file')" bash 31
[ INFO ] Pre-patch 25 | Running DietPi v7 pre-patches
[ INFO ] DietPi-Pre-patches | Debian Bullseye has been released, so we accept the repository suite change once.
[  OK  ] DietPi-Pre-patches | apt-get -o Acquire::AllowReleaseInfoChange::Suite=true -q update
[ INFO ] DietPi-Pre-patches | Connecting to Mosquitto and Webmin repositories via plain HTTP once until latest OpenSSL has been installed: https://github.com/MichaIng/DietPi/issues/4795
[  OK  ] DietPi-Pre-patches | sed -Ei s#https://(repo.mosquitto.org|download.webmin.com)#http://\1# /etc/apt/sources.list
root@DietPi:/tmp#

```

MichaIng, 2021-11-06:
So indeed the sury.org key update was not applied. Please run and show:
```sh
apt-key list '95BD4743' 2> /dev/null
apt-key list '95BD4743' 2> /dev/null | grep 'expires: 2021-'
```

generalsnipe, 2021-11-06:
```
root@DietPi:/# apt-key list '95BD4743'
pub   rsa3072 2019-03-18 [SC] [expired: 2021-03-17]
      1505 8500 A023 5D97 F5D1  0063 B188 E2B6 95BD 4743
uid           [ expired] DEB.SURY.ORG Automatic Signing Key <deb@sury.org>

root@DietPi:/# apt-key list '95BD4743' 2> /dev/null
pub   rsa3072 2019-03-18 [SC] [expired: 2021-03-17]
      1505 8500 A023 5D97 F5D1  0063 B188 E2B6 95BD 4743
uid           [ expired] DEB.SURY.ORG Automatic Signing Key <deb@sury.org>

root@DietPi:/# apt-key list '95BD4743' 2> /dev/null | grep 'expires: 2021-'
```

MichaIng, 2021-11-06:
Oh, if a key has been expired already, it shows `expired: <date>` while we check for `expires: <date>`, which only shown until it has been expired but was all that we could test with valid certs 😄. I'll fix it. Thanks for debugging.

To fix it for you:
```sh
apt-key del '95BD4743'
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg
apt update
apt upgrade
dietpi-update 1
```

Joulinar, 2021-11-06:
btw, you might have a look to our Debian Buster upgrade guide as you are running oldold Debian Stretch.

https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

generalsnipe, 2021-11-06:
> btw, you might have a look to our Debian Buster upgrade guide as you are running oldold Debian Stretch.
> 
> https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

Thanks. I'll check that out. 

@MichaIng - It worked. All good now. Thanks for the quick help.

MichaIng, 2021-11-08:
I'll mark this as closed as of the fix applied to our pre-patch.