MichaIng/DietPi #4936: Open Beta v7.8 | Please help testing and hardening the upcoming release

MichaIng, 2021-11-06:
RC version | v7.8.0
---------- | -------
v7.8.0 PR | #4934
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2021-10-16**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av7.8
____
### Beta v7.8.0
_(2021-10-13)_

#### New SBC support
- Radxa Zero | Initial support for this Raspberry Pi Zero form factored (but way more powerful) SBC has been added to DietPi with the hardware ID 74. Many thanks to @almirus and @dhry for helping with testing and debugging an early image: https://github.com/MichaIng/DietPi/issues/4831
- Raspberry Pi Zero 2 W | Initial support for this Raspberry Pi Zero/Zero W successor has been added, using hardware ID 3 for now (shared with RPi 3/3+), including overclocking profiles. Many thank to `@CassTG` for providing early hardware information and doing intensive overclocking tests: https://dietpi.com/phpbb/viewtopic.php?t=9599

#### New software
- DietPi-Dashboard | Thanks to the great work of @ravenclaw900, we are proud to announce our own official DietPi web interface for monitoring and managing your DietPi system using your web browser. It is still in beta phase as we test and gradually implement more features. We would be happy if you would try it out via "dietpi-software install 200" and give us feedback: https://github.com/MichaIng/DietPi/issues/448

#### Removed software
- Subsonic | Since it is not developed anymore and due to shared library dependencies only compatible with Debian Stretch, we removed Subsonic from DietPi-Software. With Airsonic-Advanced, we'll provide a well maintained and Bullseye-compatible alternative. If you currently have Subsonic installed, it will remain. If you want to uninstall it, follow the instructions here: https://github.com/MichaIng/DietPi/pull/4895
- emonHub | Since we do not have a single reported installation, we removed emonHub from DietPi-Software. If you currently have emonHub installed, it will remain. If you want to keep using and updating it, have a look at the official repository: https://github.com/openenergymonitor/emonhub. If you want to uninstall it, follow the instructions here: https://github.com/MichaIng/DietPi/pull/4895

#### Changes
- Odroid N1 | Removed support: We don't have a single reported system, which makes sense as this model was never really released. Only a small number of developer samples are floating around, not worth to keep maintaining an image and dedicated code. If there is really an Odroid N1 DietPi system out there, it will be automatically migrated to the Generic Rockchip RK3399 device ID on DietPi update.
- DietPi-Config | Overclocking profiles for most Raspberry Pi models have been updated to match the effective 100 MHz ARM/CPU frequency steps used by current firmware. Also an energy saving profile has been added to some models, which reduces minimum and maximum voltage a little, reducing energy consumption and heat emission a little on idle and on load.
- DietPi-Config | The option to prefer IPv4 connections when IPv6 is enabled has been removed: This only worked for APT and wget, while e.g. cURL and ping were never affected by this setting, which meant an inconsistent behaviour. If one faces issues with IPv6 enabled, it should be simply disabled instead of preferring/forcing IPv4 only for specific tools.
- DietPi-Backup | A feature has been added to enable daily system backup via cron job. Many thanks to @Lycidias93 for suggesting this feature: https://github.com/MichaIng/DietPi/issues/3871
- DietPi-Backup | Rsync/transfer logs are now done to "dietpi-backup.log" inside of the backup directory. This way they remain persistent even if DietPi-RAMlog is enabled and can be reviewed at a later time. This is helpful especially when daily backups via cron job are enabled, where the log obviously is not presented directly when the backup finished. A related option to review the last transfer log has been added to the dietpi-backup main menu. The old log file /var/log/dietpi-backup.log is moved to the new location if any backup exists during next DietPi update.
- DietPi-Backup | When using an NFS mount as backup target, it is now verified that the NFS share supports UNIX permissions to prevent the creation of a broken system backup in the first place.
- DietPi-Drive_Manager | When transferring the DietPi user data to a different drive, it is now ckecked whether the target location is located within a supported filesystem type, one with native symlink and UNIX permission support. The same is done when using the CLI script /boot/dietpi/func/dietpi-set_userdata.
- DietPi-Drive_Manager | Transferring the root filesystem is now supported on Odroid C4/HC4 models.
- DietPi-Drive_Manager | When transferring the root filesystem, further checks are performed: There needs to be a dedicated boot partition (required on Raspberry Pi and default on our current Odroid images), as this is what the transfer steps expect and the only reason for moving only the root filesystem instead of cloning the whole drive or flashing the DietPi image to an external drive in the first place. The expected boot/kernel configuration files need to be present so that the kernel can be instructed to mount the new root filesystem. The DietPi user data must not currently be located at the target partition as it would be formatted and all data hence lost.
- DietPi-Drive_Manager | The option to transfer the root filesystem is now shown as well when the drive does not contain any filesystem. The process implies formatting with a supported filesystem anyway which makes any condition obsolete. Also it is not needed anymore to unmount the filesystem manually first, as this is done automatically before formatting. In the related format menu, only supported filesystems are listed, i.e. ext4 and additionally F2FS on Raspberry Pi. The input box to enter a custom mount point is omitted as it is only temporarily used anyway until the root filesystem transfer has finished. The swap file is not disabled anymore as it can be copied and reused as is on the new root filesystem without issues.
- DietPi-Drive_Manager | On ext4 filesystems, the reserved blocks percentage is now shown and can be changed as well if the drive is not currently mounted.
- DietPi-Software | Roon Server: Added support for the new .NET core based version, released on November 3rd, which is expected to have improved performance compared to the old Mono based version. To resolve an old issue with our Roon Server implementation (see below), a reinstall is done during the DietPi update, which pulls in the .NET core dependencies as well. But this will NOT update the Roon Server version to prevent issues with potentially required migration steps. After updating DietPi, use Roon's internal updater to benefit from the new version.

#### Fixes
- DietPi-Drive_Manager | Resolved an issue where resizing F2FS filesystems failed when currently mounted. Contradicting the docs and error output, mounting the filesystem read-only is not sufficnet, but it needs to be unmounted instead, which is not done automatically before the resize and remounted afterwards.
- DietPi-Drive_Manager | Resolved an issue where resizing ext4 filesystems could have failed when not currently mounted. Especially if the drive is newly attached, a full fsck needs to be done once before it can be resized. This is now done automatically, allowing to resolve filesystem errors interactively, before resizing an offline ext4 filesystem.
- DietPi-Config | Resolved an issue where Raspberry Pi 4 overclocking profiles were offered on Raspberry Pi 400. A new set of profiles have now been added dedicated for the Raspberry Pi 400.
- DietPi-Software | IceCast: Resolved an issue where a new install failed due to an attempted operation on a non-existing file. Many thanks to @killtux for reporting this issue: https://github.com/MichaIng/DietPi/issues/4858
- DietPi-Software | Logitech Media Server: Resolved an issue where the uninstall failed as the package postinst script tried to remove the service user before the service was stopped.
- DietPi-Software | Roon Server: Resolved an issue where the internal updater purged all Roon Server data and configs, since the data directory was located within the install directory. Roon Server will now be installe [... cut ...]
- DietPi-Software | LXDE: Resolved an issue where on some cases on first desktop start, desktop icons were missing and another issue on Bullseye systems, where a "No session for pid" error message popped up on desktop start. Many thanks to @kerryland for reporting these issues: https://github.com/MichaIng/DietPi/issues/4914
- DietPi-Software | LXDE: Resolved an issue where the Firefox browser panel icon was present even if no Firefox was installed. In this case now either Chromium or the text editor is added as replacement.

JanKoudijs, 2021-11-07:
> DietPi-Software | Roon Server: Added support for the new .NET core based version, released on November 3rd

With the move to .NET the `RoonMono` directory is replaced by `RoonDotnet`. Maybe both of these directories should be listed at line 519 of `.update/patches`, to be able to make the split for both old en new installations.

MichaIng, 2021-11-07:
Makes sense, thanks for the hint 👍. I'll also test the .NET version now.

GvY85, 2021-11-09:
On a system with Wireguard, Qbitttorrent-nox, Samba and a extfat external drive no problems.

DietPi-Web Dashboard is really great! Have some remarks but will post them in https://github.com/MichaIng/DietPi/issues/448

MichaIng, 2021-11-09:
Many thanks for testing ❤️!

MichaIng, 2021-11-10:
### DietPi v7.8.1 has been pushed to the beta channel: #4945

MichaIng, 2021-11-13:
### DietPi v7.8 has been released: #4951

Many thanks to all testers ❤️!

closed 2021-11-13 by MichaIng