MichaIng/DietPi #4937: Home Assistant does not install

mokerosan, 2021-11-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.7.3
- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.10.63-v8+ #1459
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | 5V 3A
- SDcard used | SanDisk ultra

#### Steps to reproduce

1. go to dietpi-software
2. select Home Assistant
3. select install option

#### Expected behaviour

- should install Home Assistant

#### Actual behaviour

-  Checking URL: https://github.com/pyenv/pyenv/archive/master.tar.gz
                                  │  - Command: curl -ILfvm 10 https://github.com/pyenv/pyenv/archive/master.tar.gz
                                  │  - Exit code: 28
                                  │  - DietPi version: v7.7.3 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 6
                                  │  - Image creator: DietPi Core Team
                                  │  - Pre-image: From scratch
                                  │  - Error log:
                                  │   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                  │                                  Dload  Upload   Total   Spent    Left  Speed
                                  │ ^M  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0^M  0     0    0     0    0     0
                                  │ 0 --:--:-- --:--:-- --:--:--     0^M  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0^M
                                  │ 0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0^M  0     0    0     0    0     0      0      0
                                  │ --:--:--  0:00:03 --:--:--     0^M  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0*
                                  │ Trying 20.201.28.151:443...
                                  │ * Connected to github.com (20.201.28.151) port 443 (#0)
                                  │ * ALPN, offering h2
                                  │ * ALPN, offering http/1.1
                                  │ * successfully set certificate verify locations:
                                  │ *  CAfile: /etc/ssl/certs/ca-certificates.crt
                                  │ *  CApath: /etc/ssl/certs
                                  │ } [5 bytes data]
                                  │ * TLSv1.3 (OUT), TLS handshake, Client hello (1):
                                  │ } [512 bytes data]
                                  │ * TLSv1.3 (IN), TLS handshake, Server hello (2):
                                  │ { [122 bytes data]
                                  │ * TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
                                  │ { [19 bytes data]
                                  │ * TLSv1.3 (IN), TLS handshake, Certificate (11):
                                  │ { [2363 bytes data]
                                  │ * TLSv1.3 (IN), TLS handshake, CERT verify (15):
                                  │ { [79 bytes data]
                                  │ * TLSv1.3 (IN), TLS handshake, Finished (20):
                                  │ { [36 bytes data]
                                  │ * TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
                                  │ } [1 bytes data]

#### Extra details

- root@DietPi:~# dietpi-software
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | Free space check: path=/ | available=27798 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 http://ftp.us.debian.org/debian bullseye InRelease
Hit:2 http://ftp.us.debian.org/debian bullseye-updates InRelease
Hit:3 http://ftp.us.debian.org/debian bullseye-backports InRelease
Get:4 https://deb.debian.org/debian-security bullseye-security InRelease [44.1 kB]
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Fetched 44.1 kB in 2s (19.4 kB/s)
Reading package lists...
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Home Assistant: open-source home automation platform

[ INFO ] DietPi-Software | Home Assistant user:  homeassistant
[ INFO ] DietPi-Software | Home Assistant home:  /home/homeassistant
[ INFO ] DietPi-Software | pyenv activation:     ". /home/homeassistant/pyenv-activate.sh"
[ INFO ] DietPi-Software | pyenv Python version: 3.9.7
[  OK  ] DietPi-Software | Verifying user "homeassistant" to run Home Assistant
[  OK  ] DietPi-Software | usermod -aG dialout,gpio,i2c -d /home/homeassistant -s /usr/sbin/nologin homeassistant
[  OK  ] DietPi-Software | mkdir -p /home/homeassistant
[..    ] DietPi-Software | Checking URL: https://github.com/pyenv/pyenv/archive/master.tar.gz (1/2)

MichaIng, 2021-11-06:
Many thanks for your report.

The URL check timed out. You seem to have an issue with the network connection. Can you please check how long it would take:
```sh
time curl -ILf 'https://github.com/pyenv/pyenv/archive/master.tar.gz'
```

Joulinar, 2021-11-06:
I did a short test and for me this is working fine

```
[ INFO ] DietPi-Software | Home Assistant user:  homeassistant
[ INFO ] DietPi-Software | Home Assistant home:  /home/homeassistant
[ INFO ] DietPi-Software | pyenv activation:     ". /home/homeassistant/pyenv-activate.sh"
[ INFO ] DietPi-Software | pyenv Python version: 3.9.7
[  OK  ] DietPi-Software | Verifying user "homeassistant" to run Home Assistant
[  OK  ] DietPi-Software | useradd -rMU -G dialout,gpio,i2c -d /home/homeassistant -s /usr/sbin/nologin homeassistant
[  OK  ] DietPi-Software | mkdir -p /home/homeassistant
[  OK  ] DietPi-Software | Checking URL: https://github.com/pyenv/pyenv/archive/master.tar.gz
[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[ INFO ] DietPi-Software | G_THREAD_START_0 | curl -sSfL https://github.com/pyenv/pyenv/archive/master.tar.gz -o master.tar.gz
[ INFO ] DietPi-Software | APT install for: gcc libc6-dev make libssl-dev zlib1g-dev libffi-dev libbz2-dev libreadline-dev libsqlite3-dev, please wait...
```

mokerosan, 2021-11-06:
> Many thanks for your report.
> 
> The URL check timed out. You seem to have an issue with the network connection. Can you please check how long it would take:
> 
> ```shell
> time curl -ILf 'https://github.com/pyenv/pyenv/archive/master.tar.gz'
> ```

HTTP/2 302
server: GitHub.com
date: Sat, 06 Nov 2021 23:21:59 GMT
content-type: text/html; charset=utf-8
vary: X-PJAX, X-PJAX-Container, Accept-Encoding, Accept, X-Requested-With
permissions-policy: interest-cohort=()
location: https://codeload.github.com/pyenv/pyenv/tar.gz/master
cache-control: max-age=0, private
strict-transport-security: max-age=31536000; includeSubdomains; preload
x-frame-options: deny
x-content-type-options: nosniff
x-xss-protection: 0
referrer-policy: no-referrer-when-downgrade
expect-ct: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com objects-origin.githubusercontent.com www.githubstatus.com collector.githubapp.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events translator.github.com wss://alive.github.com *.actions.githubusercontent.com wss://*.actions.githubusercontent.com online.visualstudio.com/api/v1/locations raw.githubusercontent.com github-production-repository-image-32fea6.s3.amazonaws.com github-production-release-asset-2e65be.s3.amazonaws.com insights.github.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com objects-origin.githubusercontent.com; frame-ancestors 'none'; frame-src render.githubusercontent.com viewscreen.githubusercontent.com notebooks.githubusercontent.com; img-src 'self' data: github.githubassets.com identicons.github.com collector.githubapp.com github-cloud.s3.amazonaws.com secured-user-images.githubusercontent.com/ *.githubusercontent.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/
content-length: 119
x-github-request-id: 0528:3103:1BBBCA:39769D:61870E17

HTTP/2 200
access-control-allow-origin: https://render.githubusercontent.com
content-disposition: attachment; filename=pyenv-master.tar.gz
content-security-policy: default-src 'none'; style-src 'unsafe-inline'; sandbox
content-type: application/x-gzip
etag: "16ed85fe5bd114acc2f263efafb9f5fdb66f89d54eb457dfbb71fc310261e9ef"
strict-transport-security: max-age=31536000
vary: Authorization,Accept-Encoding,Origin
x-content-type-options: nosniff
x-frame-options: deny
x-xss-protection: 1; mode=block
date: Sat, 06 Nov 2021 23:22:05 GMT
x-github-request-id: 04D4:7BC7:15575:D036E:61870E1D

real    0m11,879s
user    0m0,134s
sys     0m0,052s

Joulinar, 2021-11-06:
uhh nearly 12 seconds. That's huge. Usually it should be done in less than 0.5 sec

You could try to adjust time out value inside `/boot/dietpi.txt`. Value could be set to `20`

```
# Connection timeout in seconds, before G_CHECK_URL assumes the URL is dead. Increase if you have a "flaky" connection or slow DNS resolver.
# - Set this to "0" to allow unlimited time, however this is not recommended to avoid unlimited hanging background scripts, e.g. daily DietPi update check.
# - A negative or non-integer value will result in the default of 10 seconds.
CONFIG_G_CHECK_URL_TIMEOUT=10
```

MichaIng, 2021-11-06:
IPv6 cannot be the issue as `github.com` and `codeload.github.com` have A (IPv4) records only. Probably it's temporary. I guess you don't have such long delay when accessing GitHub here via browser, right?

MichaIng, 2021-11-24:
I'll mark this as closed. Feel free to reopen when required.

closed 2021-11-24 by MichaIng