MichaIng/DietPi #4938: Locked out from system (ssh)

Dr0bac, 2021-11-07:
Hi, i'm connecting to my raspberry pi with pc and putty. Yesterday i reinstalled windows and when i tried to connect to dietpi with default

username:`root`
password:`dietpi`

I figured it out that i changed default password in `dietpi.txt` when i was installing dietpi few years ago. My putty was set to automatically enter username and password so i forgot completely about it.

Is there a way to recover/change password or i must install new dietpi?

MichaIng, 2021-11-07:
If you don't know any login password, you need to hack a script into the RPI's SD card/root filesystem to change it at boot.

Do you have another Linux system where you can mount ext4 filesystem? Otherwise e.g. booting a second DietPi via VirtualBox on Windows or macOS will work as well. The RPi drive can be attached and passed through to the virtual machine where you can mount it and hack mentioned script inside. Is it an SD card or external/USB drive you have your root filesystem stored?

Let me know which method is feasible, then we can provide you with further details.

Dr0bac, 2021-11-07:
I was using root user. Can i do something with dietpi user?

I can install virtualbox on windows. My dietpi is installed on sd card.

Joulinar, 2021-11-07:
are you able to login via SSH using user `dietpi`?

Dr0bac, 2021-11-07:
I don't know what is default password for dietpi user?

Joulinar, 2021-11-07:
did you tried `dietpi` as password?

MichaIng, 2021-11-07:
If you changed the password on first login or via `dietpi-config`, then `root` and `dietpi` share the same. But at least worth to give it a try via SSH and `dietpi` user and then the default password.

Otherwise VirtualBox should be the easiest option: https://dietpi.com/downloads/images/DietPi_VirtualBox-x86_64-Bullseye.7z

`/mnt` or any media service access is not sufficient, you need root access to `/var` or `/etc`.

When you plug the SD card into Windows, the following batch script can be used to create a VMDK file which you can attach to the VM as regular HDD:
```batch
@echo off
"C:\Program Files\Oracle\VirtualBox\VBoxManage" internalcommands createrawvmdk -filename "C:\Users\Micha\VirtualMachines\sdcard.vmdk" -rawdisk "\\.\PHYSICALDRIVE1"
mountvol e: /d
pause
```
Change the VMDK's path to a feasible one and the drive number can be received by running in `cmd`:
```
diskpart
list disk
```
Probably to start the VM it is then necessary to unmount the FAT partition of the SD card from windows via it's drive manager.

When finally mounted in the VM, run the following command to create the script which changes the root password:
```sh
cat << '_EOF_' > /var/lib/dietpi/postboot.d/change_password.sh
{
# Change root password to "password"
chpasswd <<< 'root:password'
# Make the script remove itself
rm /var/lib/dietpi/postboot.d/change_password.sh
exit 0
}
_EOF_
```

Joulinar, 2021-11-07:
@MichaIng 
probably another option https://howtoraspberrypi.com/recover-password-raspberry-pi/

MichaIng, 2021-11-07:
Ah right, I totally forgot about this option. Okay that is much easier as you don't need the VM or ext4 partition at all. But it requires a screen and keyboard attached to the RPi.

To make it even easier, it should work to run the above mentioned script directly via `init`, hence create a script right in the FAT partition (where `cmdline.txt` is located):
```sh
#!/bin/bash
{
# Mount /boot
mount /dev/mmcblk0p1 /boot
# Change root password to "password"
chpasswd <<< 'root:password'
exit 0
}
```
And then use in `cmdline.txt`:
```
init=/boot/change_password.sh
```
Then boot the RPi and wait a while. Turn it off again and revert the changes.

Dr0bac, 2021-11-07:
I managed to change the password with solution that Joulinar provided.

My only mistake was, that i didn't attached my dietpi_userdata drive to it when i was changing the password. I thought that after the line

`sync`
`exec /sbin/init`

Raspberry pi will shutdown, but instead it restarted the system, and i got errors that dietpi_userdata is not attached and swapfile is not present. And system i guess froze after messages about starting services (bazarr, radarr, etc). And than i pull out the power cable from raspberry pi.

Did i messed up something because dietpi_userdata hdd wasn't connected?

MichaIng, 2021-11-07:
No worries, you didn't mess up anything. Since the swap file was not mounted, parts of the boot sequence were skipped and as you said not all services could be started, so also pulling the power cable most likely did not cause any damage and an fsck will run automatically on next boot.

Instead of `exec /sbin/init`, `poweroff` is probably the safer command to end the steps. Also the explanation what will happen does not match, as `exec /sbin/init` runs the init system, hence starts it and will not "go out" 🤔.

Dr0bac, 2021-11-07:
Excellent. Thank you both for helping me.

closed 2021-11-07 by Dr0bac