MichaIng/DietPi #4941: Jellyfin

Sabering1, 2021-11-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=0
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'

- Distro version | bullseye 1

- Kernel version | Linux DietPi 5.10.63+ #1459 Wed Oct 6 16:40:27 BST 2021 armv6l GNU/Linux

- SBC model | RPi Zero W (armv6l)
- Power supply used | Standard USB A to micro USB cable
- SDcard used | SanDisk

#### Additional Information (if applicable)
- Software title | Jellyfin
Jellyfin is listed in the official dietpi website under Media Servers
https://dietpi.com/docs/software/media/#jellyfin
However its not available for installation with dietpi-software or apt

#### Steps to reproduce

1. ... Go to dietpi-software
2. ... Type Jellyfin

#### Expected behaviour

- ... Software should be listed there

#### Actual behavior

- ...Its not listed

#### Extra details

- ... Add software or remove it from the website

Joulinar, 2021-11-07:
Probably disabled for your device. Could be checked as follow

`dietpi-software list | grep jell`

MichaIng, 2021-11-07:
Not every software option is available on every device/architecture/distro version. Run the following to see which ones are disabled in your case:
```sh
dietpi-software list | grep DISABLED
```
In case of Jellyfin, it is disabled for ARMv6 RPi models (RPi 1 and Zero), as there are simply no binaries provided for this architecture. And it isn't possible since the underlying .NET runtime does not support ARMv6: https://github.com/dotnet/runtime/issues/7764

And aside of that, you won't be able to have any joy with Jellyfin on a Raspberry Pi Zero W. This is not a multi media RPi model but a typical headless server non-GUI lower power device 😉.

Sabering1, 2021-11-07:
Oh i didn't know that!
Thanks for the quick response!

MichaIng, 2021-11-08:
I'll mark this as closed then. Feel free to reopen if required.

closed 2021-11-08 by MichaIng