MichaIng/DietPi #4943: Portainer installation problems

deckspace, 2021-11-08:
Hiya Micha, 
I am seeing this on my 32bit Dietpi install when attempting to install Portainer alongside Docker from within Dietpi-software 
Seems to have become an issue since recent update of DietPi -  am checking other instances where we have working versions to gather more info but perhaps you have comment in the meantime. thanks

#### Details:
- Date           | Mon Nov  8 10:44:11 UTC 2021
- DietPi version | v7.7.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux bxhsdiet 4.19.49v6v7-aufs #1 SMP Tue Jun 11 15:13:27 CEST 2019 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `docker run -d -p 9002:9000 --name=portainer --restart=always -v /run/docker.sock:/var/run/docker.sock -v portainer_data:/data portainer/portainer-ce`
- Exit code      | 125
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Unable to find image 'portainer/portainer-ce:latest' locally
latest: Pulling from portainer/portainer-ce
7721cab3d696: Pulling fs layer
0645e7e2a110: Pulling fs layer
bfbb644d4a9f: Pulling fs layer
0645e7e2a110: Verifying Checksum
0645e7e2a110: Download complete
7721cab3d696: Verifying Checksum
7721cab3d696: Download complete
7721cab3d696: Pull complete
0645e7e2a110: Pull complete
bfbb644d4a9f: Verifying Checksum
bfbb644d4a9f: Download complete
bfbb644d4a9f: Pull complete
Digest: sha256:af387baba14e0342e40d274c0c894fd333d3cca0d6737a8e1e0d6d9523c87a8a
Status: Downloaded newer image for portainer/portainer-ce:latest
70895397a87626aef2607a3a5f3c4f78ceedf50d0ff7f2356486acd283ebb940
docker: Error response from daemon: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: process_linux.go:508: setting cgroup config for procHooks process caused: bpf_prog_query(BPF_CGROUP_DEVICE) failed: function not implemented: unknown.
```

Joulinar, 2021-11-08:
thats interesting. According your information you are running Debian Bullseye but with a very old kernel `4.19.49v6v7`. Are you using any custom kernel or probably berryboot system?

MichaIng, 2021-11-08:
Exactly. And Docker reports quite clearly that the kernel feature `BPF_CGROUP_DEVICE` is missing which is not with the official Raspberry Pi kernel that (our) DietPi images ship with.

On Berryboot, the kernel cannot be upgraded the regular way via APT package and I'm not aware of any internal updater either. Furthermore the RPi 1/2/3 images have not been updated anymore after the RPi 4 release, so probably a new Berryboot kernel for those models has never been built.

I like the idea behind Berryboot, but it has its implied downsides, and additionally Berryboot is developed very slowly, very unresponsive on GitHub and seems to intentionally only provide new images for the latest RPi model. I hence wouldn't recommend to use it as long as you don't have strong reasons to do so. For "multi-boot", on an RPi wrapping the SD card or USB drive is a much better alternative if you have physical access to it.

deckspace, 2021-11-09:
Thanks for the comments, yes this one is running berryboot so that explains the issue.. what a drag! Our non berryboot installations run fine with Portainer and docker I had no idea the kerne was shared in the way described as that's a show stopper.

MichaIng, 2021-11-09:
Yes, it is probably a problem specific to the RPi, at least with the official bootloader: Other than common bootloaders like GRUB or U-Boot, it is not flashed to the start of the drive, outside of any filesystem, from where it can load different partitions with kernel, device tree and in case initramfs, but it is (on models prior to RPi 4, where it is located at the EEPROM of the board) a single closed source binary file `bootcode.bin` loaded on the first FAT partition of the boot drive, which loads kernel, device tree and further bootloader files from the same partition. Probably it could be done differently, but to Berryboot makes use of the official bootloader blob and kernel (sources) and an initramfs to show its OS selection menu and then only the userland as overlay from the selected OS. This works, but you then cannot update the kernel/firmware from the selected OS, even if you would mount the original actual Berryboot partition, due to its custom kernel, initramfs, bootloader and boot/firmware configuration it depends on.

Classic bootloaders are self-contained and can show an OS selection menu without loading a dedicated kernel, device tree and initramfs first, but loading those from the OS/partition you select. So aside of the bootloader itself, you can fully configure any update the kernel for each OS individually from within the OS. On RPi this is of further importance, since the firmware can be configured, features enabled/disabled significantly via `config.txt` and device tree overlays, loaded by the bootloader. Without direct access to it, the options you have from within the loaded OS is pretty limited. And due to the low update rate of Berryboot, new kernel features and fixes are provided with a large delay, compared to using the official kernel packages, or well maintained ones from other distributions.

Btw, there is an update available with Linux 5.10, which may contain the required kernel feature for Docker/Portainer: https://www.berryterminal.com/doku.php/berryboot
Other than the filename says, it works fine on RPi 3, only earlier RPi models are not supported anymore (strangely the `config.txt` still contains the reference to the RPi 1/2 compatible kernel image, but that image is not present anymore). According to [this](https://raspberrypi.stackexchange.com/questions/75348/upgrading-berryboot-version) you should be able to manually replace the old files with the new ones.

MichaIng, 2021-11-11:
I'll mark this as closed. Feel free to reopen if any questions are left.

closed 2021-11-11 by MichaIng