MichaIng/DietPi #4944: DietPi-Software | Deluge fails on ARMv6

MichaIng, 2021-11-08:
Spit from original report: #4785
Reported at Raspbian: https://bugs.launchpad.net/raspbian/+bug/1945530

MichaIng, 2021-11-09:
Hmm, shall we disable it on Raspbian Bullseye until it gets fixed? As a workaround, on ARMv7+ the Debian packages could be tried: https://packages.debian.org/bullseye/deluged
But I'm not keen to implement this hassle into our code 🤔.

Joulinar, 2021-11-10:
let's disable if it is to complicated.

MichaIng, 2021-11-10:
Would be just bad if it got fixed right after our release 😄. Zero response at our bug report, but I hope that this issue gets more attention/reports now since Raspberry Pi OS Bullseye was released yesterday. Same for our reports at the RPi repo branch for RealVNC and CEC. Looks like the official release was no guarantee that all software works, as those are still broken and Kodi is still missing 🙄.

MichaIng, 2021-11-12:
Disabled for Raspbian Bullseye systems: https://github.com/MichaIng/DietPi/commit/74e0e73c800543c343acfd4f2d5dbc9a4a178563

themagicbullet, 2021-11-25:
I'm pretty sure that the problem that's occurring is that the build flags that were used to build v1.2.9 of libtorrent didn't manually link libatomic (see arvidn/libtorrent#4657 and arvidn/libtorrent#5117). While waiting for the libtorrent-rasterbar package to be updated to a version that is built with these flags, you can use the [`LD_PRELOAD` trick](https://www.baeldung.com/linux/ld_preload-trick-what-is) to manually cause the dynamic loader to load the libatomic library.

I don't use DietPi so I'm not sure how you use Deluge, but the basic idea is that you need to run `export LD_PRELOAD` with the path to the library and then call `/usr/bin/deluged -d`. In my case, I use a systemd service so instead of executing the `/usr/bin/deluged` directly, I instead have the service execute a Bash script that looks like this: 

```shell
#!/usr/bin/bash
export LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1
/usr/bin/deluged -d
```

MichaIng, 2021-11-25:
Great find, many thanks for sharing ❤️! I'll test it and in case apply it to our systemd unit. Would you mind to add this info to the Raspbian but report? https://bugs.launchpad.net/raspbian/+bug/1945530

I can also try to collect it from the issues you linked, but better to have it added by you being closer involved.

Gurky-Kronos, 2021-12-06:
Is there a way to fix this on my end while waiting?

MichaIng, 2021-12-06:
Thanks for the reminder, I missed to test and fix this for the upcoming release. I just verified that the solution provided by @themagicbullet works well and applied it hence to `dietpi-software` to be shipped with next beta iteration and DietPi v7.9. You can apply the fix manually by running these commands: https://github.com/MichaIng/DietPi/commit/aefd867
```sh
G_CONFIG_INJECT 'Environment=' 'Environment=LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1' /etc/systemd/system/deluged.service '\[Service\]'
systemctl daemon-reload
systemctl restart deluged
```
And I guess if you upgraded from Buster to Bullseye, you also need to fix the web service, which requires an additional flah:
```sh
sed -i '/^ExecStart/s/deluge-web$/deluge-web -d/' /etc/systemd/system/deluge-web.service
systemctl daemon-reload
systemctl restart deluge-web
```

Gurky-Kronos, 2021-12-06:
> You can apply the fix manually by running these commands: aefd867

That worked, Thank you

MichaIng, 2021-12-06:
Great, thanks for the feedback. I mark the issue as closed then. When this got fixed Raspbian wise, I should get a notification from the bug report.

closed 2021-12-06 by MichaIng

MichaIng, 2021-12-07:
Great, it has been fixed package wise now 👍. The workaround can hence be reverted, but it doesn't cause any harm to leave it in place.