MichaIng/DietPi #4947: Install phones-home even if opted out

kevcmk, 2021-11-10:
If a user is opted out, the install should not phone home.

https://github.com/MichaIng/DietPi/blob/92a41a9c44c9e1f45a78f4029c66ee9766b89cfd/dietpi/dietpi-survey#L106

Joulinar, 2021-11-10:
This is uploading an empty file only to override all existing files on our server. no user data are stored at this point.

kevcmk, 2021-11-10:
That's clear. But why would anything be uploaded in the first place?

ravenclaw900, 2021-11-10:
It's uploaded to overwrite any previously sent data, like if the user was using survey before.

kevcmk, 2021-11-10:
Deleting already submitted survey data is a separate (and, from my perspective, nearly useless [?]) scenario.

If a user explicitly opts out, why would the install process phone anything home in the first place?￼

MichaIng, 2021-11-10:
It is a simple opt-out + delete combination, to not have three options there. Also it helps us to know how many user are opted in and out and estimate an overall user count. If you have a problem with an empty file uploaded to our server after a DietPi update and a dietpi-software installation, you could either block `ssh.dietpi.com` or remove `/boot/dietpi/dietpi-survey`. Any failure with the script call or upload won't break the parent update/install. A very simple effective solution:
```sh
echo '0.0.0.0 ssh.dietpi.com' >> /etc/hosts
```

I'm all in when its about preventing software spying on their users, I'm actively blocking all unnecessary home calls from Windows that I know about and which do not break essential things, I use custom Android ROMs to avoid Google services, use slim open source software where I can, block commercial trackers when browsing the web, though it is practically impossible to block every unwanted request if you want to use the internet in a reasonable way. Not sure if you're aware how much data you are uploading to various commercial 3rd parties on every single website load you do, also here on GitHub (AWS cloud at least)? A direct single empty SFTP upload to an open source project server without any answer or any 3rd party involved, shouldn't be something you worry about. Just to bring things into perspective 😉.

kevcmk, 2021-11-10:
Nothing personal, but I suspect that this is a violation of both GDPR and CCPA.

MichaIng, 2021-11-10:
I'm not sure what you are trying to achieve here or what your actual problem is. Every DietPi (or any other) update or software install naturally involves a whole bunch of request WITH data to a bunch of different servers to resolve hostnames, sync the system time, check the current DietPi version, download it's files, respectively update the APT lists from Debian/Raspbian/RPi foundation/Radxa/Armbian, in case download the packages, etc etc. Why are you complaining about a single empty request?

There is not a single bit of private data involved in this request, so it does not violate any law of user data privacy. And we do zero logging on our server, if this is for some reason something you worry about, even that you'd need to worry about this for every single other request. Though zero logging this is something you need to trust us, like you need to trust every other single server you send any request to about this, respectively accept that your requests with your public IP can always be logged by every server.

If you would explain your actual concern, I mean the actual final concern, not "I don't like that a request is done", please explain, as I don't see any problem here.

kevcmk, 2021-11-11:
```
DietPi-Survey would like to collect anonymous usage statistics.
This allows us to focus development based on popularity.
 - NO private data will be collected and NO ONE can identify you based on it.
 - The data is sent via secured SFTP connection, stored unreadable to the public upload user.
 - If you agree, your uploaded data will be updated after each DietPi-Update and DietPi-Software install.
 - Current survey statistics can be reviewed at: https://dietpi.com/survey/
\nWould you like to join DietPi-Survey?
```

Makes no mention of "We'll send your data regardless, but we'll later on 'delete' it if you've opted out". For someone so privacy conscious as you have specified, I'm surprised you find my asking about this to be so surprising.

MichaIng, 2021-11-11:
> "We'll send your data regardless, but we'll later on 'delete' it if you've opted out"

This would be a false info, as no data is send but only directly deleted if any there. Again, uploaded is an empty file that overwrites the existing file, hence existing data, if that file was not empty already. And no data is initially uploaded as long as you do not manually select "Opt IN".

kevcmk, 2021-11-11:
Ok, clear that you're dumping an empty file. But still seriously concerning that DietPi install phones-home even if the user opts out. As you've described yourself, I'm of the same mind on data privacy; if you're like me, "trust us we don't log" doesn't hold water. Is there any reason this couldn't be arranged to only invoke Send_File if the user isn't opted out?

MichaIng, 2021-11-11:
Data privacy, or better the choice whom you want to share your data with is important, no question, but as no data is sent, it simply does not apply here. If you really worry about your public IP being logged for whatever reason by servers, which is the only info we are able to get from this request, then you need to use a VPN for each and every internet request you do, so that at least only the VPN provider gets your IP. If you do not use a VPN for everything, then there is not argument left why this request would further violent your privacy.

What you request was the case before DietPi v6.9 and was changed to the way it is now with the v6.10: https://github.com/MichaIng/DietPi/commit/055563094aca775a5a48c4c6d05da71a048aed2f#

This could be reverted, but it would break our ability to know about the overall user count: https://dietpi.com/survey/
While this is not perfectly accurate, it is good for a small project like use to have a reference about an overall user number, so that e.g. SBC manufacturers can get an idea whether its worth to provide us with developer samples, list DietPi on their wiki/software/download pages etc, something where all our users have benefit from. Also the ratio get know how well DietPi-Survey is accepted is valuable to us, so we know whether we need to change something about it, and can monitor the effect of changes, and also the effect when advertising the survey. And the survey itself is very valuable to us to estimate which software titles and SBCs are well used and which one not so well, e.g. we remove Subsonic and emonHub and Odroid N1 support with next release, as Subsonic would be a pain to further maintain, emonHub and Odroid N1 had both zero users, the Odroid as it was never publicly release but only a few developer samples are floating around. This way we can keep our maintenance efforts in borders while adding support for new software titles and new SBCs. Without the survey + an estimate how accurate statistically the counts are (taking into account optin/optout users, statistically), we could be much less certain whether dropping support for something would hurt too many users or not.

So no, as long as no one provides us with an argument why an empty ping hurts ones privacy at the end of an update/install process that involves a bunch of non-empty requests and downloads, I'm not keen to loose the above mentioned benefits.

MichaIng, 2021-11-11:
Btw, do not take my statements as offence. I'm actually happy that this topic comes up here, as it did a while back internally and we asked us exactly the same question. I understand that there are some principles behind it, and the generally healthy scepticism when any request is sent that was not explicitly requested (compared to a software download or update check when requesting a software install or update).

What I want to make clear here is that it is a difference if closed source software from a tech giant is sending various collected information from your system back home, without any transparency and limited end user control, which is in theory able to create user profiles due to combined data from various services, connected with actual user accounts, probably search machines, ads and CDN related cross-origin requests etc, so that a single request with an IP can contribute to a user profile that can be associated to an actual real person, compared to a request which provides the IP only to a project like us, which has no ability and no aim to associate this IP with timestamp to any other information. Since HTTP requests are passed through Cloudflare, we are not even able to associate it with forum accounts or blog comments.

So as it has some benefits for us how it is, before changing things, I need a convincing argument why this empty SFTP upload does or can indeed violate our users privacy, if we would be able to misuse it in any way, or if it violates any related law anywhere in the world. I did think and research quite some time about this but couldn't find more than a subjective negative feeling about a network request that was not explicitly requested without factual concerns. But I may be wrong.

MichaIng, 2022-02-06:
Marking this as closed. However, anyone is invited to continue/contribute to this discussion, and we'll reopen the issue then. I'm generally open for factual arguments and want to handle this transparently, but currently it see no factual harm to users privacy but a real downside of our project, when removing the empty file upload.

closed 2022-02-06 by MichaIng