MichaIng/DietPi #4949: dietpi-software failed to install ffmpeg

juarezpedra, 2021-11-10:
#### Details:
- Date           | Wed Nov 10 19:50:07 -03 2021
- DietPi version | v7.7.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux bob 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux`
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | `apt-get -qq --allow-change-held-packages install ffmpeg`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. sudo dietpi-software install 7

#### Expected behaviour:
Install ffmpeg
- ...
#### Actual behaviour:
Exit the installation before ends 
- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch http://raspbian.c3sl.ufpr.br/raspbian/pool/main/libt/libtheora/libtheora0_1.1.1+dfsg.1-14_armhf.deb  Connection failed
E: Failed to fetch http://raspbian.c3sl.ufpr.br/raspbian/pool/main/x/x264/libx264-148_0.148.2748+git97eaef2-1+rpi1_armhf.deb  Connection failed
E: Failed to fetch http://raspbian.c3sl.ufpr.br/raspbian/pool/main/f/ffmpeg/libavcodec57_3.2.15-0+deb9u4_armhf.deb  Connection failed
E: Failed to fetch http://raspbian.c3sl.ufpr.br/raspbian/pool/main/libo/libopenmpt/libopenmpt0_0.2.7386~beta20.3-3+deb9u4_armhf.deb  Connection failed
E: Failed to fetch http://raspbian.c3sl.ufpr.br/raspbian/pool/main/f/ffmpeg/libavformat57_3.2.15-0+deb9u4_armhf.deb  Connection failed
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

MichaIng, 2021-11-10:
The connection to the Raspbian mirror failed. Did you choose this manually or was it the result of the mirror director? I.e. can you show:
```sh
cat /etc/apt/sources.list
```
The mirror works fine here though. Probably it's a temporary connection or server issue only and a simple retry may work. You can also test the connection manually:
```sh
curl -IL 'http://raspbian.c3sl.ufpr.br/raspbian/'
```

MichaIng, 2021-11-24:
I'll mark this as closed. Feel free to reopen if the issue persists.

closed 2021-11-24 by MichaIng

juarezpedra, 2021-11-25:
Sorry for the delay, I will test again as soon as I get home. Thanks.