MichaIng/DietPi #4952: Command        | `ping -c 1 -W 10 9.9.9.9` - Exit code      | 2

seafile007, 2021-11-11:
#### Details:
- Date           | Thu Nov 11 22:58:13 GMT 2021
- DietPi version | v7.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi B (armv6l) (ID=0)
- Kernel version | `Linux RaspPiDietPiB 5.10.52+ #1441 Tue Aug 3 18:08:18 BST 2021 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `ping -c 1 -W 10 9.9.9.9`
- Exit code      | 2
- Software title | DietPi-Login
#### Steps to reproduce:

1. remained after reboot
2. on rescue shell ping is fine to 9.9.9.9
#### Expected behaviour:

- errors when dietpi-config
#### Actual behaviour:

- ...
#### Extra details:

- new installed on Raspi Version B
#### Additional logs:
```
ping: connect: Network is reachable
root@RaspPiDietPiB:~# ping 9.9.9.9
PING 9.9.9.9 (9.9.9.9) 56(84) bytes of data.
64 bytes from 9.9.9.9: icmp_seq=1 ttl=59 time=32.9 ms
64 bytes from 9.9.9.9: icmp_seq=2 ttl=59 time=33.4 ms
64 bytes from 9.9.9.9: icmp_seq=3 ttl=59 time=31.2 ms
64 bytes from 9.9.9.9: icmp_seq=4 ttl=59 time=31.4 ms
64 bytes from 9.9.9.9: icmp_seq=5 ttl=59 time=30.0 ms

Joulinar, 2021-11-11:
at which point you have that issue?

seafile007, 2021-11-11:
hi, thanks for comment
after first boot from new dietpi sd card start

seafile007, 2021-11-11:
now rebooting with this this after ssh login:
Checking network connectivity                                                                                 │
│  - Command: ping -c 1 -W 10 9.9.9.9                                                                           │
│  - Exit code: 2                                                                                               │
│  - DietPi version: v7.4.2 (MichaIng/master) | HW_MODEL: 0 | HW_ARCH: 1 | DISTRO: 6                            │
│  - Image creator: DietPi Core Team                                                                            │
│  - Pre-image: from scratch                                                                                    │
│  - Error log:                                                                                                 │
│ ping: connect: Network is unreachable                                                                         │
│                                                                                                               │
│                      Retry          : Re-run the last command that failed                                     │
│                      DietPi-Config  : Edit network, APT/NTP mirror settings etc                               │
│                      Open subshell  : Open a subshell to investigate or solve the issue                       │
│                      Send report    : Uploads bugreport containing system info to DietPi                      │
│                                     ●─ Devs only ──────────────────────────────────────●                      │
│                      Change command : Adjust and rerun the command                                            │
│                                                                                                               │
│                                                                                                               │
│                                <Ok>                                    <Exit>                                 │
│

Joulinar, 2021-11-11:
can you go to subshell and try

`ping -c 1 -W 10 9.9.9.9`

seafile007, 2021-11-11:
from subshell ping is fine:
 DietPi v7.4.2 : 23:13 - Thu 11/11/21
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.178.130 (eth0)
[FAILED] DietPi-Login | Checking network connectivity
 - Command: ping -c 1 -W 10 9.9.9.9
root@RaspPiDietPiB:~# ping -c 1 -W 10 9.9.9.9
PING 9.9.9.9 (9.9.9.9) 56(84) bytes of data.
64 bytes from 9.9.9.9: icmp_seq=1 ttl=59 time=32.2 ms

--- 9.9.9.9 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 32.171/32.171/32.171/0.000 ms
root@RaspPiDietPiB:~# ping -c 3 -W 10 9.9.9.9
PING 9.9.9.9 (9.9.9.9) 56(84) bytes of data.
64 bytes from 9.9.9.9: icmp_seq=1 ttl=59 time=31.2 ms
64 bytes from 9.9.9.9: icmp_seq=2 ttl=59 time=37.0 ms
64 bytes from 9.9.9.9: icmp_seq=3 ttl=59 time=31.7 ms

--- 9.9.9.9 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2004ms
rtt min/avg/max/mdev = 31.175/33.293/36.988/2.621 ms
root@RaspPiDietPiB:~#

Joulinar, 2021-11-11:
and if you go back and retry?

seafile007, 2021-11-11:
any idea? thanks

root@RaspPiDietPiB:~# exit
exit
[  OK  ] DietPi-Login | Checking network connectivity
[  OK  ] DietPi-Login | Checking DNS resolver
[  OK  ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (1/60)
[  OK  ] DietPi-Run_NTPD | systemd-timesyncd synced
[  OK  ] Network time sync | Completed

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
curl: (6) Could not resolve host: raw.githubusercontent.com
[FAILED] DietPi-Update | Unable to get latest version.
[ INFO ] DietPi-Update | Please check the target Git repository and your network connection, then rerun dietpi-update.
[ INFO ] DietPi-Update | If this error persists, please report at: https://github.com/MichaIng/DietPi/issues
[ INFO ] DietPi-Update | APT update, please wait...
Err:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
  Could not resolve 'raspbian.raspberrypi.org'
Err:2 https://archive.raspberrypi.org/debian bullseye InRelease
  Could not resolve 'archive.raspberrypi.org'
Reading package lists...
W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease  Could not resolve 'raspbian.raspberrypi.org'
W: Failed to fetch https://archive.raspberrypi.org/debian/dists/bullseye/InRelease  Could not resolve 'archive.raspberrypi.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
[FAILED] DietPi-Update | APT update

seafile007, 2021-11-11:
root@RaspPiDietPiB:/tmp/DietPi-Update# ping archive.raspberrypi.org
PING lb.raspberrypi.org (46.235.231.151) 56(84) bytes of data.
64 bytes from 46.235.231.151 (46.235.231.151): icmp_seq=1 ttl=54 time=32.5 ms
64 bytes from 46.235.231.151 (46.235.231.151): icmp_seq=2 ttl=54 time=31.7 ms
64 bytes from 46.235.231.151 (46.235.231.151): icmp_seq=3 ttl=54 time=34.5 ms

seafile007, 2021-11-11:
root@RaspPiDietPiB:/tmp/DietPi-Update# ping raspbian.raspberrypi.org
PING mirrordirector.raspbian.org (93.93.128.193) 56(84) bytes of data.
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=1 ttl=54 time=41.3 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=2 ttl=54 time=40.6 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=3 ttl=54 time=39.8 ms

Joulinar, 2021-11-11:
strange that the system is failing in first place.

seafile007, 2021-11-11:
I've never had that with DietPi.
I known DietPi is very great!
But now I have no idea.

Joulinar, 2021-11-11:
if you are on command line, can you do `apt update && apt upgrade`

MichaIng, 2021-11-11:
Looks like the network is somehow flaky. First it fails, then it work, same with ping and with resolving both hosts. Which DNS server do you use? Default `9.9.9.9`?
```sh
cat /etc/resolv.conf
```
And how much time does it take to ping + resolve a new host?
```sh
time ping -c 1 dietpi.com
time ping -c 1 wikipedia.org
```

seafile007, 2021-11-11:
> if you are on command line, can you do `apt update && apt upgrade`

many errors like this:
W: Problem unlinking the file /var/cache/apt/archives/partial/libraspberrypi-bin_1%3a2+git20210928~141951+6e8f786-2_armhf.deb - PrepareFiles (5: Input/output error)
W: Problem unlinking the file /var/cache/apt/archives/partial/libraspberrypi0_1%3a2+git20210928~141951+6e8f786-2_armhf.deb - PrepareFiles (5: Input/output error)
W: Problem unlinking the file /var/cache/apt/archives/partial/raspberrypi-bootloader_1%3a1.20211029-1_armhf.deb - PrepareFiles (5: Input/output error)
W: Problem unlinking the file /var/cache/apt/archives/partial/raspberrypi-sys-mods_20211005+bullseye_armhf.deb - PrepareFiles (5: Input/output error)
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-brcm80211_20210315-3%2brpt3_all.deb  Could not open file /var/cache/apt/archives/partial/firmware-brcm80211_1%3a20210315-3+rpt3_all.deb - open (30: Read-only file system) [IP: 46.235.231.145 443]
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-iwlwifi_20210315-3%2brpt3_all.deb  System error resolving 'archive.raspberrypi.org:443' - getaddrinfo (5: Input/output error) [IP: 46.235.231.145 443]
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-misc-nonfree_20210315-3%2brpt3_all.deb  System error resolving 'archive.raspberrypi.org:443' - getaddrinfo (5: Input/output error) [IP: 46.235.231.145 443]

seafile007, 2021-11-11:
> ```shell
> time ping -c 1 wikipedia.org
> ```

no problems:
root@RaspPiDietPiB:/tmp/DietPi-Update# time ping -c 1 dietpi.com
bash: /usr/bin/ping: Input/output error

real	0m0.074s
user	0m0.000s
sys	0m0.060s
root@RaspPiDietPiB:/tmp/DietPi-Update# time ping -c 1 wikipedia.org
bash: /usr/bin/ping: Input/output error

real	0m0.071s
user	0m0.000s
sys	0m0.060s
root@RaspPiDietPiB:/tmp/DietPi-Update#

MichaIng, 2021-11-11:
This looks like filesystem errors. Can you check:
```sh
dmesg -l emerg,alert,crit,err
```

Joulinar, 2021-11-11:
jup the is a `Read-only file system`. Probably a faulty SD card?

seafile007, 2021-11-11:
Can I change something on the SD card on my desktop again to start from the beginning?
Can I change the boot.txt or config.txt file?

> ```shell
> dmesg -l emerg,alert,crit,err
> ```

dmesg not found on RasPi

Joulinar, 2021-11-11:
did you tried to download a new image from our side and flash the image again.

seafile007, 2021-11-11:
> did you tried to download a new image from our side and flash the image again.

i will try now

seafile007, 2021-11-12:
> did you tried to download a new image from our side and flash the image again.

I have done that.
Still not working. I'll check tomorrow with another SD card.
Thanks a lot guys for your "live" support.
Best regards from Germany.

seafile007, 2021-11-14:
DietPi is running again :-)
With a new SD card everything works now.
Thank you all!

closed 2021-11-14 by seafile007