MichaIng/DietPi #4956: Docker Install error while installing Roon Extension

Rickyrivits, 2021-11-14:
---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Sun Nov 14 12:31:58 GMT 2021
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -qq --allow-change-held-packages install xserver-xorg-core xserver-xorg-input-libinput xinit dbus-x11 xfonts-base x11-xserver-utils x11-utils libgl1-mesa-dri mesa-utils mesa-utils-extra xserver-xorg-video-fbdev`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Setting up docker-ce (5:20.10.10~3-0~debian-bullseye) ...
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
invoke-rc.d: initscript docker, action "start" failed.
● docker.service - Docker Application Container Engine
     Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Sun 2021-11-14 12:31:58 GMT; 26ms ago
TriggeredBy: ● docker.socket
       Docs: https://docs.docker.com
    Process: 30765 ExecStart=/usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock (code=exited, status=1/FAILURE)
   Main PID: 30765 (code=exited, status=1/FAILURE)
        CPU: 479ms
dpkg: error processing package docker-ce (--configure):
 installed docker-ce package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 docker-ce
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

MichaIng, 2021-11-14:
Many thanks for your report.

I see you had a kernel upgrade but didn't reboot yet, hence loaded kernel and installed modules do not match. Please `reboot` and retry.

Rickyrivits, 2021-11-14:
I updated dietpi using Bullseye_AlloGUI.img
Then installed a few software packages using dietpi-software
When i installed 86 and 162 the software failed
Later I tried to install 113 and this failed. 
At this point I copied the aforementioned data and stopped.

MichaIng, 2021-11-14:
I just checked the code, since we already have a check for the kernel match to prevent a Docker installation in the first place and inform about a reboot: https://github.com/MichaIng/DietPi/blob/0c59b07/dietpi/dietpi-software#L2182-L2194

But I see this is only done if Docker was explicitly selected for installation. If you e.g. select only Roon Extension Manager, then Docker is selected automatically as dependency, but after this kernel check is done. But we also cannot do the check afterwards, else we could install obsolete dependencies. I have to think about a logic which covers all cases cleanly.

Rickyrivits, 2021-11-14:
Thanks, The software installed after I re-booted. I have a roon problem but not your concern. Thanks again

MichaIng, 2021-11-14:
Okay great. I adjusted this issue so we make the mentioned changes to prevent a Docker installation also when it gets pulled in as dependency, if a reboot is outstanding after a kernel upgrade. This can go hand in hand with: #4732

Rickyrivits, 2021-11-14:
Thanks and you are on the Ball! All I can say is you service is TOP NOTCH. Thanks for the help.

MichaIng, 2021-11-15:
Marking as closed in favour of: #4959

closed 2021-11-15 by MichaIng