MichaIng/DietPi #4957: Second USB port not working on NanoPi R1

ckuethe, 2021-11-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bullseye`
- Kernel version | `uname -a`
`Linux r1 5.10.60-sunxi #21.08.2 SMP Tue Sep 14 16:28:44 UTC 2021 armv7l GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`NanoPi R1 (armv7l)`
- Power supply used | (EG: 5V 1A RAVpower)
`Vilros 5V 2.5A`
- SDcard used | (EG: SanDisk ultra)
`SAMSUNG 32GB Evo Plus Class 10 Micro SDHC`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
`dmesg`, `minicom`
- Was the software title installed freshly or updated/migrated?
Fresh Install, with latest updates
- Can this issue be replicated on a fresh installation of DietPi?
Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Write fresh dietpi image
2. Boot dietpi image
3. Plug in a couple of USB devices
4. `lsusb -tv`

#### Expected behaviour

- dmesg or lsusb should show devices plugged into both usb ports (a ZWO camera and a usb-serial in my case)

#### Actual behaviour

- dmesg shows only the device plugged into the first (left) usb port
- right usb port does receive power though, the usb-serial powers up but does not identify

#### Extra details

My particular R1 came with internal eMMC preloaded with FriendlyWRT which does see both usb ports. I've attached a tarball of its `/boot` partition - I haven't yet determined what they're doing to enable the second port.

##### dmesg from dietpi
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.60-sunxi (root@runner4) (arm-linux-gnueabihf-gcc (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 8.3.0, GNU ld (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 2.32.0.20190321) #21.08.2 SMP Tue Sep 14 16:28:44 UTC 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=50c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: FriendlyElec NanoPi-R1
[    0.000000] earlycon: uart0 at MMIO32 0x01c28000 (options '115200n8')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x77c00000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x000000006fffffff]
[    0.000000]   HighMem  [mem 0x0000000070000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[... 1035 lines cut ...]
[   23.410256] dwmac-sun8i 1c30000.ethernet eth0: PTP not supported by HW
[   23.417702] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   24.139549] ieee80211 phy0: brcmf_net_attach: couldn't register the net device
[   24.146853] ieee80211 phy0: brcmf_ap_add_vif: Registering netdevice failed
[   24.154347] ieee80211 phy0: brcmf_cfg80211_add_iface: add iface wlan0 type 3 failed: err=-52
[   24.238045] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   24.246891] br-lan: port 2(wlan0) entered blocking state
[   24.252346] br-lan: port 2(wlan0) entered disabled state
[   24.258717] device wlan0 entered promiscuous mode
[   24.263751] br-lan: port 2(wlan0) entered blocking state
[   24.269184] br-lan: port 2(wlan0) entered forwarding state
[   24.336469] br-lan: port 1(eth1) entered disabled state
[   24.342614] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   24.355977] br-lan: port 2(wlan0) entered disabled state
[   24.534324] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   24.541144] br-lan: port 2(wlan0) entered blocking state
[   24.546567] br-lan: port 2(wlan0) entered forwarding state
[   28.308017] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[  412.416229] usb 7-1: new full-speed USB device number 2 using ohci-platform
[  412.652269] usb 7-1: New USB device found, idVendor=067b, idProduct=2303
[  412.659023] usb 7-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  412.666223] usb 7-1: Product: USB-Serial Controller
[  412.671112] usb 7-1: Manufacturer: Prolific Technology Inc.
```
[friendlywrt-boot-tgz.zip](https://github.com/MichaIng/DietPi/files/7534841/friendlywrt-boot-tgz.zip)

MichaIng, 2021-11-15:
Many thanks for your report.

The second USB port works and the serial device is detected ~3 minutes after boot, but disconnected 2 seconds later:
```
[  184.643397] usb 6-1: new full-speed USB device number 2 using ohci-platform
[  184.870381] usb 6-1: New USB device found, idVendor=067b, idProduct=2303, bcdDevice= 4.00
[  184.870409] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  184.870426] usb 6-1: Product: USB-Serial Controller
[  184.870441] usb 6-1: Manufacturer: Prolific Technology Inc.
[  184.950305] usbcore: registered new interface driver usbserial_generic
[  184.950406] usbserial: USB Serial support registered for generic
[  184.969560] usbcore: registered new interface driver pl2303
[  184.969774] usbserial: USB Serial support registered for pl2303
[  184.970151] pl2303 6-1:1.0: pl2303 converter detected
[  184.996857] usb 6-1: pl2303 converter now attached to ttyUSB0
[  186.657008] usb 6-1: USB disconnect, device number 2
[  186.658151] pl2303 ttyUSB0: pl2303 converter now disconnected from ttyUSB0
[  186.658308] pl2303 6-1:1.0: device disconnected
```
Can you try to attach it before booting the system? And can you also try to attach it as only USB device, leaving the camera unplugged to see whether this makes a difference? Bad cables and voltage issues can lead to flaky peripheral connections, so good to test things isolated. FriendlyWRT ships with a much older kernel, so things may be loaded in a different way/order, CPU frequency may be lower by default, or other reasons why the USB device stays connected.

ckuethe, 2021-11-15:
That second connect was me plugging and unplugging for testing. Enabling `usbhost2` fixes the second port and allows me to use the usb-serial device that I wanted to use. I'm not using ethernet at all on this device but `ip link` and `dmesg` still show two interfaces after enabling `usbhost2`

<details>
<summary>dmesg from R1 with `usbhost2` enabled</summary>
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.60-sunxi (root@runner4) (arm-linux-gnueabihf-gcc (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 8.3.0, GNU ld (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 2.32.0.20190321) #21.08.2 SMP Tue Sep 14 16:28:44 UTC 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=50c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: FriendlyElec NanoPi-R1
[    0.000000] earlycon: uart0 at MMIO32 0x01c28000 (options '115200n8')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x77c00000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x000000006fffffff]
[    0.000000]   HighMem  [mem 0x0000000070000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] On node 0 totalpages: 262144
[    0.000000]   Normal zone: 1728 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] percpu: Embedded 20 pages/cpu s50764 r8192 d22964 u81920
[    0.000000] pcpu-alloc: s50764 r8192 d22964 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 260416
[    0.000000] Kernel command line: root=UUID=1c77f31f-94b7-432e-94d3-73ea68d8a05d rootwait rootfstype=ext4 earlycon console=ttyS1,115200 hdmi.audio=EDID:0 disp.screen0_output_mode=1920x1080p60 consoleblank=0 loglevel=4 ubootpart=453dbe72-01 ubootsource=mmc usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u net.ifnames=0 sunxi_wdt.nowayout=1  sunxi_ve_mem_reserve=0 sunxi_g2d_mem_reserve=0 sunxi_fb_mem_reserve=16
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:on, heap free:off
[    0.000000] Memory: 882264K/1048576K available (9216K kernel code, 1460K rwdata, 3232K rodata, 1024K init, 333K bss, 35240K reserved, 131072K cma-reserved, 131060K highmem)
[    0.000000] random: get_random_u32 called from __kmem_cache_create+0x1b/0x298 with crng_init=0
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 42134 entries in 83 pages
[    0.000000] ftrace: allocated 83 pages with 4 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000012] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000033] Switching to timer-based delay loop, resolution 41ns
[    0.000492] Console: colour dummy device 80x30
[    0.000592] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.000624] pid_max: default: 32768 minimum: 301
[    0.001004] LSM: Security Framework initializing
[    0.001117] Yama: becoming mindful.
[    0.001438] AppArmor: AppArmor initialized
[    0.001585] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001613] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003560] CPU: Testing write buffer coherency: ok
[    0.004324] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005647] Setting up static identity map for 0x40100000 - 0x40100054
[    0.005861] rcu: Hierarchical SRCU implementation.
[    0.007966] smp: Bringing up secondary CPUs ...
[    0.019385] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.030923] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.042310] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[... 417 lines cut ...]
[   11.751933] usbcore: registered new interface driver usbserial_generic
[... cut ...]
[   12.133714] r8152 2-1:1.0 eth1: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[   12.463904] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   12.465995] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   12.467167] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Mar 30 2016 11:30:56 version 7.45.77.h8.4 FWID 01-ee8a6268
[   35.831375] vcc3v0: disabling
[   35.831405] vcc5v0: disabling

</details>

MichaIng, 2021-11-15:
Okay great. I'll add this overlay by default to our image.

MichaIng, 2021-11-15:
Also new images will have it enabled now by default: https://github.com/MichaIng/DietPi/commit/cc154ff

ckuethe, 2021-11-15:
I'd call this issue fixed, then. Thanks for the speedy resolution.

closed 2021-11-15 by ckuethe