MichaIng/DietPi #4958: Pi-hole installation fails on buster+PineA64

lwqcz, 2021-11-15:
#### Details:
- Date           | Mon Nov 15 09:45:58 GMT 2021
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | PINE A64 (aarch64) (ID=40)
- Kernel version | `Linux central 5.10.60-sunxi64 #21.08.1 SMP Wed Aug 25 18:29:57 UTC 2021 aarch64 GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `pihole -l off`
- Exit code      | 127
- Software title | DietPi-Software
#### Steps to reproduce:
- I've used `dietpi-software` tool to install Pi-hole and tried serval times, it fails all the time.
#### Expected behaviour:
- Pi-hole shoutld be correctly installed
#### Actual behaviour:

- The process is failing producing errors as shown below.
#### Extra details:
- I've even tried different power supply solution and even plug it directly to main network switch, instead of switch in my LAB.
- ...
#### Additional logs:
```
/boot/dietpi/func/dietpi-globals: line 863: pihole: command not found
```
---------------------------------------------------------------------

And here goes another errors I've found in console, regarding my attempt to install Pi-hole:

```
  [i] Update repo in /etc/.pihole...
  : Could not update local repository. Contact support.
[FAILED] DietPi-Software | ./install.sh --disable-install-webserver
```
#### Details:
- Date           | Mon Nov 15 09:44:57 GMT 2021
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | PINE A64 (aarch64) (ID=40)
- Kernel version | `Linux central 5.10.60-sunxi64 #21.08.1 SMP Wed Aug 25 18:29:57 UTC 2021 aarch64 GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `./install.sh --disable-install-webserver`
- Exit code      | 1
- Software title | DietPi-Software

Joulinar, 2021-11-15:
could you try to reboot your system and re-run PiHole installation. Pls share the entire log from the install. It should be possible if you scroll up on your SSH terminal

lwqcz, 2021-11-15:
- and here goes even printscreen of error dialog I'm getting all the time:
[image]

lwqcz, 2021-11-15:
The whole log for `dietpi-software`, after another reboot:
```
root@central:~# dietpi-software 
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[FAILED] DietPi-Config | An interface of type "wlan" does not exist.
[  OK  ] DietPi-Software | Free space check: path=/ | available=12081 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask 
[  OK  ] DietPi-Services | unmask : avahi-daemon
[  OK  ] DietPi-Services | unmask : php7.3-fpm
[  OK  ] DietPi-Services | unmask : lighttpd
[  OK  ] DietPi-Services | unmask : home-assistant
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : home-assistant
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : avahi-daemon
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Hit:4 https://deb.debian.org/debian buster-backports InRelease
Hit:5 http://mirrors.dotsrc.org/armbian-apt buster InRelease
Reading package lists...
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[ INFO ] DietPi-Software | Git will be installed
[ INFO ] DietPi-Software | SQLite will be installed
[ INFO ] DietPi-Software | PHP will be installed
[ INFO ] DietPi-Software | Lighttpd will be installed

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Git: Clone and manage Git repositories locally

[ INFO ] DietPi-Software | APT install for: git, please wait...
[  OK  ] DietPi-Software | APT install for: git

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Lighttpd: Extremely lightweight webserver

[ INFO ] DietPi-Software | APT install for: lighttpd perl, please wait...
[  OK  ] DietPi-Software | APT install for: lighttpd perl
[ INFO ] DietPi-Software | Removing obsolete SysV lighttpd service
[  OK  ] DietPi-Software | update-rc.d lighttpd remove
[  OK  ] DietPi-Software | Desired setting in /etc/lighttpd/lighttpd.conf was already set: server.document-root = "/var/www"
[  OK  ] DietPi-Software | rmdir --ignore-fail-on-non-empty /var/www/html

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing SQLite: Persistent single-file database system

[ INFO ] DietPi-Software | APT install for: sqlite3, please wait...
[  OK  ] DietPi-Software | APT install for: sqlite3

 DietPi-Software
[... 201 lines cut ...]
- Kernel version | `Linux central 5.10.60-sunxi64 #21.08.1 SMP Wed Aug 25 18:29:57 UTC 2021 aarch64 GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `pihole -l off`
- Exit code      | 127
- Software title | DietPi-Software
#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ...
2. ...
#### Expected behaviour:
<!-- What SHOULD happen? -->
- ...
#### Actual behaviour:
<!-- What IS happening? -->
- ...
#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...
#### Additional logs:
```
/boot/dietpi/func/dietpi-globals: line 863: pihole: command not found
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.
```

Joulinar, 2021-11-15:
ok no need to do print screens. let's remove the repository directory

```
rm -Rf /etc/.pihole
dietpi-software install 93
```

lwqcz, 2021-11-15:
Ok. Did what you've suggested, but no luck. I've just uploaded a bug report from `dietpi-software` dialog.

Joulinar, 2021-11-15:
I don't have access to the bug reports, This can be done by our developer only. Due to whatever reason the PiHole installer is not able to write/update to `/etc/.pihole`. Could you check if it is existing,

```
ls -la /etc/.pihole
```

lwqcz, 2021-11-15:
That directory is writeable by the root user I assume:
```
root@central:~# ls -la /etc/.pihole
total 140
drwxr-xr-x  8 root root  4096 Nov 15 10:22  .
drwxr-xr-x 80 root root  4096 Nov 15 10:22  ..
-rw-r--r--  1 root root   779 Nov 15 10:22  .editorconfig
drwxr-xr-x  8 root root  4096 Nov 15 10:22  .git
-rw-r--r--  1 root root   688 Nov 15 10:22  .gitattributes
drwxr-xr-x  3 root root  4096 Nov 15 10:22  .github
-rw-r--r--  1 root root    90 Nov 15 10:22  .gitignore
-rw-r--r--  1 root root    83 Nov 15 10:22  .stickler.yml
-rw-r--r--  1 root root  6223 Nov 15 10:22  CONTRIBUTING.md
-rw-r--r--  1 root root 14204 Nov 15 10:22  LICENSE
-rw-r--r--  1 root root  9325 Nov 15 10:22  README.md
drwxr-xr-x  6 root root  4096 Nov 15 10:22  advanced
drwxr-xr-x  2 root root  4096 Nov 15 10:22 'automated install'
-rwxr-xr-x  1 root root 33596 Nov 15 10:22  gravity.sh
drwxr-xr-x  2 root root  4096 Nov 15 10:22  manpages
-rwxr-xr-x  1 root root 16711 Nov 15 10:22  pihole
drwxr-xr-x  2 root root  4096 Nov 15 10:22  test
root@central:~# ls -la /etc/ | grep .pihole
drwxr-xr-x  8 root root    4096 Nov 15 10:22 .pihole
drwxr-xr-x  2 root root    4096 Nov 15 10:21 pihole
```

lwqcz, 2021-11-15:
I've tried this and it worked 🥂 
```
chmod 775 /etc/.pihole/ && rm -rf /var/www/html/admin
```
... it seems to me, that installed is using different mode then just root.

Joulinar, 2021-11-15:
hmm usually not 🤔 . I did a test installation on my RPi3b+ Bullseye ARMv7 and for me it's working fine ootb

```
  [i] Logging On.
  [✗] Check for existing repository in /etc/.pihole
  [i] Clone https://github.com/pi-hole/pi-hole.git into /etc/.pihole...HEAD is now at b5e0f14 Merge pull request #4405 from pi-hole/development
  [✓] Clone https://github.com/pi-hole/pi-hole.git into /etc/.pihole

  [✗] Check for existing repository in /var/www/html/admin
  [i] Clone https://github.com/pi-hole/AdminLTE.git into /var/www/html/admin...HEAD is now at ee50954 Merge pull request #1935 from pi-hole/devel
  [✓] Clone https://github.com/pi-hole/AdminLTE.git into /var/www/html/admin

```

MichaIng, 2021-11-15:
`chmod 775 /etc/.pihole` cannot make any difference. It was `755`, the owner is and stays `root:root` and nothing is (or should be) member of the `root` group which is basically used to declare that no group should get additional permissions.

It seems that disk reads or writes were flaky. Please check for kernel errors, to assure that any issues are detected as early as possible before more problems happen:
```sh
dmesg -l emerg,alert,crit,err
```

MichaIng, 2021-11-15:
> Ok. Did what you've suggested, but no luck. I've just uploaded a bug report from dietpi-software dialog.

Please paste the provided ID so that I can find the bug report. I'll have a look whether kernel or disk I/O issues can be seen.
```sh
echo $G_HW_UUID
```

MichaIng, 2021-11-15:
I guess I found the bug report. No kernel errors visible there, though the errored out at `/var/www/html/admin` already existing, which is strange. This should only happen if the directory exists but the Git metadata are damaged, e.g. the contained `.git` dir does not exist or so.

One service was failing. Can you show:
```sh
systemctl status systemd-modules-load
journalctl -u systemd-modules-load
systemctl restart systemd-modules-load
```
I guess one of the modules listed in `/etc/modules-load.d` does not exist.

lwqcz, 2021-11-16:
Thx for your answers. I'll add more data as soon as I will arrive home to access the device. The install is old (img install was made from) but clean. The bug report I've found in console history is `f1625446-9c30-4c0e-aeea-9a6e8f1753b0`.

MichaIng, 2021-12-08:
Any new, or can the issue otherwise be closed?

MichaIng, 2022-01-04:
I'll mark this as closed. Feel free to reopen when required.

closed 2022-01-04 by MichaIng