MichaIng/DietPi #4964: UUID problem when copeling new disk on Bullseye

cocoflan, 2021-11-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_Bullseye`

When i connect external disk on USB on a running RPI.., my UUID was again gone.

Is this normal? only solution was to re-anble the names by deleting them on /mnt and re-anble them in dietpi-drive_manager by changing the name again to the original?

So short saying, if you connect a disk on a running system you still get in trouble with UUID?

MichaIng, 2021-11-16:
I'm not sure whether I understand your issue:
- Every filesystem has an unique UUID. So if you format a drive with a new filesystem, it'll have a new UUID. If you reattach the same drive without formatting it, it'll have the same UUID than before.
- But the UUID and the mount point are two different things, we only use the UUID as default mount point directory, just because we know it is unique.
- If you unmount a drive via `dietpi-drive_manager`, it's mount point in `/mnt` and the entry from `/etc/fstab` will be removed. So when you attach the same drive later and mount it via `dietpi-drive_manager`, the previously chosen mount point won't be known anymore, but the UUID will again be shown again as default in the input box.

Does this answer your question? Else please try to be more precise about which menu or command output you expect and what you see instead.

cocoflan, 2021-11-16:
No it was an disk never connected to the system caused that problem.

MichaIng, 2021-11-16:
Please describe the issue then:
> try to be more precise about which menu or command output you expect and what you see instead.

cocoflan, 2021-11-16:
So after connecting a disk on a running system i got this error with the UUID, so the given names are gone of all my other disks like i indicated in the past i posted. to keep it clear names that are given on mount are gone.

MichaIng, 2021-11-16:
> names that are given on mount are gone

So you mean the mount points, like `/mnt/my_backup_drive`, which are gone when you remount a drive, right? As I said, when you unmount a drive via `dietpi-drive_manager`, this mount point is indeed removed, so you need to reenter it when remounting the drive.

There is currently no temporary mount or temporary unmount functionality. What I can generally think of is:
1. An option to temporarily mount a drive, so that the `/etc/fstab` entry is skipped. That way it won't be automatically mounted on reboot and unplugging it doesn't cause issues with the systemd automount units, which are created otherwise.
2. An option to temporarily unmount a drive, so that the `/etc/fstab` entry is preserved, which allows to keep the mount point and other possible mount options.

But both isn't that trivial to implement the way `dietpi-drive_manager` does things currently, where `/etc/fstab` is rewritten based on which drives are currently mounted, so that all mounts and unmounts are inherently "permanent". We'd need an additional config file to store the additional information which fstab entries to skip and which to preserve.

If you want to temporarily unmount a drive which usually stays attached to the system, you could do:
```sh
umount /mnt/my_backup_drive
```
Then you can detach the drive. After reattaching it:
```sh
mount /mnt/my_backup_drive
```
Since `/etc/fstab` does not change, the mount point information remains the same. But there is one issue while the drive is detached. Then try to access the mount point, the systemd automount unit will hang the console for 90 seconds (timeout), e.g.:
```
ls /mnt/my_backup_drive
```
This is as it tries to automatically mount the drive, as fast as the mount point is accessed. But without the drive attached, this of course fails, i.e. it times out after 90 seconds.

MichaIng, 2022-01-04:
I'll mark this as closed. Please try to clarify the issue, based on my question above, if the issue persists. I still didn't 100% get it 😅.

closed 2022-01-04 by MichaIng