MichaIng/DietPi #4966: Influxdb is not starting after install

blablazzz, 2021-11-17:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.8.2
- Distro version | bullseye
- Kernel version | Linux DietPi 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64 GNU/Linux
- SBC model | ROCK64
- Power supply used | 5v3A

#### Steps to reproduce
1. Install Influxdb

#### Expected behaviour
Should be active

#### Actual behaviour
But its written - activating

after running 'systemctl restart influxdb' got error:
Job for influxdb.service failed because a timeout was exceeded.
See "systemctl status influxdb.service" and "journalctl -xe" for details.

and it gives:
Nov 17 08:09:22 DietPi influxd-systemd-start.sh[189435]: InfluxDB API unavailable after 74 attempts...
Nov 17 08:09:23 DietPi influxd-systemd-start.sh[189435]: InfluxDB API unavailable after 75 attempts...
Nov 17 08:09:24 DietPi influxd-systemd-start.sh[189435]: InfluxDB API unavailable after 76 attempts...

Joulinar, 2021-11-17:
Hmm I testes InfluxDB on a RPi4B 64bit and it's working fine without issues. Can you share following

```
journalctl -u influxdb
```

Are there any files located in `/var/log/influxdb` ?

blablazzz, 2021-11-17:
`/var/log/influxdb ` is empty
`journalctl -u influxdb` gives only - InfluxDB API unavailable

Joulinar, 2021-11-17:
hmm looks like the database is taking quite long to start up. I guess the counter is still increasing if your run `journalctl -u influxdb` again?

blablazzz, 2021-11-17:
Nov 17 12:17:02 DietPi systemd[1]: influxdb.service: start operation timed out. Terminating.
Nov 17 12:17:02 DietPi systemd[1]: influxdb.service: Failed with result 'timeout'.
Nov 17 12:17:02 DietPi systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
Nov 17 12:17:02 DietPi systemd[1]: influxdb.service: Consumed 5.284s CPU time.
Nov 17 12:17:02 DietPi systemd[1]: influxdb.service: Scheduled restart job, restart counter is at 166.
Nov 17 12:17:02 DietPi systemd[1]: Stopped InfluxDB is an open-source, distributed, time series database.
Nov 17 12:17:02 DietPi systemd[1]: influxdb.service: Consumed 5.284s CPU time.
Nov 17 12:17:02 DietPi systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Nov 17 12:17:02 DietPi influxd-systemd-start.sh[247313]: Merging with configuration at: /etc/influxdb/influxdb.conf
Nov 17 12:17:02 DietPi influxd-systemd-start.sh[247311]: ts=2021-11-17T12:17:02.944710Z lvl=info msg="InfluxDB starting" log_id=0XsT6Tt0000 version=1.8.10 branch=1.8 commit=688e697c51fd
Nov 17 12:17:02 DietPi influxd-systemd-start.sh[247311]: ts=2021-11-17T12:17:02.944858Z lvl=info msg="Go runtime" log_id=0XsT6Tt0000 version=go1.13.8 maxprocs=4
Nov 17 12:17:03 DietPi influxd-systemd-start.sh[247330]: Merging with configuration at: /etc/influxdb/influxdb.conf
Nov 17 12:17:03 DietPi influxd-systemd-start.sh[247310]: InfluxDB API unavailable after 1 attempts...
Nov 17 12:17:04 DietPi influxd-systemd-start.sh[247310]: InfluxDB API unavailable after 2 attempts...
Nov 17 12:17:05 DietPi influxd-systemd-start.sh[247310]: InfluxDB API unavailable after 3 attempts...
Nov 17 12:17:05 DietPi influxd-systemd-start.sh[247311]: run: open server: listen: listen tcp 127.0.0.1:8088: bind: address already in use

Joulinar, 2021-11-17:
there is an already running instance on port 8088. Did you install DietPi Dashboard?

blablazzz, 2021-11-17:
yes, i have dashboard

Joulinar, 2021-11-17:
pls remove the dashboard and reboot your system

MichaIng, 2021-11-17:
Ayayay, did we cause a port conflict? InfluxDB should listen on port `8086` by default for database queries, but probably there is another listener active for other tasks which we didn't think about.

Joulinar, 2021-11-17:
yes there are 2

```
tcp   LISTEN 0      4096       127.0.0.1:8088      0.0.0.0:*    users:(("influxd",pid=7945,fd=3))
tcp   LISTEN 0      4096               *:8086            *:*    users:(("influxd",pid=7945,fd=14))
```

MichaIng, 2021-11-17:
No need to reboot the system or uninstall dashboard. This is sufficient for testing:
```sh
systemctl stop dietpi-dashboard
systemctl restart influxdb
```
If it works then, you can change the port which dietpi-dashboard uses in `/opt/dietpi-dashboard/config.toml`, e.g.:
```sh
G_CONFIG_INJECT 'port[[:blank:]]' 'port = 8089' /opt/dietpi-dashboard/config.toml
systemctl restart dietpi-dashboard
```

Joulinar, 2021-11-17:
it is the default setting for the RPC service

```
# Bind address to use for the RPC service for backup and restore.
# bind-address = "127.0.0.1:8088"
```

blablazzz, 2021-11-17:
uninstalled dashboard - InfluxDB got running. All good.
All the ports where standard.

MichaIng, 2021-11-17:
Since `dietpi-dashboard` is still in beta anyway, I guess it's easier and less painful to adjust it's default port and leave the InfluxDB RPC port untouched. @ravenclaw900 ftr. Next, `8089`, or `8090` as seen easier to remember?

We need to add the RPC port info to the InfluxDB docs as well.

MichaIng, 2021-11-17:
@blablazzz 
You can keep using the dashboard perfectly fine, see: https://github.com/MichaIng/DietPi/issues/4966#issuecomment-971538695

Joulinar, 2021-11-17:
@MichaIng 
Should I open an issue at the Dashboard GitHub?

MichaIng, 2021-11-17:
Yes, though it also needs to be changed in `dietpi-software` where the config file is created.

ravenclaw900, 2021-11-17:
Note that we don't have to clutter the 80xx port numbers. I just chose 8080 originally as a placeholder. Something like 3030 or so would work just as well.

MichaIng, 2021-11-17:
True. Though ports in all ranges are in use, so not sure whether using something below `808*` or `303*` makes much difference. At some point we need to stay with one and then change the default port of newly added software, in case it conflicts with another one.

MichaIng, 2021-12-02:
@ravenclaw900 
What do we want to do about that? Override the default port in `dietpi-software` or changing the default in `dietpi-dashboard`? I'm at this stage basically fine with every solution and with every alternative 4 digits port. Probably it indeed makes sense to use a less common port (range), like `5252` or so, at least this and everything around it seems to be pretty unused among our software implementations.

ravenclaw900, 2021-12-02:
I'd change it in the dashboard itself, just to keep it consistent even without config file, though that breaks anything written about the dashboard beforehand. Nothing wrong with `5252`, probably even more memorable then `8088`.

MichaIng, 2021-12-02:
@Joulinar 
As discussed that it makes sense to inform users of this, even that currently installed dashboards are not affected (as the port is explicitly set in the config and the config not touched on reinstalls), probably additionally we can add a yesno to the update when explicit port 8088 is found in the config and ask whether this shall be changed or not?

Joulinar, 2021-12-02:
Probably that would make sense during update to avoid people running on that conflicting port. As well our docs will not fit once we changed.

MichaIng, 2021-12-04:
The TCP port change is not offered during next DietPi update: https://github.com/MichaIng/DietPi/commit/91e792d

closed 2021-12-11 by MichaIng