MichaIng/DietPi #4967: DietPi-Backup | logfile not created correctly

Joulinar, 2021-11-17:
Since the last update 7.8 backup log file is not created correctly on new located backup target dir. As well transition of backup log during `dietpi-update` is failing if the target location is not mounted (like a disk unpluged)

empty log file - https://dietpi.com/phpbb/viewtopic.php?t=9678
I did a test and could verify it

```
root@DietPi4:~# cat /mnt/test_backup/dietpi-backup.log
Backup log from 2021-11-17_16:45:30

root@DietPi4:~#
```

according the user there is a well a try to backup a file not existing as the user is not using default location

```
[  OK  ] DietPi-Backup | Free space check: path=/mnt/dd6b05a5-a3bb-429a-95a8-b2cf5189f7ff/dietpi-backup/data | available=204553 MiB | required=704 MiB
[ INFO ] DietPi-Backup | Backup to /mnt/dd6b05a5-a3bb-429a-95a8-b2cf5189f7ff/dietpi-backup in progress, please wait...
rsync: [client] failed to open log-file /mnt/dietpi-backup/dietpi-backup.log: No such file or directory (2)
Ignoring "log file" setting.
```

---

issue during `dietpi-update` - https://dietpi.com/phpbb/viewtopic.php?t=9658

```
[ INFO ] DietPi-Patch | Moving DietPi-Backup log file to /mnt/dietpi-backup/dietpi-backup.log
[FAILED] DietPi-Patch | mv /var/log/dietpi-backup.log /mnt/dietpi-backup/dietpi-backup.log
```

MichaIng, 2021-11-17:
I see why the wrong log file is used: https://github.com/MichaIng/DietPi/blob/792a406/dietpi/dietpi-backup#L57
The `FP_TARGET` variable is evaluated and stored to readonly `aRSYNC_LOGGING_OPTIONS` before the config file is loaded. Also it needs to be re-evaluated whenever the target directory changes. Hmm, at best it is evaluated on `rsync` call directly. I'll remove `aRSYNC_LOGGING_OPTIONS` add add those to the two related `rsync` calls directly.

But I do not understand the upgrade error yet. It checks `[[ $FP_TARGET && -d $FP_TARGET && ! -f $FP_TARGET/dietpi-backup.log ]]` hence only attempts to move the file when the target directory exists. But indeed better is to check whether it contains an actual backup (either the `data` directory or the `.dietpi-backup_stats` files), which implies whether it is mounted or not. Was the automount unit probably active but the drive not attached so that systemd automount timed out?

MichaIng, 2021-11-19:
Issue solved for v7.9 and via live patch for v7.8

closed 2021-11-19 by MichaIng