MichaIng/DietPi #4969: radxa zero  1Gb + SD 32 Gb hangs on reboot

promotux, 2021-11-18:
#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
- Distro version | bullseye`
- Kernel version | Linux DietPi 5.10.69-4-amlogic-g37ae4051315a #g37ae4051315a SMP PREEMPT Tue Oct 26 10:16:23 CST 2021 aarch64 GNU/Linux

- SBC model | Radxa Zero (aarch64)
- SDcard used | Gigastone

#### Steps to reproduce

reboot 

#### Actual behaviour

after few second screen become black , keyboard still on and nothing more happen

MichaIng, 2021-11-18:
AFAIK the reboot issue was solved with the 5.10.69-5 kernel package. Please try the newest package from the testing suite:
```sh
cd /tmp
curl -f 'https://raw.githubusercontent.com/radxa/apt/gh-pages/buster-testing/pool/main/l/linux-5.10.69-7-amlogic-gfd159ba07d5c/linux-image-5.10.69-7-amlogic-gfd159ba07d5c_5.10.69-7-amlogic-gfd159ba07d5c_arm64.deb' -o linux.deb
dpkg -i linux.deb
rm linux.deb
reboot
```
After reboot you can purge the old package:
```sh
apt purge --autoremove 'linux-5.10.69-4-amlogic-g37ae4051315a'
```

MichaIng, 2021-11-19:
I uploaded a new image with latest kernel. Please test whether this boots and works fine: https://dietpi.com/downloads/images/testing/DietPi_RadxaZero-ARMv8-Bullseye.7z

MichaIng, 2021-11-19:
New image boots fine and has been moved into stable place. Marking this as closed.

promotux, 2021-11-19:
hi,  tested this last img but with no luck on reboot. ( it didnt work neither with debian or ubuntu downloaded from radxa zero site )

MichaIng, 2021-11-20:
Hmm, @almirus reboot works fine in your case, doesn't it?

almirus, 2021-11-20:
@MichaIng as before - no

on  DietPi v7.8.2
<pre>
[  OK  ] Reached target Reboot.
[94838.399711] systemd-shutdown[1]: Syncing filesystems and block devices.
[94838.419683] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[94838.427842] systemd-journald[1456]: Received SIGTERM from PID 1 (systemd-shutdow).
[94838.444733] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[94838.453703] systemd-shutdown[1]: Unmounting file systems.
[94838.455573] [4384]: Remounting '/' read-only in with options '(null)'.
[94838.471873] EXT4-fs (mmcblk1p2): re-mounted. Opts: (null)
[94838.477438] systemd-shutdown[1]: All filesystems unmounted.
[94838.477567] systemd-shutdown[1]: Deactivating swaps.
[94838.483085] systemd-shutdown[1]: All swaps deactivated.
[94838.487579] systemd-shutdown[1]: Detaching loop devices.
[94838.495351] systemd-shutdown[1]: All loop devices detached.
[94838.498388] systemd-shutdown[1]: Stopping MD devices.
[94838.503549] systemd-shutdown[1]: All MD devices stopped.
[94838.508685] systemd-shutdown[1]: Detaching DM devices.
[94838.513848] systemd-shutdown[1]: All DM devices detached.
[94838.519069] systemd-shutdown[1]: All filesystems, swaps, loop devices, MD devices and DM devices detached.
[94838.543134] systemd-shutdown[1]: Syncing filesystems and block devices.
[94838.544364] systemd-shutdown[1]: Rebooting.
[94838.548340] kvm: exiting hardware virtualization
</pre>

promotux, 2021-11-20:
changed hardware with another one ( i bought 4 of them )  and have hte same issue. In a forum a read that can be a SD type problem. I'll try to buy a new one of aanother brand ( using Gigastone 32 Gb  now )

MichaIng, 2021-11-20:
Ah I mixed it up, though it was solved with one of the recent kernel updates. Okay so this is then the last still outstanding issue. Already reported at Radxa: https://github.com/radxa/radxa-zero-images-released/issues/3

Framelony, 2021-11-23:
Just as a notice, the last  debian "Buster" img 22-11-2021 has still the reboot problem.

MichaIng, 2021-11-23:
Yes, still needs to be resolved kernel wise: https://github.com/radxa/radxa-zero-images-released/issues/3

almirus, 2021-11-23:
@MichaIng   problem is in bootloader
the solution is here https://forum.radxa.com/t/instruction-of-how-to-fix-2-seconds-delay-at-boot-and-not-working-reboot/7898

MichaIng, 2021-11-23:
Note that the two seconds boot delay which is removed by this may break boot in other cases, when the root drive/card takes a little while to be fully accessible. So I prefer to wait for a solution from Radxa end instead of compiling an own U-Boot.

MichaIng, 2021-12-06:
Seems to be solved with the latest Radxa Debian image. I rebuilt ours based on it. Sadly there is not U-Boot package to fix it without either extracting and flashing the bootloader manually of flashing the new image 😢. Probably the first 15 MiB of the image can be `dd`ed into the first 15 MiB of the running system, but if something wrong is flashed, it can render the system unbootable. We'd need to find out where and with what size the U-Boot files are location, I think this is always moreless the same with modern aarch64 U-Boot. I'll try to find out and in case provide files + a script to update it.

However, if a new image is fine, it's uploaded: https://dietpi.com/downloads/images/

Framelony, 2021-12-06:
still no luck with DietPi_RadxaZero-ARMv8-Bullseye.7z  from 06/12/2021

MichaIng, 2021-12-06:
Dammit, I thought so because the Radxa Zero image maintainer said so: https://github.com/radxa/radxa-zero-images-released/issues/7#issuecomment-955179003

Samples still did not arrive, has been >1 month now, I'll check back. When those are here I can try to compile U-Boot with the fix.

Framelony, 2021-12-06:
> Samples still did not arrive, has been >1 month now, I'll check back. When those are here I can try to compile U-Boot with the fix.
waited almost two month for five pieces, but i think it worth the waiting.

MichaIng, 2021-12-10:
A new kernel seems to solve it. Could you try:
```sh
cd /tmp
curl -sSfL 'https://raw.githubusercontent.com/radxa/apt/gh-pages/buster-testing/pool/main/l/linux-5.10.69-8-amlogic-g8e2cfaa5faa5/linux-image-5.10.69-8-amlogic-g8e2cfaa5faa5_5.10.69-8-amlogic-g8e2cfaa5faa5_arm64.deb' -o linux-image-5.10.69-8.deb
dpkg -i linux-image-5.10.69-8.deb
rm linux-image-5.10.69-8.deb
```

Framelony, 2021-12-10:
> A new kernel seems to solve it. Could you try:
> 
> ```shell
> cd /tmp
> curl -sSfL 'https://raw.githubusercontent.com/radxa/apt/gh-pages/buster-testing/pool/main/l/linux-5.10.69-8-amlogic-g8e2cfaa5faa5/linux-image-5.10.69-8-amlogic-g8e2cfaa5faa5_5.10.69-8-amlogic-g8e2cfaa5faa5_arm64.deb' -o linux-image-5.10.69-8.deb
> dpkg -i linux-image-5.10.69-8.deb
> rm linux-image-5.10.69-8.deb
> ```

it works! 
done  and rebooted two times without problems.

MichaIng, 2021-12-10:
Great, I'll upgrade it in our image 🙂:+1:

jpramosi, 2022-04-16:
I can also confirm that the latest kernel image fixes the issue. I just updated the kernel and everything was smooth.
```bash
cat /etc/apt/sources.list.d/dietpi-radxa.list
curl -s http://apt.radxa.com/buster-stable/dists/buster/main/binary-arm64/Packages | grep -h -P -o "^Package: \K.*" | grep -e "linux-image-.*amlogic"
sudo apt update
sudo apt install linux-image-5.10.69-10-amlogic-g617a45dd0fce
sudo reboot now
# optional step
sudo apt remove --purge linux-image-5.10.69-7-*
```

<details><summary>cat /boot/uEnv.txt</summary>

```
verbosity=7
console=ttyAML0,115200
overlay_prefix=meson
rootfstype=ext4
fdtfile=amlogic/meson-g12a-radxa-zero.dtb
overlays=meson-g12b-spi-b-gpioh-4-5-6-7 meson-g12a-pwm-c-on-gpiox-8
param_spidev_spi_bus=1
param_spidev_max_freq=4000000
rootuuid=0a23bd6c-0d3b-458f-9256-9a345cd0e89b
initrdsize=0xc81498
kernelversion=5.10.69-10-amlogic-g617a45dd0fce
initrdimg=initrd.img-5.10.69-10-amlogic-g617a45dd0fce
kernelimg=vmlinuz-5.10.69-10-amlogic-g617a45dd0fce
```

</details>

<details><summary>sudo dmesg</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.69-10-amlogic-g617a45dd0fce (stephen@lara) (gcc (Ubuntu/Linaro 7.5.0-6ubuntu2) 7.5.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #amlogic SMP PREEMPT Wed Jan 26 09:18:39 UTC 2022
[    0.000000] Machine model: Radxa Zero
[    0.000000] efi: UEFI not found.
[    0.000000] OF: fdt: Reserved memory: failed to reserve memory for node 'secmon@5000000': base 0x0000000005000000, size 3 MiB
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000b000000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x1fea5100-0x1fea6fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   DMA zone: 2048 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 131072 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.1
[    0.000000] percpu: Embedded 23 pages/cpu s56920 r8192 d29096 u94208
[    0.000000] pcpu-alloc: s56920 r8192 d29096 u94208 alloc=23*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Policy zone: DMA
[    0.000000] Kernel command line: initrd=initrd.img-5.10.69-10-amlogic-g617a45dd0fce root=UUID=0a23bd6c-0d3b-458f-9256-9a345cd0e89b rootwait rw rootfstype=ext4 console=tty1 console=ttyAML0,115200 panic=10 consoleblank=0 loglevel=7   cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[... 413 lines cut ...]

```
ATTENTION: default value of option vblank_mode overridden by environment.
3984 frames in 5.0 seconds = 796.766 FPS
4014 frames in 5.0 seconds = 802.605 FPS
3999 frames in 5.0 seconds = 799.704 FPS
```

</details>

Device tree overlays, wifi, panfrost and everything else is working!
I will do later some more tests but it's really great now.
But the provided image on the download page

```
FILE:   DietPi_RadxaZero-ARMv8-Bullseye.img
DATE:   Sat Dec 18 00:45:40 CET 2021
MD5:    0a1cc3fd6bc238b3bf6c3de8ec3555bf
SHA1:   bf9b601500d4d4bbfcccf519c089b4306e4d1318
SHA256: d12353c78a34c9e8a9c26ebde74b7b9fe715cf17fbb937854c5e80774f910d80
```

is still an old image with the 5.10.69-7 kernel.
Nonetheless this board (and the OS image) are now perfect for my use case.
Thank you all.

almirus, 2022-04-16:
checked today, reboot works.

MichaIng, 2022-04-16:
Nice, finally there is a "latest" kernel package in the stable repository. I'll re-create our image, using that, so kernel upgrades can be applied via `apt upgrade`: https://github.com/radxa/apt/tree/gh-pages/buster-stable/pool/main/l/linux-5.10-radxa-zero-latest
```sh
apt update
apt install linux-5.10-radxa-zero-latest
```

MichaIng, 2022-05-03:
New image with new kernel has been uploaded 🙂. Marking this as solved.

closed 2022-05-03 by MichaIng