MichaIng/DietPi #4971: Issue with dietpi-update

martingtaylor, 2021-11-18:
#### Details:
- Date           | Thu Nov 18 16:04:38 GMT 2021
- Bug report     | e4027db9-45fd-4e4f-ad37-669d60d3e66a
- DietPi version | v7.7.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | `Linux DietPi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -qq upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Setting up nginx-full (1.14.2-2+deb10u4) ...
Job for nginx.service failed because the control process exited with error code.
See "systemctl status nginx.service" and "journalctl -xe" for details.
invoke-rc.d: initscript nginx, action "start" failed.

Joulinar, 2021-11-18:
Pls can you share following

```
journalctl -u nginx.service
```

martingtaylor, 2021-11-18:
***@***.***:/$ su root
Password:

 DietPi v7.7.3 : Update available

 - Device model : RPi 2 Model B (armv7l)
 - CPU temp : 31'C : 87'F (Cool runnings)
 - LAN IP : 192.168.0.100 (eth0)

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 - MOTD : Learn how to upgrade your system to Bullseye:
          https://dietpi.com/blog/?p=811
 Image by        : DietPi Core Team (pre-image: Raspberry Pi OS (32-bit) Lite)
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Patreon Legends : Camry2731
 Contribute      : https://dietpi.com/contribute.html
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-update   : Run now to update DietPi from v7.7.3 to v7.8.2

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

***@***.***:/# journalctl -u nginx.service
-- Logs begin at Thu 2021-11-18 16:10:52 GMT, end at Thu 2021-11-18 16:21:29 GMT. --
Nov 18 16:11:38 DietPi systemd[1]: Starting A high performance web server and a reverse proxy server...
Nov 18 16:11:38 DietPi nginx[285]: nginx: [alert] could not open error log file: open() "/var/log/nginx/error.log" failed (2: No such file or directory)
Nov 18 16:11:38 DietPi nginx[285]: 2021/11/18 16:11:38 [emerg] 285#285: open() "/var/log/nginx/access.log" failed (2: No such file or directory)
Nov 18 16:11:38 DietPi nginx[285]: nginx: configuration file /etc/nginx/nginx.conf test failed
Nov 18 16:11:38 DietPi systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Nov 18 16:11:38 DietPi systemd[1]: nginx.service: Failed with result 'exit-code'.
Nov 18 16:11:38 DietPi systemd[1]: Failed to start A high performance web server and a reverse proxy server.
Nov 18 16:11:44 DietPi systemd[1]: Starting A high performance web server and a reverse proxy server...
***@***.***:/# Nov 18 16:11:44 DietPi nginx[469]: nginx: [alert] could not open error log file: open() "/var/log/nginx/error.log" failed (2: No such file or directory)
Nov 18 16:11:44 DietPi nginx[469]: 2021/11/18 16:11:44 [emerg] 469#469: open() "/var/log/nginx/access.log" failed (2: No such file or directory)
Nov 18 16:11:44 DietPi nginx[469]: nginx: configuration file /etc/nginx/nginx.conf test failed
Nov 18 16:11:44 DietPi systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Nov 18 16:11:44 DietPi systemd[1]: nginx.service: Failed with result 'exit-code'.
Nov 18 16:11:44 DietPi systemd[1]: Failed to start A high performance web server and a reverse proxy server.

Thanks - Martin

Joulinar, 2021-11-18:
Looks like error log is missing. Could you share following

```
ls -la /var/log/
```

martingtaylor, 2021-11-18:
***@***.***:/# ls -la /var/log/
total 16
***@***.***:/# drwxr-xr-t  8 root     root      320 Oct  8 13:24 .
drwxr-xr-x 12 root     root     4096 Aug 11  2020 ..
-rw-r--r--  1 root     root        0 Oct 13 11:04 alternatives.log
drwxr-xr-x  2 root     root      100 Nov 18 16:04 apt
-rw-rw----  1 root     utmp        0 Nov 18 16:10 btmp
-rw-r--r--  1 root     root     1585 Nov 18 16:39 dataplicity.log
-rw-r--r--  1 root     root        0 Oct  8 14:17 dpinstall.log
-rw-r--r--  1 root     root        0 Nov 18 16:04 dpkg.log
-rw-rw-r--  1 root     utmp        0 Nov 18 16:10 lastlog
drwxr-x---  2 www-data www-data   60 Aug 11  2020 lighttpd
drwxr-s---  2 mysql    adm        60 Aug 11  2020 mysql
-rw-------  1 root     root      185 Nov 18 16:11 php7.3-fpm.log
drwx------  2 root     root       40 Aug 11  2020 private
drwxr-s---  2 redis    adm        60 Aug 11  2020 redis
drwxr-xr-x  2 root     root       60 Oct  8 13:24 supervisor
-rw-rw-r--  1 root     utmp     1536 Nov 18 16:11 wtmp

MichaIng, 2021-11-18:
Whyever it got removed, recreate Nginx log directory:
```sh
mkdir /var/log/nginx
chown www-data:www-data /var/log/nginx
```
I see you have Lighttpd installed as well? Is it active and a specific reason why you run two webservers concurrently? Naturally they cannot run both on regular ports 80/443 and usually there is no need to run two of them as every webserver can serve multiple websites or even vhosts and nearly all features are available on all of them.

Joulinar, 2021-11-18:
Maybe Lighttpd is a leftover from the past?

martingtaylor, 2021-11-18:
I think it might have been installed with Dataplicity. I’ll try and remove it.

Thanks.

Martin

MichaIng, 2021-11-29:
Is this issue solved, either with removing Nginx and using Lighttpd or creating the Nginx log directory?

MichaIng, 2021-12-08:
I'll mark this as closed due to outstanding reply. Feel free to reopen if required.

closed 2021-12-08 by MichaIng