MichaIng/DietPi #4973: dietpi-software reinstall 122 failing

fmuntean, 2021-11-19:
#### Details:
- Date           | Thu Nov 18 19:03:20 EST 2021
- Bug report     | bd2852e4-45ff-488b-b617-8dd65be89618
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi Zero (armv6l) (ID=1)
- Kernel version | `Linux rpi-zero 5.10.63+ #1459 Wed Oct 6 16:40:27 BST 2021 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `sudo -u nodered npm i --no-audit node-red`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
TypeError: Class extends value undefined is not a constructor or null
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1149:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:999:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1149:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:999:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
```

please advise

MichaIng, 2021-11-19:
Many thanks for your report.

On x86_64 it works fine, including the install of `socks-proxy-agent` and `make-fetch-happen` modules. Will try to replicate on RPi Zero W.

Do you remember whether the Node version changed, respectively what the Node version was before?

fmuntean, 2021-11-19:
the node version has changed successfully to the latest

MichaIng, 2021-11-19:
I'm wondering whether a new Node version can invalidate the installed modules.

fmuntean, 2021-11-21:
Any Idea how to fix this issue ?

MichaIng, 2021-11-21:
Can you try to run the command manually and print the full output here:
```sh
cd /mnt/dietpi_userdata/node-red
sudo -u nodered npm i --no-audit node-red
```

fmuntean, 2021-11-21:
cd /mnt/dietpi_userdata/node-red
sudo -u nodered npm i --no-audit node-red
TypeError: Class extends value undefined is not a constructor or null
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1149:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:999:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1149:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:999:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)

MichaIng, 2021-11-21:
Hmm, it generally works here on RPi Zero W, so not is not a general issue with the Node build for ARMv6, same v17.1.0. Let's see if we can get more details about at which module or dependency it fails exactly:
```sh
sudo -u nodered npm i --no-audit --verbose node-red
```

And as the modules named in the error trace are globally installed, maybe those are the issue, so let's also try to reinstall those:
```sh
npm i -g --unsafe-perm --no-audit socks-proxy-agent make-fetch-happen
```

fmuntean, 2021-11-21:
sudo -u nodered npm i --no-audit --verbose node-red

TypeError: Class extends value undefined is not a constructor or null
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1149:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:999:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1149:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:999:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)

npm i -g --unsafe-perm --no-audit socks-proxy-agent make-fetch-happen
TypeError: Class extends value undefined is not a constructor or null
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1149:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:999:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1149:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:999:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)

MichaIng, 2021-11-21:
So there is an issue with `node`, `npm` or one of the global modules and it is not related to Node-RED. Please try to start with a fresh Node instance:
```sh
dietpi-software uninstall 9
dietpi-software reinstall 122
```

fmuntean, 2021-11-21:
these two commands have run successfully.
However the node-red is not working:
Node-RED is not yet fully installed
/usr/bin/node-red-start: line 82: update-nodejs-and-nodered: command not found

MichaIng, 2021-11-21:
That is the wrong command location. Node-RED is installed as local module and running as system service, so you don't need to (and concurrently cannot) manually start it. Check service status:
```sh
systemctl status node-red
```
The executable is located at `/mnt/dietpi_userdata/node-red/node_modules/.bin/node-red`, but it is not intended to be manually called form console, and if so, only as user `nodered` user. The CLI command can however be called from console:
```sh
node-red-admin
```
It is an alias which calls the actual `node-red-admin` executable via `sudo -u nodered`.

Check out the docs: https://dietpi.com/docs/software/hardware_projects/#node-red

fmuntean, 2021-11-21:
Thank you for your help. 
After restarting the board the new node-red version started up.

just FYI: It seems that reinstalling node-red deleted my local project and had to reconfigure it from scratch.

Maybe in the future reinstall should keep the setting.

closed 2021-11-21 by fmuntean

MichaIng, 2021-11-21:
> just FYI: It seems that reinstalling node-red deleted my local project and had to reconfigure it from scratch.

Hmm, that is unexpected. I'll do some tests about that. Probably your Node-RED instance was a global module previously (also as you were expecting `/usr/bin/node-red-start`) and not fully migrated when reinstalling it as local module. The change was done at the beginning of this year with DietPi v7.0, and I need to review whether we did a migration automatically or expected a reinstall to do this cleanly. The data directory was always at `/mnt/dietpi_userdata/node-red` though.

Generally if you want to update Node-RED only, as stated in our docs, run the following commands:
```sh
systemctl stop node-red
cd /mnt/dietpi_userdata/node-red
sudo -u nodered npm up --no-audit node-red
systemctl start node-red
```
This should definitely preserve all data and configs.

If you want to update Node.js only:
```sh
dietpi-software reinstall 9
```

MDAR, 2021-12-30:
Just a bump to say that I experienced  the same thing just now on an Odroid C2, but after following these instructions, everything is working well.

Happy new year everyone

MichaIng, 2022-01-03:
@MDAR 
Was it exactly the same error message as OP reported, or which part?

MDAR, 2022-01-05:
> Just a bump to say that I experienced the same thing just now on an Odroid C2, but after following these instructions, everything is working well.
> 
> Happy new year everyone

😄 

Hard to answer that, given the nature of the OP.

and of course that I was looking for a solution to try, rather than reporting a fault afresh.

To be equally vague, I can tell you that this resolved the issue

>Generally if you want to update Node-RED only, as stated in our docs, run the following commands:
>
> ```
>systemctl stop node-red
>cd /mnt/dietpi_userdata/node-red
>sudo -u nodered npm up --no-audit node-red
>systemctl start node-red
>```
>This should definitely preserve all data and configs.
>
>If you want to update Node.js only:
>
>```
>dietpi-software reinstall 9
>```

To summarise the issue I got...

Odroid C2
Updated to latest DietPi versions
tried to update Node-RED using
```dietpi-software reinstall 122```
that just didn't work

found this

followed last step

**Everything working well now**
😄

MichaIng, 2022-01-05:
This error I meant:
```
TypeError: Class extends value undefined is not a constructor or null
...
```
However, must be somehow an issue when the Node.js version is updated above a certain version, leading to a Node-RED `npm` reinstall/update at least on first attempt.

MDAR, 2022-01-05:
> This error I meant:
> 
> ```
> TypeError: Class extends value undefined is not a constructor or null
> ...
> ```
> 
> However, must be somehow an issue when the Node.js version is updated above a certain version, leading to a Node-RED `npm` reinstall/update at least on first attempt.

I wish I could be more helpful, I just didn't pay attention enough to find the real error message.

I'll look in more detail if it shows itself again.

MichaIng, 2022-01-05:
> I just didn't pay attention enough to find the real error message.

No problem, at least you got it resolved 🙂.