MichaIng/DietPi #4974: DietPi-VPN installs OpenVPN on first run without asking

GvY85, 2021-11-19:
Good morning,

On my systems I manually installed Wireguard 2 years ago after using NordVPN via DietPi before that.
Yesterday I thought "lets see what this DietPi-VPN" actually is and does and started it.
I noticed that it instantly starts installing OpenVPN (that I dont use) and perhaps makes some other changes to the system?
After that was done I saw it offered nothing extra for my particular setup but after exiting I suddenly had OpenVPN installed and perhaps changes made to my system so I ended up restoring a DietPi-Backup and rebooting because my system felt "violated".

All in all no big deal but I think it would be better if DietPi-VPN would allow you to have a look at it to see if it is something for you before it starts makeing changes to the system. 
Perhaps an improvement?

MichaIng, 2021-11-19:
Many thanks for your report.

Strange, I was pretty sure that we already changed it to install OpenVPN on "Apply" instead of when starting the script. But we didn't 🤔. I'll do that, makes sense.

DietPi-VPN uses OpenVPN, like DietPi-NordVPN did before. This is the only protocol which can consistently be used across all providers. Generally APT packages can be uninstalled cleanly:
```sh
apt purge openvpn
```
and in `dietpi-vpn` you can use "Reset" to have all configurations removed and as well offered to uninstall OpenVPN.

MichaIng, 2021-11-19:
Done: https://github.com/MichaIng/DietPi/commit/05d9289

MichaIng, 2021-11-19:
> making system changes

There are btw no other system changes done. Our scripts are generally very transparent/verbose about what they are doing, so you can review everything in the command line history (though I know it cannot be scrolled up on all local consoles). Aside of the OpenVPN APT package install, files and settings are stored to `/etc/openvpn` and `/var/lib/dietpi/dietpi-vpn` only, and are completely removed removed when using the "Reset" option 🙂. I don't like left overs myself, so we did take care to be clean in this regards.

MichaIng, 2021-11-19:
I'll mark this as closed/solved.

closed 2021-11-19 by MichaIng