MichaIng/DietPi #4975: Auto redirect to wifi connection for radxa zero

almirus, 2021-11-19:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
Radxa Zero have only wifi module and doesn't LAN
On the first launch, I have to take a few steps before setting up the wifi (which is turned off by default)

#### Describe the solution you'd like:
When checking the network (first start), immediately launch dietpi-config and start wifi connection wizard

Joulinar, 2021-11-19:
Hi,

Basically you can activate WiFi before first initial boot already by setting the respective value inside `dietpi.txt`

MichaIng, 2021-11-19:
You can also enter `dietpi-config` from the error prompt when network connection is checked.

MichaIng, 2022-02-06:
Marking this as closed. Feel free to reopen when required.

closed 2022-02-06 by MichaIng