MichaIng/DietPi #4977: Slow Recursive DNS Resolving using pihole and unbound

AlexHuebi, 2021-11-19:
The following is a direct Copy/Paste from https://www.reddit.com/r/pihole/comments/qxgq2v/slow_recursive_dns_resolving_using_unbound/ with the hope that I might get here some help in resolving this too.

```
Hello everyone..

I've been using PiHole for a while now (and this Installation since ~4-5 Months on a 4B / 8GB Model - so absolutly Overkill but I've been running other stuff too). The performance (seemed) great until the last month or two. (Although it might be for longer but I didn't monitor the performance for longer than the 1-2 mo. )

So since I've noticed delays while opening *some* websites and not on others, I decided to Test the Resolution Speed. Cached Speeds were perfect at around 3 ms while Uncached results did take ~2 seconds on Average with peaks at ~6 Seconds. I would expect an avg. performance arround 200-300 ms.

Thats when I decided that something should change and I did some research and found this: https://www.reddit.com/r/pihole/comments/d9j1z6/unbound_as_recursive_dns_server_slow_performance/

After one Week, the performance still hasn't improved ( https://imgur.com/a/4gBdaiH ).

Performance Compared to using Cloudflare via PiHole ( https://imgur.com/a/GBzU37w )

Note: Both Screenshots were made with PiHole's Cache Size set to 0. Software thats beeing used is DNSBench from Precision Freeware.

So thats why I am here.
The System Running PiHole is:
As OS: DietPi on the current release.
RasPi 4B / 8 GB Model
500GB Boot SSD (Nextcloud is also installed as a Datashare)
100/10 Mbit connection with a relativly low latency and high reliability.
(No SD card installed)

I've installed PiHole and Unbound via the DietPI interface and configured each like shown in this Guide ( https://docs.pi-hole.net/guides/dns/unbound/ ). (edit: although the config was already set by the DietPi Installer)

All installed Software is on the latest Version.Average CPU and RAM usage is <10%.

My guess would be that the Problem lies somewhere at Unbound but since it seems like there isn't an official forum, I've got to ask here: What can I do to improve the performance?

The Unbound Config File currently in use is here: https://pastebin.com/tB3UERaE
```
Additional Info: 
Installed Packets: 
 44   Transmission: BitTorrent server with web interface (C)
114  Nextcloud: File sync, sharing and collaboration platform
115  Webmin: web interface system management
73   Fail2Ban: prevents brute-force attacks with ip ban
81   LLSP: Lighttpd | SQLite  | PHP
82   LLMP: Lighttpd | MariaDB | PHP
92   Certbot: Obtain and renew Let's Encrypt SSL certs for HTTPS
93   Pi-hole: block adverts for any device on your network
182  Unbound: validating, recursive, caching DNS resolver
96   Samba Server: Feature-rich SMB/CIFS server
0    OpenSSH Client: Feature-rich SSH, SFTP and SCP client
87   SQLite: Persistent single-file database system
88   MariaDB: Persistent cached file-per-table database server
91   Redis: Volatile in-memory non-SQL database server
16   Build-Essential: GNU C/C++ compiler, development libraries and headers
17   Git: Clone and manage Git repositories locally
130  Python 3: Runtime system, pip package installer and development headers

Thanks in advance for any help.

Joulinar, 2021-11-19:
You could try using a different DNS server inside your Pihole configuration. Just uncheck the custom entry and select Quad9 + Cloudflare for example. This way you could see if it improves.

MichaIng, 2021-11-19:
If I understand, a direct upstream DNS indeed helps, but Unbound is used as recursive resolver. If you used https://docs.pi-hole.net/guides/dns/unbound/ on top of the existing Unbound configuration, please try to revert this first to assure it is slow with the default configuration we apply.

There is no need and no benefit in disabling the Pi-hole cache. It doesn't cost something and implies the chance that one hop is skipped when Pi-hole serves from cache instead of forwarding to Unbound first. But in your case it's about uncached queries, so that is not directly related.

One thing you could try is disable DNSSEC validation completely. For this:
```sh
echo -e 'server:\n    module-config: "iterator"' > /etc/unbound/unbound.conf.d/disable-dnssec.conf
systemctl restart unbound
```

But this is no production solution, as DNSSEC is a major brick in DNS security and should definitely be enabled, this is only to identify which part delays the query.

AlexHuebi, 2021-11-19:
@Joulinar 
> You could try using a different DNS server inside your Pihole configuration. Just uncheck the custom entry and select Quad9 + Cloudflare for example. This way you could see if it improves.
Already did that - Performance was as expected at ~60ms resolve time

Joulinar, 2021-11-19:
@MichaIng 
DNSSEC is something done by unbound already. I guess it doesn't make sense having it activate in Pihole as well.

AlexHuebi, 2021-11-19:
@MichaIng 
> If I understand, a direct upstream DNS indeed helps, but Unbound is used as recursive resolver. If you used https://docs.pi-hole.net/guides/dns/unbound/ on top of the existing Unbound configuration, please try to revert this first to assure it is slow with the default configuration we apply.

I forgot to edit the Post, I did Check the config and it seemed like the one you provide (might not have changed anything). Where can I download the original file to compare it? 

> One thing you could try is to disable DNSSEC validation completely. For this:
> 
> ```shell
> echo -e 'server:\n    module-config: "iterator"' > /etc/unbound/unbound.conf.d/disable-dnssec.conf
> systemctl restart unbound
> ```

I'll try that and update you asap.

AlexHuebi, 2021-11-19:
Pre-Change:
[image]

After the Change:
[image]

Reverting the Change:
[image]

AlexHuebi, 2021-11-19:
One thing to Note: It seems that theres a difference in Speed when I changed PiHole from Unbound to Cloudflare and back again. This *change* does not persist since I rebooted the Pi last week where it showed a similar behaviour afterwards.

MichaIng, 2021-11-19:
> DNSSEC is something done by unbound already. I guess it doesn't make sense having it activate in Pihole as well. Or you mean to disable on Unbound? Probably I misunderstood your comment.

Where did I write that DNSSEC should be enabled in Pi-hole 🤔? I mean to test disabling it in Unbound 😄.

> Where can I download the original file to compare it?

https://github.com/MichaIng/DietPi/blob/dev/.conf/dps_182/unbound.conf
But the port is changed to `5335` and the interface bound to `127.0.0.1` when Pi-hole is installed, and `do-ip6` disabled when IPv6 is disabled.

> It seems that theres a difference in Speed when I changed PiHole from Unbound to Cloudflare and back again.

That is not possible, as long as it is not related a the cache in Unbound or Pi-hole. There are natural variations of course, and like you can see in your three tests, if a network connection was established once and frequently has an effect on subsequent connection speeds as well. A similar average below 200 ms in all three cases looks pretty fine now?

Joulinar, 2021-11-19:
Basically I did the same test now. And using PiHole + Unbound is on average even faster than using Cloudflare directly

[image]

MichaIng, 2021-11-20:
That is indeed possible if by chance the closest/chosen upstream provider server is not so close or temporarily slow. The amount of stations which DNS queries pass is the same in both cases:
```
client => Pi-hole => Unbound => DNS root
client => Pi-hole => Cloudflare => DNS root
```

AlexHuebi, 2021-11-20:
SO.... I just got to comparing the currently set unbound config and the preset made by you and the differences are:
Increased num-threads to 2
Commented out the verbosity level
added: prefer-ip6: no

the rest is the same. 

Also I tested if it makes a difference when I use another pi.
Current "Production" Server (Pi4)
[image]

Testserver (Pi3)
[image]

@Joulinar 
Both are setup to use Unbound. So I am wondering why your Pi is as fast/ if not faster than Cloudflare 🤔 
(My Cloudflare speed:)
[image]

AlexHuebi, 2021-11-20:
I completly ignored the fact that I can test both nameservers at the same time and thus might eliminate some variables. 
[image]

So it seems that my Internet might play a bigger role in that than I thought.. Which seems a bit wierd since a mean rtt of ~10-12 should be fast enough.

Joulinar, 2021-11-20:
Could it be you are loosing packages somewhere? Because reliability is around 90% only if using local DNS server. I would expect 100%. I guess it has a huge impact on the max value.

AlexHuebi, 2021-11-20:
I think that the Packetloss origins from PiHole blocking some Requests: 
[image]

The rest of the log seems clean.

But I tried some Packetloss tests anyway:
https://devicetests.com/packet-loss-test
[image]

and
https://packetlosstest.com/
[image]

I am not sure though how reliable these kinds of tests are.

Joulinar, 2021-11-20:
My Pihole is blocking around 1.6 million domains and I have a reliability of 100% thinking.

AlexHuebi, 2021-11-23:
Since (for whatever reason) the Resolving seems to be back at speed, I'll close the Issue and might reopen another one referencing this one if it comes back. 

[image]

Thanks a lot for the suggestions!
Cheers!

closed 2021-11-23 by AlexHuebi