MichaIng/DietPi #4980: ASUS Tinker Board - Can't run docker containers on Bullseye

jroblesruiz, 2021-11-20:
### Creating a bug report/issue

#### Details:
- Date           | Sat Nov 20 16:16:20 CST 2021
- Bug report     | fa9de131-1a53-4064-9802-1771cbde2f7e
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | ASUS Tinker Board (armv7l) (ID=52)
- Kernel version | `Linux econ-tb 5.10.60-rockchip #21.08.1 SMP PREEMPT Wed Aug 25 19:43:43 UTC 2021 armv7l GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `docker run -d -p 9002:9000 --name=portainer --restart=always -v /run/docker.sock:/var/run/docker.sock -v portainer_data:/data portainer/portainer-ce`
- Exit code      | 125
- Software title | DietPi-Software
#### Steps to reproduce:

1. Install DietPi_ASUSTB-ARMv7-Bullseye.img (10/22/2021), new image on a fresh micro-SD.
2. Complete setup process.
3. Run dietpi-software and install Docker, Docker-compose and Portainer.
#### Expected behaviour:

- Portainer should be installed and run/started by docker.
#### Actual behaviour:

- Setup ends with an error, and if you try to start Portainer with `docker start Portainer`, you receive the following error. The same happened with every docker container I tried to start, including Hello-word container) 
`Error response from daemon: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: process_linux.go:508: setting cgroup config for procHooks process caused: bpf_prog_query(BPF_CGROUP_DEVICE) failed: function not implemented: unknown
Error: failed to start containers: portainer`
#### Extra details:

- I searched for the error and similar posts points that's a compatibility problem witch cgroup v2, and that you must edit /boot/boot.ini with this line `systemd.unified_cgroup_hierarchy=0` but on my Tinker Board I don't have that file and I don't find any information on how to set kernel boot options on the Tinker Board with DietPi.
#### Additional logs:
```
docker: Error response from daemon: Conflict. The container name "/portainer" is already in use by container "17d1487760eb0665cf320048f184ec795c413737103d88fa07793c70509754cd". You have to remove (or rename) that container to be able to reuse that name.
See 'docker run --help'.
```
Regards

MichaIng, 2021-11-20:
Many thanks for your report.

Was the kernel upgraded and you did not yet reboot?

```
bpf_prog_query(BPF_CGROUP_DEVICE) failed: function not implemented
```
Hmm, that should be definitely the case with the used Armbian kernel 🤔.

It is expected that calling `docker run` two times cannot work as the container is deployed already. The deployment went well, only the container startup failed. In such case it makes sense to ignore the error and fix/debug container startup afterwards, I'd say. You can do that via "Change command" and replacing everything with a colon `:` or `true`.

jroblesruiz, 2021-11-21:
Hi MichaIng, thanks for your response.

I did reboot my board, and I'm using the kernel shipped with the bullseye image.

Its correct what you said, docker installs correctly, but it can't run any container, even the simplest one as the `hello-world` one.

```
$ docker run hello-world
Unable to find image 'hello-world:latest' locally
latest: Pulling from library/hello-world
9b157615502d: Pull complete
Digest: sha256:cc15c5b292d8525effc0f89cb299f1804f3a725c8d05e158653a563f15e4f685
Status: Downloaded newer image for hello-world:latest
docker: Error response from daemon: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: process_linux.go:508: setting cgroup config for procHooks process caused: bpf_prog_query(BPF_CGROUP_DEVICE) failed: function not implemented: unknown.
ERRO[0003] error waiting for container: context canceled
```
If you point me in the right direction to set the kernel parameter `sytemd.unified_cgroup_hierarchy=0` I can try It, but I don't know how to doit on the Tinker Board as /boot/boot.ini file doesn't exist, I've tried creating It but didn't work for me.

MichaIng, 2021-11-21:
Confusing, I never heard of any such issue with the Armbian kernel, and this is the generic 32-bit Rockchip kernel image, hence not something specific for ASUS Tinker Board 🤔. Also I cannot find any report on the Armbian forum about this: https://forum.armbian.com/forum/21-rockchip-3288-3328/

This shouldn't trigger this error, but can you check whether `docker_optimization=on` has been added to `/boot/armbianEnv.txt`?

I checked the kernel configurations for both, the Rockchip 32-bit and 64-bit kernel images from Armbian and indeed `BPF_CGROUP_DEVICE` is not set explicitly in either of them.

> as /boot/boot.ini file doesn't exist

The new U-Boot does not use `boot.ini` but `boot.scr`. Generally you can edit `boot.cmd` the first and use the command provided at the bottom of the file to compile `boot.scr`. But a safer way to add boot parameters it the `armbianEnv.txt` which is loaded automatically:
```sh
sed -i '/^extraargs=/"$/ sytemd.unified_cgroup_hierarchy=0"/' /boot/armbianEnv.txt
```

jroblesruiz, 2021-11-21:
Thanks, I edited /boot/boot.cmd and regenerate boot.src as you mentioned, adding the ystemd.unified_cgroup_hierarchy=0 parameter. Now if I run `cat /proc/cmdline` I get:

Here I found some useful steps, but the one I was missing was where to add it and how to regenerate the boot.src

```
earlyprintk root=UUID=f9d46ad4-5004-44cd-9239-a862108641c5 rootwait rootfstype=ext4  consoleblank=0 loglevel=4 ubootpart=ecdd8161-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u "net.ifnames=0"  systemd.unified_cgroup_hierarchy=0 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
```
And now I can run dockers containers successfully.

closed 2021-11-21 by jroblesruiz

onorbe, 2022-10-08:
I have the same issue.
Someone can explain me the steps to solve it?
Thanks in advanced