MichaIng/DietPi #4981: Nextcloud not upgrading

jahway603, 2021-11-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applied'
G_LIVE_PATCH_STATUS[1]='not applied'
G_LIVE_PATCH_STATUS[2]='not applied'
- Distro version | bullseye
- Kernel version | Linux solarship 4.14.241+ #1 SMP PREEMPT Wed Jul 28 16:55:16 UTC 2021 armv7l GNU/Linux
- SBC model | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)
- Power supply used | 12v2a
- SDcard used | no SDcard, USB hard drive has boot and rootfs

#### Additional Information (if applicable)
- Software title | Nextcloud
- Was the software title installed freshly or updated/migrated? Attempt at updating it
- Can this issue be replicated on a fresh installation of DietPi? As I do not have another of these boards, I tried the x86 VirtualBox image and it successfully installed the correct latest-stable, 22.2.3

- Bug report ID | N/A

#### Steps to reproduce

1. ran `sudo dietpi-software reinstall 114` to reinstall Nextcloud, with the intention of upgrading to the latest-stable 22.2.3 as 22.1.x was currently installed and they are both in the same branch (22.x)
1. After the reinstall, the version remained unchanged and was not the latest-stable

#### Expected behaviour

- After the reinstall, the version would be the latest-stable, which is currently 22.2.3

#### Actual behaviour

- After the reinstall, the version of Nextcloud is not updated and remains unchanged

#### Extra details

- The logic in dietpi-sofware around line 5000 is not pulling the latest if you have it already installed, leaving a user not able to upgrade. Unless there's another way to do this via dietpi- scripts that I'm not aware of.

Joulinar, 2021-11-20:
Usually NC is updated via internal update function inside web UI. As well you could run following to check

`ncc update:check`

MichaIng, 2021-11-20:
Yes, upgrading via `dietpi-software reinstall` is prevented intentionally as an upgrade involves migration steps and is done pretty well with the internal updater. There is also a notification printed during the reinstall about this.

If you need to do it via console:
```sh
sudo -u www-data php /var/www/nextcloud/updater/updater.phar
```

> Unless there's another way to do this via dietpi- scripts that I'm not aware of.

Generally when there is an internal updater (e.g. `pihole -up`) or the software was installed with any package manager (`apt`, `pip`, `npm`, ...), then this is the way to go. In case of package managers, also a `dietpi-software reinstall` implies the upgrade, but is usually non-required overhead around the actual package manager call. When unsure, our online documentations contains update info for most software titles, e.g. here for Nextcloud: https://dietpi.com/docs/software/cloud/#nextcloud

jahway603, 2021-11-21:
Thank you both. I was able to upgrade to the latest version using that console command being run as www-data that @MichaIng listd.

closed 2021-11-21 by jahway603