MichaIng/DietPi #4982: Issue while trying to upgrade to bullseye

rsmith98146, 2021-11-21:
##### Details:
- Date           | Sat Nov 20 20:45:58 PST 2021
- Bug report     | 965a26d9-dacc-467a-8874-2bdba59ebdba
- DietPi version | v7.5.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | TinkerOS 2.0.7
- Hardware       | ASUS Tinker Board (armv7l) (ID=52)
- Kernel version | `Linux DietPi 4.4.103+ #1 SMP Fri Jun 22 16:14:49 CST 2018 armv7l GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `apt-get --allow-releaseinfo-change-suite -q update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://downloads.plex.tv/repo/deb public InRelease
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Ign:6 http://repo.linaro.org/ubuntu/linaro-overlay bullseye InRelease
Err:7 http://repo.linaro.org/ubuntu/linaro-overlay bullseye Release
  404  Not Found [IP: 54.192.73.115 80]
Reading package lists...
E: The repository 'http://repo.linaro.org/ubuntu/linaro-overlay bullseye Release' does not have a Release file.
```

Joulinar, 2021-11-21:
The issue is with the 3rd party repository you have added manually.

`http://repo.linaro.org/ubuntu/linaro-overlay`

You would need to check with the provider what the correct repository might be on Bullseye. Furthermore at the moment it is pointing to Ubuntu, which is basically incorrect as DietPi is based on Debian.

A small Google search was pointing me to this

`http://obs.linaro.org/linaro-overlay-bullseye/`

MichaIng, 2021-11-21:
Btw, you run a very old image which is based on TinkerOS with an old Linux version 4.4 and it seems linke the Linaro repository was used for some kernel-close software and driver packages.

If you find some more time, I recommend to flash our new image based on Armbian with Linux 5.10: https://dietpi.com/downloads/images/

You could also try to migrate your current system if you have a full SD card/drive backup, but at least we'd need to check whether the partitioning is correct:
```sh
fdisk -l "$(lsblk -npo PKNAME "$G_ROOTFS_DEV")"
```

MichaIng, 2022-01-04:
I'll mark this as closed. Feel free to reopen when required.

closed 2022-01-04 by MichaIng