MichaIng/DietPi #4983: Failure on apache reinstall when upgrading to bullseye

jdhaggerty, 2021-11-21:
During the apache reinstall I get these errors:
E: Package 'php7.4-apcu' has no installation candidate
apache2_invoke: Enable module php7.4
apache2_reload: Your configuration is broken. Not restarting Apache 2
apache2_reload: [Sun Nov 21 08:46:17.305567 2021] [:crit] [pid 19432] (38)Function not implemented: AH00141: Could not initialize random number generator

 DietPi v7.8.2
 Device model : Odroid C2 (aarch64)

MichaIng, 2021-11-21:
Many thanks for your report.

Did you follow https://dietpi.com/blog/?p=811#2.2-manual-upgrade?

Please show the output of the following commands:
```sh
apt update
apt policy php-apcu
apt install php7.4-apcu
```

jdhaggerty, 2021-11-21:
Yes, this is following the manual upgrade instructions.

# apt update
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://dietpi.com/meveric all InRelease
Hit:6 https://dietpi.com/meveric bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

# apt policy php7.4-apcu
php7.4-apcu:
  Installed: (none)
  Candidate: (none)
  Version table:
     5.1.19+4.0.11-7+0~20210222.23+debian9~1.gbp08eb57 -1
        100 /var/lib/dpkg/status

# apt install php7.4-apcu
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package php7.4-apcu is a virtual package provided by:
  php-apcu 5.1.19+4.0.11-3 [Not candidate version]

E: Package 'php7.4-apcu' has no installation candidate

MichaIng, 2021-11-21:
I edited the second command, please show the following:
```sh
apt policy php-apcu
```
It is picked as expected by `apt install php7.4-apcu` but it looks like an APT preference blocks it from being installed, probably an obsolete APT pinning, let's see.

jdhaggerty, 2021-11-23:
```
apt policy php-apcu
php-apcu:
  Installed: 5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d
  Candidate: 5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d
  Version table:
 *** 5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d 100
        100 /var/lib/dpkg/status
     5.1.19+4.0.11-3 500
        500 https://deb.debian.org/debian bullseye/main arm64 Packages
```
Changing the apache reinstall command to use php-acpu instead of php7.4-acpu allows the reinstall to proceed, but both mariadb and apache don't start:

```
systemctl status apache2.service
● apache2.service - The Apache HTTP Server
     Loaded: loaded (/lib/systemd/system/apache2.service; disabled; vendor>
     Active: failed (Result: exit-code) since Tue 2021-11-23 10:09:42 PST;>
       Docs: https://httpd.apache.org/docs/2.4/
    Process: 22090 ExecStart=/usr/sbin/apachectl start (code=exited, statu>

Nov 23 10:09:42 OdroidPi systemd[1]: Starting The Apache HTTP Server...
Nov 23 10:09:42 OdroidPi apachectl[22093]: [Tue Nov 23 10:09:42.324477 202>
Nov 23 10:09:42 OdroidPi apachectl[22090]: Action 'start' failed.
Nov 23 10:09:42 OdroidPi apachectl[22090]: The Apache error log may have m>
Nov 23 10:09:42 OdroidPi systemd[1]: apache2.service: Control process exit>
Nov 23 10:09:42 OdroidPi systemd[1]: apache2.service: Failed with result '>
Nov 23 10:09:42 OdroidPi systemd[1]: Failed to start The Apache HTTP Serve>
```

```
systemctl status mysql.service
● mariadb.service - MariaDB 10.5.12 database server
     Loaded: loaded (/lib/systemd/system/mariadb.service; disabled; vendo>
     Active: activating (auto-restart) (Result: signal) since Tue 2021-11>
       Docs: man:mariadbd(8)
             https://mariadb.com/kb/en/library/systemd/
    Process: 25639 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root >
    Process: 25641 ExecStartPre=/bin/sh -c systemctl unset-environment _W>
    Process: 25644 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recover>
    Process: 25692 ExecStart=/usr/sbin/mariadbd $MYSQLD_OPTS $_WSREP_NEW_>
   Main PID: 25692 (code=killed, signal=ILL)
```
Error logs are empty.

Joulinar, 2021-11-23:
MariaDB service usually has a different name and since Bullseye, error.log file has been disabled by default by MariaDB developer. Pls can you share

```
systemctl restart mariadb.service
journalctl -u mariadb
```

jdhaggerty, 2021-11-23:
```
systemctl restart mariadb.service
Job for mariadb.service failed because a fatal signal was delivered to the control process.
See "systemctl status mariadb.service" and "journalctl -xe" for details.

journalctl -xe
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A stop job for unit mariadb.service has finished.
░░
░░ The job identifier is 15009 and the job result is done.
Nov 23 11:52:22 OdroidPi systemd[1]: Starting MariaDB 10.5.12 database server...
░░ Subject: A start job for unit mariadb.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit mariadb.service has begun execution.
░░
░░ The job identifier is 15009.
Nov 23 11:52:23 OdroidPi systemd[1]: mariadb.service: Main process exited, code=killed, status=4/ILL
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit mariadb.service has exited.
░░
░░ The process' exit code is 'killed' and its exit status is 4.
Nov 23 11:52:23 OdroidPi systemd[1]: mariadb.service: Failed with result 'signal'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit mariadb.service has entered the 'failed' state with result 'signal'.
Nov 23 11:52:23 OdroidPi systemd[1]: Failed to start MariaDB 10.5.12 database server.
░░ Subject: A start job for unit mariadb.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit mariadb.service has finished with a failure.
░░
░░ The job identifier is 15009 and the job result is failed.
```

MichaIng, 2021-11-23:
Please show the full service log as stated, otherwise we do not see more than that it exited with `SIGILL`:
```sh
journalctl -u mariadb
```

Joulinar, 2021-11-23:
ok let's activate error.log to get to more details

```
sed -i 's/#log_error/log_error/' /etc/mysql/mariadb.conf.d/50-server.cnf
systemctl restart mariadb.service
cat /var/log/mysql/error.log
```

MichaIng, 2021-11-23:
That doesn't change something as without `log_error` MariaDB simply prints everything to syslog/journal. But yeah, other than I thought `journalctl -xe` does not show only error messages and we can see that service start log entry.

Another idea is to increase the log level:
```sh
G_CONFIG_INJECT 'log_warnings=' 'log_warnings=3' /etc/mysql/mariadb.conf.d/50-server.cnf
```

Joulinar, 2021-11-23:
@MichaIng 
for me it remove the `#` and activate the `error.log`

```
root@DietPi4:~# cat /etc/mysql/mariadb.conf.d/50-server.cnf | grep log_error
log_error = /var/log/mysql/error.log
root@DietPi4:~#
```

MichaIng, 2021-11-23:
As said, the error log is done in every case, just to STDERR => journald if no log file is defined.

jdhaggerty, 2021-11-27:
Doesn't seem to be all that useful:
```
Nov 27 11:56:23 OdroidPi systemd[1]: mariadb.service: Scheduled restart job, restart counter is at 42.
Nov 27 11:56:23 OdroidPi systemd[1]: Stopped MariaDB 10.5.12 database server.
Nov 27 11:56:23 OdroidPi systemd[1]: Starting MariaDB 10.5.12 database server...
Nov 27 11:56:24 OdroidPi systemd[1]: mariadb.service: Main process exited, code=killed, status=4/ILL
Nov 27 11:56:24 OdroidPi systemd[1]: mariadb.service: Failed with result 'signal'.
Nov 27 11:56:24 OdroidPi systemd[1]: Failed to start MariaDB 10.5.12 database server.
```

MichaIng, 2021-11-27:
[Raising the log_level](https://github.com/MichaIng/DietPi/issues/4983#issuecomment-977192197) didn't change something about that?

SIGILL isn't something I've seen before, especially since it seems to run into this before starting up any actual database engine. Probably the binaries are broken. Please try to reinstall every related package:
```sh
systemctl disable --now mariadb
apt install --reinstall mariadb-server mariadb-server-10.5 mariadb-common mariadb-client-10.5 mariadb-server-core-10.5 mariadb-client-core-10.5 mysql-common libmariadb3
```

MichaIng, 2022-01-04:
Did the above help?

MichaIng, 2022-02-06:
Marking as closed due to outstanding reply. Feel free to reopen when required.

MichaIng, 2022-03-17:
>  The fact that it's an illegal instruction seems to imply that it might have been compiled wrong for this chipset.

This is however very unlikely as long as you did not replace the Debian repository with something invalid manually, especially with an aarch64 SoC which is compatible with all ARM builds. I guess it is some corruption elsewhere, probably a library used by MariaDB.

Please also raise the log level to get more details, as suggested above.

jdhaggerty, 2022-03-17:
Turning on logging just generates an empty log file. It appears that it hits the illegal instruction pretty early on during program load.

MichaIng, 2022-03-17:
I wonder whether there are more packages for Stretch installed, like the `php7.4-apcu`. We should clean this up first:
```sh
dpkg -l | grep -E 'debian9|bpo9'
```

jdhaggerty, 2022-03-17:
ii  libpcre3:arm64                   2:8.44-2+0~20210301.9+debian9~1.gbpa278ad         arm64        Perl 5 Compatible Regular Expression Library - runtime files
ii  libpcre3-dev:arm64               2:8.44-2+0~20210301.9+debian9~1.gbpa278ad         arm64        Perl 5 Compatible Regular Expression Library - development files
ii  libpcre32-3:arm64                2:8.44-2+0~20210301.9+debian9~1.gbpa278ad         arm64        Perl 5 Compatible Regular Expression Library - 32 bit runtime files
ii  libpcrecpp0v5:arm64              2:8.44-2+0~20210301.9+debian9~1.gbpa278ad         arm64        Perl 5 Compatible Regular Expression Library - C++ runtime files
ii  libsodium23:arm64                1.0.18-1+0~20191009.1+debian9~1.gbpb6823f         arm64        Network communication, cryptography and signaturing library
ii  libxml2:arm64                    2.9.12+dfsg-0+0~20210621.9+debian9~1.gbp43e861    arm64        GNOME XML library
ii  libxml2-dev:arm64                2.9.12+dfsg-0+0~20210621.9+debian9~1.gbp43e861    arm64        Development files for the GNOME XML library
ii  libzip4:arm64                    1.7.3-1+0~20210114.10+debian9~1.gbp4c125d         arm64        library for reading, creating, and modifying zip archives (runtime)
ii  php-apcu                         5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d arm64        APC User Cache for PHP
ii  php-common                       2:84+0~20210621.36+debian9~1.gbp28513e            all          Common files for PHP packages
ii  php-gd                           2:8.0+84+0~20210621.36+debian9~1.gbp28513e        all          GD module for PHP [default]
ii  php-mbstring                     2:8.0+84+0~20210621.36+debian9~1.gbp28513e        all          MBSTRING module for PHP [default]
ii  php-mysql                        2:8.0+84+0~20210621.36+debian9~1.gbp28513e        all          MySQL module for PHP [default]
ii  php-xml                          2:8.0+84+0~20210621.36+debian9~1.gbp28513e        all          DOM, SimpleXML, WDDX, XML, and XSL module for PHP [default]
ii  php-zip                          2:8.0+84+0~20210621.36+debian9~1.gbp28513e        all          Zip module for PHP [default]
rc  php5.6-apcu                      5.1.19+4.0.11-7+0~20210222.23+debian9~1.gbp08eb57 arm64        APC User Cache for PHP
rc  php5.6-common                    5.6.40-47+0~20210227.51+debian9~1.gbpd522a2       arm64        documentation, examples and common module for PHP
rc  php7.0-apcu                      5.1.19+4.0.11-7+0~20210222.23+debian9~1.gbp08eb57 arm64        APC User Cache for PHP
rc  php7.0-common                    7.0.33-45+0~20210220.50+debian9~1.gbp703a20       arm64        documentation, examples and common module for PHP
rc  php7.1-apcu                      5.1.19+4.0.11-7+0~20210222.23+debian9~1.gbp08eb57 arm64        APC User Cache for PHP
rc  php7.1-common                    7.1.33-34+0~20210228.54+debian9~1.gbpb1fb69       arm64        documentation, examples and common module for PHP
rc  php7.2-apcu                      5.1.19+4.0.11-7+0~20210222.23+debian9~1.gbp08eb57 arm64        APC User Cache for PHP
rc  php7.2-common                    7.2.34-18+0~20210223.60+debian9~1.gbpb21322       arm64        documentation, examples and common module for PHP
rc  php7.3-apcu                      5.1.19+4.0.11-7+0~20210222.23+debian9~1.gbp08eb57 arm64        APC User Cache for PHP
rc  php7.4-apcu                      5.1.19+4.0.11-7+0~20210222.23+debian9~1.gbp08eb57 arm64        APC User Cache for PHP
rc  php7.4-cli                       7.4.15-2+0~20210212.37+debian9~1.gbp5dee47        arm64        command-line interpreter for the PHP scripting language
rc  php7.4-gd                        7.4.15-2+0~20210212.37+debian9~1.gbp5dee47        arm64        GD module for PHP
rc  php7.4-json                      7.4.15-2+0~20210212.37+debian9~1.gbp5dee47        arm64        JSON module for PHP
rc  php7.4-mbstring                  7.4.15-2+0~20210212.37+debian9~1.gbp5dee47        arm64        MBSTRING module for PHP
rc  php7.4-mysql                     7.4.15-2+0~20210212.37+debian9~1.gbp5dee47        arm64        MySQL module for PHP
rc  php7.4-opcache                   7.4.15-2+0~20210212.37+debian9~1.gbp5dee47        arm64        Zend OpCache module for PHP
rc  php7.4-readline                  7.4.15-2+0~20210212.37+debian9~1.gbp5dee47        arm64        readline module for PHP
rc  php7.4-xml                       7.4.15-2+0~20210212.37+debian9~1.gbp5dee47        arm64        DOM, SimpleXML, XML, and XSL module for PHP
rc  php7.4-zip                       7.4.15-2+0~20210212.37+debian9~1.gbp5dee47        arm64        Zip module for PHP
ii  php8.0-apcu                      5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d arm64        APC User Cache for PHP
ii  php8.0-common                    8.0.10-1+0~20210826.23+debian9~1.gbp9ab3fd        arm64        documentation, examples and common module for PHP
rc  php8.0-curl                      8.0.10-1+0~20210826.23+debian9~1.gbp9ab3fd        arm64        CURL module for PHP
ii  php8.0-gd                        8.0.10-1+0~20210826.23+debian9~1.gbp9ab3fd        arm64        GD module for PHP
ii  php8.0-mbstring                  8.0.10-1+0~20210826.23+debian9~1.gbp9ab3fd        arm64        MBSTRING module for PHP
ii  php8.0-mysql                     8.0.10-1+0~20210826.23+debian9~1.gbp9ab3fd        arm64        MySQL module for PHP
ii  php8.0-xml                       8.0.10-1+0~20210826.23+debian9~1.gbp9ab3fd        arm64        DOM, SimpleXML, XML, and XSL module for PHP
ii  php8.0-zip                       8.0.10-1+0~20210826.23+debian9~1.gbp9ab3fd        arm64        Zip module for PHP

MichaIng, 2022-03-18:
Okay that became a mess. Let's clean this up. Which web application do your use which uses PHP and MariaDB?

Currently you seem to use PHP 8.0, but Apache2 should be configure to use PHP 7.4, being part of the conflict. Original reason for the unintended PHP version install was that you or another installer installed the non-versioned "php-*" packages instead of versioned "php7.3-*" like ones, and the non-versioned ones were a moving target with Ondrej's PHP package repository, which we used on Debian Stretch to keep up support (which was not a good idea, it turned out).

I suggest to stay with PHP 7.4, which is natively shipped by Debian Bullseye. Let's purge all of them first and reinstall afterwards:
```sh
apt autopurge 'php-*' 'php5.6-*' 'php7.0-*' 'php7.1-*' 'php7.2-*' 'php7.3-*' 'php7.4-*' 'php8.0-*'
```
Then the other packages, first autoremoving what is obsolete:
```sh
apt-mark auto libpcre3 libpcre3-dev libpcre32-3 libpcrecpp0v5 libsodium23 libxml2 libxml2-dev libzip4
apt autopurge
dpkg -l | grep -E 'debian9|bpo9'
```

jdhaggerty, 2022-03-18:
ii  libargon2-1:arm64                0~20190702-0.1+0~20190710.3+debian9~1.gbp2fb167 arm64        memory-hard hashing function - runtime library
ii  libicu65:arm64                   65.1-1+0~20200223.8+debian9~1.gbp519cf3         arm64        International Components for Unicode
ii  libpcre2-8-0:arm64               10.36-2+0~20210212.6+debian9~1.gbp6138a4        arm64        New Perl Compatible Regular Expression Library- 8 bit runtime files
ii  libpcre3:arm64                   2:8.44-2+0~20210301.9+debian9~1.gbpa278ad       arm64        Perl 5 Compatible Regular Expression Library - runtime files
ii  libpcrecpp0v5:arm64              2:8.44-2+0~20210301.9+debian9~1.gbpa278ad       arm64        Perl 5 Compatible Regular Expression Library - C++ runtime files
ii  libsodium23:arm64                1.0.18-1+0~20191009.1+debian9~1.gbpb6823f       arm64        Network communication, cryptography and signaturing library
ii  libxml2:arm64                    2.9.12+dfsg-0+0~20210621.9+debian9~1.gbp43e861  arm64        GNOME XML library

BTW, I switched from apache to nginx because I never could get apache to start. It kept saying that it couldn't connect to the random number generator, and I figured that this kernel is so old that apache had introduced an incompatibility somewhere. nginx starts fine. I have also switched back to php 7.3 because php 7.4 doesn't seem to have php-curl in it, and it would be nice to have that module.

Anyway, it's hard to think of a way that any php installation, no matter how messed up, could cause mariadb to crash on initialization with an illegal instruction.

MichaIng, 2022-03-19:
Okay, let's try to downgrade those packages to the ones from Debian Bullseye, assured to be compatible:
```sh
apt install --reinstall --allow-downgrades 'libpcre3=2:8.39-13' 'libpcrecpp0v5=2:8.39-13' 'libsodium23=1.0.18-1' 'libxml2=2.9.10+dfsg-6.7'
```
And let's see whether the others can be autoremoved as well, they were not shown in your first list:
```sh
apt-mark auto libargon2-1 libicu65 libpcre2-8-0
apt autopurge
dpkg -l | grep 'debian9'
```

jdhaggerty, 2022-03-20:
ii  libargon2-1:arm64                0~20190702-0.1+0~20190710.3+debian9~1.gbp2fb167 arm64        memory-hard hashing function - runtime library
ii  libpcre2-8-0:arm64               10.36-2+0~20210212.6+debian9~1.gbp6138a4        arm64        New Perl Compatible Regular Expression Library- 8 bit runtime files

Same illegal instruction clash after a reboot, sadly.

MichaIng, 2022-03-20:
Two more packages to switch to Bullseye version, and reinstall PHP:
```sh
apt install --reinstall --allow-downgrades 'libargon2-1=0~20171227-0.2' 'libpcre2-8-0=10.36-2'
dietpi-software reinstall 89
```

jdhaggerty, 2022-03-20:
I am running several low-volume test and production wordpress sites. So while it was possible to switch from Apache to nginx, both PHP and MariaDB are required. (There is some sketchy plugin that hasn't been updated for 8 years and was removed from the Wordpress repository for some violation that claims to let you run with SQLite, but on the whole that path seems a whole lot worse than trying to figure out whatever this is).

Doing the above results in the same behavior (Illegal instruction on mariadb initialization). And again, I think the PHP thing is probably a red herring. In my understanding, PHP only communicates (via php-fpm) with the web server, and the web server, in turn talks to MariaDB via standard sockets. And the nginx/php stuff seems to work fine in any of these configurations. But I am willing to be corrected on this.

Joulinar, 2022-03-20:
MariaDB in its own needs to start successful, independent from any web server or PHP module. Probably there are corruption on MariaDB executable. @MichaIng maybe it's needed to reinstall the database software

MichaIng, 2022-03-20:
Hmm, MariaDB should have been upgraded already and the PHP reinstall should have installed `php7.4-mysql`, which allows PHP to communicate with MariaDB via standard socket, indeed. Let's try to reinstall MariaDB packages then:
```sh
apt install --reinstall mariadb-server mariadb-common mysql-common mariadb-server-10.5 mariadb-server-core-10.5 mariadb-client-10.5 mariadb-client-core-10.5 libmariadb3
journalctl -u mariadb
cat /var/log/mysql/error.log
```

Joulinar, 2022-03-20:
`/var/log/mysql/error.log` might not be available as error logging has been disabled by default on MariaDB 10.5. (Bullseye)

MichaIng, 2022-03-20:
Ah right, thanks for the reminder.

Joulinar, 2022-03-20:
following should do it

```
sed -i 's/#log_error/log_error/' /etc/mysql/mariadb.conf.d/50-server.cnf
systemctl restart mariadb.service
```

MichaIng, 2022-03-20:
This doesn't make sense, the logs double with `journalctl -u mariadb`, or reduce those. It makes very much sense to skip file loggin in favour of journal logging 😉.

Joulinar, 2022-03-20:
this is off-topic but based on my experience, the `error.log` gives way more inside compare to `journalctl`

MichaIng, 2022-03-20:
It is the same when file logging is disabled. Basically enabling file logging removes internal errors from journal, skipping file logging makes MariaDB printing errors to STDERR => journal: https://mariadb.com/kb/en/error-log/#writing-the-error-log-to-stderr-on-unix

jdhaggerty, 2022-03-20:
Yes, of course you're right--php does talk to the db. Silly me. Reinstalling (and trying to increase error logging) gives exactly the same result:
Mar 20 16:13:07 OdroidPi systemd[1]: Starting MariaDB 10.5.12 database server...
Mar 20 16:13:08 OdroidPi systemd[1]: mariadb.service: Main process exited, code=killed, status=4/ILL
Mar 20 16:13:08 OdroidPi systemd[1]: mariadb.service: Failed with result 'signal'.
Mar 20 16:13:08 OdroidPi systemd[1]: Failed to start MariaDB 10.5.12 database server.

Empty /var/log/mysql/error.log, FWIW. It really does seem  to be crashing almost immediately on startup, before anything like logging starts.

jdhaggerty, 2022-03-20:
And just to check, I deleted all my databases just to make sure that some sort of weird data corruption wasn't causing the problem with no love. Clearly I'm grasping at straws at this point.

MichaIng, 2022-03-20:
Nasty. No the databases are not the issue, it is something with the binary itself, respectively the libraries it is loading.

Lol, you know what, I just checked old issues and found: #4790
```sh
G_CONFIG_INJECT 'setenv mesontimer' 'setenv mesontimer "0"' /boot/boot.ini
reboot
```

Might be worth to migrate to our new image, which is based on mainline kernel from Armbian, instead of the very old Hardkernel kernel. But one needs to flash the new image, there is no migration path since the new kernel requires a different partitioning. The above however should solve it.

However, not bad that we cleaned up all the remaining packages from Stretch/Ondrej's PHP repo.

jdhaggerty, 2022-03-21:
Yeah, this has gotten a lot more time consuming than I had originally planned. It's probably time to cut my losses. I'll stick with buster until I have the energy to do a clean upgrade (or move to a different platform).

Thanks for all of your patience and help. This is a great distro.

closed 2022-03-21 by jdhaggerty

MichaIng, 2022-03-22:
The simple fix I provided did not work?